lvgl/lvgl #2193: Linux framebuffer sometime show part of blue background

choujayyl, 2021-04-12:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I use lvgl7.0 with linux framebuffer.But sometimes the screen display part of blue backgrond in place not refresh realtime.

### To Reproduce
I dont konw,but when exit app,it sometime show full blue backgrond.

### Expected behavior

I dont konw,but when exit app,it sometime show full blue backgrond.
### Screenshots or video

[image]

embeddedt, 2021-04-12:
Please try with 7.11; 7.0 is old and has a number of bugs which have since been fixed.

choujayyl, 2021-04-13:
> Please try with 7.11; 7.0 is old and has a number of bugs which have since been fixed.

sorry,I just tried with 7.11.The above is wrong.

kisvegabor, 2021-04-15:
It seems like another app (e.g. window manager) is also using the frame buffer. LVGL draws the screen, the other app overwrites it with blue, LVGL renders the changed areas and they remain visible.

choujayyl, 2021-04-16:
> It seems like another app (e.g. window manager) is also using the frame buffer. LVGL draws the screen, the other app overwrites it with blue, LVGL renders the changed areas and they remain visible.

Yes,It is linux kernel virtual terminal sometime clear the screen.You need to changed it mode to KD_GRAPHICS that only LVGL can use it.
So close issue.

closed 2021-04-16 by choujayyl