lvgl/lvgl #2194: Missing allocation result check

MaJerle, 2021-04-13:
https://github.com/lvgl/lvgl/blob/04f0ffc7756a5f623a3d3ab916efd82250c36e3a/src/core/lv_obj_class.c#L64

xiaoxiang781216, 2021-04-13:
@MaJerle can you provide a patch?

MaJerle, 2021-04-13:
I found this by accident. Later I realized that none of the realloc calls is properly handled. For example:

```
    size_t new_len = ins_len + old_len + 2; /* +2 for terminating NULL and possible \n */
    ext->options        = lv_mem_realloc(ext->options, new_len + 1);
    LV_ASSERT_MEM(ext->options);
    if(ext->options == NULL) return;
```

If realloc fails, this is a memory leak. This example would need to be re-coded to:
```
    size_t new_len = ins_len + old_len + 2; /* +2 for terminating NULL and possible \n */
    void* new_options        = lv_mem_realloc(ext->options, new_len + 1);
    LV_ASSERT_MEM(new_options);
    if(new_options == NULL) return;
    ext->options = new_options;   /* Now we are sure all is OK and new pointer is replacing old one */
```

Proposal is to create a *safe* realloc that accepts *pointer-to-pointer* and returns true if realloc is successful or 0 otherwise.

See my code as a reference: https://github.com/MaJerle/lwmem/blob/5351899d7acb95233d0ed8a789979ccab285751c/lwmem/src/lwmem/lwmem.c#L842-L891

Then you can call the function like:
```
void* p = your_malloc(1234);
/* For the sake of this example, we assume malloc is OK */

/* Try to realloc to bigger size */
if (lwmem_realloc_safe_version(&p, 23456)) {
    /* Pointer p now points to new memory since realloc is successful */
} else {
    /* Pointer p points to old memory since allocation failed */
}

/* Pointer p is here still valid */

kisvegabor, 2021-04-15:
Thaks Tilen, it's really worth considering. 

Do you think this proper error handling is really required? I mean if LVGL is out of memory how do you recover? Will you say "okay, I couldn't set these options on the dropdown, so I delete some screens and try to set options again"?

IMO, `LV_ASSERT_MALLOC()` that halts on malloc/realloc failure is a good compromise between readability, maintainability and safety.

What do you think?

MaJerle, 2021-04-16:
There are some rules with malloc, realloc implementations and so far I'm not sure these are properly used with lv api.
Library sometimes assumes too much and that may have negative effect.

Especially for realloc, if it fails, you have still capability to show something, app just reports to user that no memory is available anymore, *to do something*, but this won't trigger (for example) hard-fault or black screen.

> Do you think this proper error handling is really required? I mean if LVGL is out of memory how do you recover? Will you say "okay, I couldn't set these options on the dropdown, so I delete some screens and try to set options again"?

Yes, it is better to say this rather than entering hard-fault. Believe me, it is easier to debug, maintain and at the end, do proper action if memory allocation fails, which cannot be done if hard-fault occurs.

Just my 2 cents :)

kisvegabor, 2021-04-19:
As these functions "work normally", I think the users are not motivated in checking return values. I mean for example if you try to open a file a lot of issues can happen, so you are motivated in checking the return value of `open`. But in case of widget API the users assume that it works.  We can add an error log on fail though. 

@embeddedt @xiaoxiang781216 What do you think? Is it enough to stop on a `while(1)` and add log-on critical error or shall we use return values to indicate failures?

MaJerle, 2021-04-19:
> As these functions "work normally", I think the users are not motivated in checking return values. I mean for example if you try to open a file a lot of issues can happen, so you are motivated in [...]

Thanks for the answer. Doing this, it means that if application is of type "add new field" and it there is no memory, user gets while loop instead of (for example) message on screen of no-memory-anymore dialog box.

You can trust me that everyone prefer to have return statement checked. You must not assume anything.

xiaoxiang781216, 2021-04-19:
LV_ASSERT_MALLOC can help to find the bug(e.g. memory leak?), but can't replace the runtime check. For example, applicaiton may contain some cache to improve the performance, but can release the cache if LVGL has memory pressure. Returning NULL point give the developer the last chance to recovery if they really want. So, both are useful.

kisvegabor, 2021-04-19:
My problem is that if there is no memory for a button there won't be a memory for an "Out of memory" pop-up either. 
So the question is what the developer can do if he gets out of memory return value?

xiaoxiang781216, 2021-04-19:
@kisvegabor application may manage the cache internally, so LVGL is out of memory, doesn't mean the whole system out of memory:).

kisvegabor, 2021-04-19:
I've checked that there are about 100 malloc/realloc in LVGL. Not that few but managable. Can you imagine other generic error codes too that can be returned by widgets?

embeddedt, 2021-04-19:
> there won't be a memory for an "Out of memory" pop-up either.
So the question is what the developer can do if he gets out of memory return value?

In most cases, deleting every other widget should free enough memory to display a simple out-of-memory popup. If there is really no memory left after that, the device probably doesn't have enough heap to run stably anyways.

For my projects I'd be satisfied with a panic only (e.g. with a callback to reset the device) but this isn't practical for a device in the field that needs to run long-term.

MaJerle, 2021-04-19:
In the runtime, you may have widgets that are added dynamically. Out of memory must not crash the application.

C47D, 2022-01-25:
As an exercise I ran `grep -r lv_mem_realloc --exclude-dir=tests | wc` on the current master (71c739cc) to try to calculate how much time it would be required to fix the reallocation failure. There are 43 places where lv_mem_realloc` is being used, not that many. Do you have an action plan on this @kisvegabor ?

kisvegabor, 2022-01-25:
@C47D The main question in what to do if any malloc fails. Some people support the idea of introducing the concept of returning error codes from almost all function. It'd be a huge work and I don't think it's the best direction to take. 

IMO adding ASSERTs is enough for the vast majority of the cases and it can be done with much less effort.

C47D, 2022-01-25:
Can we do something like this?

```c
void* p = your_malloc(1234);
/* For the sake of this example, we assume malloc is OK */

/* Try to realloc to bigger size */
tmp = lv_mem_realloc(&p, 23456)
if (tmp) {
    /* Pointer p now points to new memory since realloc is successful */
    p = tmp;
} else {
    /* Pointer p points to old memory since allocation failed */
    LV_ASSERT_MALLOC(tmp);
}

/* Pointer p is here still valid */
```

embeddedt, 2022-01-25:
We still need a way of conveying the failure in a way that makes sense. Suppose, to build on your example, that there was an allocation involved in changing the state of a button to disabled. If the allocation fails, leaving the button enabled could be disastrous, as a lot of UIs would want to disable the button immediately after the first click to prevent double clicks.

If this were C++ it would make sense to throw an exception here, but we can't do that in C.

C47D, 2022-01-25:
Do we know how other UI libraries handle memory allocation failures?

MaJerle, 2022-01-25:
Functions that can take an action shall return either error (0) or a NULL pointer, on failure. These are mostly functions to create widgets.

> If this were C++ it would make sense to throw an exception here, but we can't do that in C.

That is the worst idea for embedded systems you could imagine, unless you target Linux

kisvegabor, 2022-01-25:
> Can we do something like this?

For `realloc` it's fine, but IMO `NULL` is expressive enough so no need for `LV_MEM_REALLOC_FAILED`.

> We still need a way of conveying the failure in a way that makes sen

I still vote for ASSERTs. During the development you should see if the system is out of RAM. 

> Functions that can take an action shall return either error (0) or a NULL pointer, on failure. These are mostly functions to create widgets.

LVGL performs a lot of `malloc`s during rendering too. Handling and returning errors form there would be a mess.

MaJerle, 2022-01-25:
@kisvegabor are you sure final application is based on development only? GUI can be dynamic (that's why you have dynamic allocation aka heap) story. What if user wants to display list of MP3 songs on display in list-view? Crashing the system because of no available memory is a criminal in that case.

I do vote for `return NULL`, if allocation fails.

My 2cnts

kisvegabor, 2022-01-25:
@MaJerle 
I'm thinking a little bit differently about it. What can you do if you are out of RAM? Show a popup about "too many songs?". It'd be odd.  But what if there is no RAM for the popup either? Will you delete some screens? If you can delete/recreate screens run time in case of a failure why don't you do it by default?

So my approach for mp3 list thing is you should know that you have RAM to display max 20 items and implement pagination if there are more items.

As RAM is a finite you should design the UI to respect these limitations and ensure that there is a healthy amount of free RAM.
And it can be tested and measured during development too. 

> I do vote for return NULL, if allocation fails.

Speaking only about widget creation, returning `NULL` on failure seems possible. 
But what about setting too many options for a drop down, too long text for a label, writing too many letters in a text area, etc?

The case of  text area is really interesting. IMO it's the developer's task to ensure that the user can't write infinite long text there instead of doing some error handling after failing to add the 15,000th character.  

If normal `malloc` is used and and there in no more heap other parts of the system might crash too (external libs in which you have no control). Assuming that everything will operate in healthy way when out of ram is not realistic in my opinion.  

Instead we should avoid being out of RAM.

---

Adding return values would be a huge conceptual rework. Adding asserts is a low hanging fruit and already solves a lot of issues.

MaJerle, 2022-01-25:
Every return malloc must be rechecked against NULL. If it is not, its a huge problem.

IMO, lib must be resistant regardless to what user does. If user will display popup or not its up to them. Maybe they will create popup earlier during init ajd hide it. Maybe they will start shortening list on runtime. Point of heap memory is to allow reuse, but if its not available, you shall not asume anything.

Setting textarea text shall return false if buffer is not long enough. Instead of void functions, make them uint8_t to return state.

[image]

xiaoxiang781216, 2022-01-25:
From practice, I would expect LVGL could release the internal cache(e.g. image cache) and provide some hook to platform in case of out of memory(like https://en.cppreference.com/w/cpp/memory/new/set_new_handler), and try allocate again. this infrastructure is enough to handle the abnormal case.

C47D, 2022-01-27:
Maybe we can start by applying this pattern (assigning return of `lv_mem_realloc` to a temporary variable) when we found `lv_mem_realloc` being used, I found 7 usages while working on `lv_table`.

```c
    size_t new_len = ins_len + old_len + 2; /* +2 for terminating NULL and possible \n */
    void* new_options        = lv_mem_realloc(ext->options, new_len + 1);
    LV_ASSERT_MEM(new_options);
    if(new_options == NULL) return;
    ext->options = new_options;   /* Now we are sure all is OK and new pointer is replacing old one */
```

We can left the Out Of Memory handling for later.

MaJerle, 2022-01-27:
Thanks to consider. A safe realloc function could be implemented, then you reduce caller code size and gain readability and maintenance. See my original post and my first reply for idea.

kisvegabor, 2022-01-27:
@xiaoxiang781216 
> provide some hook to platform in case of out of memory

I think it's a wonderful idea. Easy to implement and solves a lot of possible issues.

@MaJerle 
>  if (!lv_mem_realloc_s(&ext->options, new_len + 1)) {

This is also a great idea. We can update to `lv_mem_realloc_s` gradually without changing the whole concept of LVGL's error handling.

@C47D if you also like this approach, I suggest using it. (I'd call it `lv_mem_realloc_safe`).

C47D, 2022-01-28:
I'm checking the default mem handler and it seems like it already handles the reallocation failure, I guess we only need to update `lv_mem_realloc_safe` when `LV_MEM_CUSTOM` is enabled.

```
/*
** The TLSF block information provides us with enough information to
** provide a reasonably intelligent implementation of realloc, growing or
** shrinking the currently allocated block as required.
**
** This routine handles the somewhat esoteric edge cases of realloc:
** - a non-zero size with a null pointer will behave like malloc
** - a zero size with a non-null pointer will behave like free
** - a request that cannot be satisfied will leave the original buffer
**   untouched
** - an extended buffer size will leave the newly-allocated area with
**   contents undefined
*/
void * lv_tlsf_realloc(lv_tlsf_t tlsf, void * ptr, size_t size)
```

kisvegabor, 2022-01-29:
> a non-zero size with a null pointer will behave like malloc

Some thing similar is [already handled](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_mem.c#L191) but we can  add this too: 
```c
    if(data_p == NULL) return lv_mem_alloc(new_size);
```

> a zero size with a non-null pointer will behave like free

It's already handled by [lv_mem_realloc](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_mem.c#L185-L189)

> a request that cannot be satisfied will leave the original buffer untouched

I think @MaJerle's suggestion with passing the pointer of pointer is better as it results in simpler code. 
With tlsf way of thinking
```c
void * data_tmp = lv_mem_realloc(data, 100);
if(data_tmp) data = data_tmp;
else ...error...
```

With double pointer
```
if(!lv_mem_realloc_safe(&data, 100)) ...error...
```

> I guess we only need to update lv_mem_realloc_safe when LV_MEM_CUSTOM is enabled.

If you agree with the above we should use a different approach as tlsf handles the the out of memory with realloc. So all I'd do is something like:
```c
bool lv_mem_realloc_safe(void ** p, size_t size)
{
   void * p_tmp = lv_mem_realloc(*p, size);
   if(p_tmp == NULL)  return false;
   
   *p = p_tmp;
   return true;    
}
```

MaJerle, 2022-01-29:
@kisvegabor if you pass 0 to realloc safe, it should set pointer to null and return true right?

kisvegabor, 2022-01-29:
`lv_mem_realloc` return `&zero_mem` if size is 0. It saves some error checks later as there is difference between a pointer is `NULL` (not initialized, not used yet) or used but actually it points to a zero sized array.

MaJerle, 2022-01-29:
realloc that frees the memory,should set pointer to NULL. Thats the only way to tell user that it cannot use the pointer anymore.

Zero sized array should be 0. User cannot write nor read with it. Seems arhitecture is a bit odd then.

Am I wrong?

embeddedt, 2022-01-29:
Returning a pointer is also legal, according to [Open Group](https://pubs.opengroup.org/onlinepubs/007904975/functions/realloc.html#tag_03_598_04).

> If size is 0, either a null pointer or a unique pointer that can be successfully passed to free() shall be returned.

C47D, 2022-01-31:
I've added handling of the special cases and a couple of silly tests for them in #3055, should we continue the discussion over there?

kisvegabor, 2022-01-31:
> should we continue the discussion over there?

Yes, but let's leave it open as [the hook](https://github.com/lvgl/lvgl/issues/2194#issuecomment-1021425748) suggested by @xiaoxiang781216 shall be added too.

X-Ryl669, 2022-02-03:
I don't understand the new safe realloc interface.
Most other library providing safe realloc have an invariant: either the allocation worked or it failed. This greatly simplifies the design. 
Typically, lvgl could be changed with a very simple scheme without modifying the code everywhere:
```
void * lv_mem_realloc(void * ptr, size_t new_size) {
  void * tmp = realloc(ptr, new_size);
  if (tmp == NULL) free(ptr);
  return tmp;
}
```
If realloc fails, you have a lot more problems than trying to deal with reusing the previous pointer. 
Using a different signature (with a LV_RES_T) and a pointer to pointer is nonsense, IMHO, because it'll bloat the code with dealing with issues you can't solve anyway.
You previous code doing this:
```
void some_func() {
[part 1...]
ptr = realloc(ptr, new_size);
LV_ASSERT_MALLOC(ptr);
[part 2...]
}
```
is now doing this:
```
lv_res_t some_func() {
[part 1...]
res = realloc(&ptr, new_size);
if (res == FAILED) {
   unwind part 1;
   return FAILED;
}
[part 2...]
}
```

Except.... that you might have effects in [part 1]  you can't unwind (like, for example, one of the pointer was realloc'd also in part 1, calling constructors for object etc...). Also, it force user of some_func to deal with the memory error and hell starts to freeze.

Right now, the application crashes or assert exactly where it should, and it's much easier to debug.

Fixing the leak is easy (it's one line), but changing the signature seems like a huge PITA.

MaJerle, 2022-02-03:
@X-Ryl669 if realloc fails, you are not supposed to free previous buffer. You should keep previous allocated value and return FALSE to user, saying realloc was not possible. Then its up to app to handle it they way they want..

X-Ryl669, 2022-02-03:
> if realloc fails, you are not supposed to free previous buffer. You should keep previous allocated value and return FALSE to user, saying realloc was not possible. 

There is no supposition here. realloc does not free the previous buffer. A safe realloc could free the previous buffer (in fact most safe realloc implementation does this, since it's the most logical action to do). realloc does not return false, it returns NULL. And most of the code will already deal with this NULL return (either by asserting/stopping in debugger, either by some error handling).

By changing the signature of the function, you're forcing the error handling of the code (because if you ignore it, the later code will crash when it tries to use the new allocated element that don't exist). So you end up with some very manual and labor intensive code where you must return earlier, undo what you might already have done (like, to take an example, when changing the dimension of a table, the error might happen when you're reallocating some text on the last label of the last cell of the table => you must recreate a table that's smaller than it is at the time of the crash: it means calling the destructors of all chidren objects and so on). 
That's plenty of code (understand: flash space on embedded system) for what? Even if you clean correctly, the next function call will behave the same because of the lack of heap, and mostly your application is broken (expected actions don't happen if you have to rewind them all). 

The only solution is to free some heap (a large amount), so that's what @xiaoxiang781216 proposed. At least, you'll gain some time until you're out of heap again. And it's even better if all hidden from the user, that is, mem_realloc failed, called a hook, the hook freed some memory from the cache, and mem_realloc retried and succeeded.

By changing the contract of the mem_realloc to free the buffer like I suggested, you're not saving the planet, but at least, most code will not be leaking the ptr anymore (since many code ends up in `ptr = realloc(ptr...)`) so you don't increase the issue you're already in.

> Btw, we are in C, not C++, so constructors, and such story, doesn't apply at all.

That's not how LVGL is working. When you work with lv_obj_t, to create a widget, it'll call its constructor chain like in C++. And if you look at the place where realloc is used, many of them are dealing/creating with lv_obj_t. So you end up with allocation chains that's not trivial to delete / restore to a previous state.

MaJerle, 2022-02-03:
I disagree with these comments and will step out of this discussion.

Realloc is now used like this:
```c
    size_t new_len = ins_len + old_len + 2; /* +2 for terminating NULL and possible \n */
    ext->options        = lv_mem_realloc(ext->options, new_len + 1);
    LV_ASSERT_MEM(ext->options);
    if(ext->options == NULL) return;
```
That's obvious memory leak on failure, that needs to be corrected.

> That's not how LVGL is working. When you work with lv_obj_t, to create a widget, it'll call its constructor chain like in C++. And if you look at the place where realloc is used, many of them are [...]

Then we have a fundamental architecture failure.

MaJerle, 2022-02-03:
How would you solve this. You have a string for button (label) and user wants to change label. If realloc fails, naturally function to set label returns false aka cannot do it (or returns `falseMemoryNotAvail`..). In your example, button would disappear :)

> Yes. You are completely right. And changing the semantic of lv_mem_realloc to free the pointer on failure is the best solution to avoid this leak, IMHO. It's what most safe_realloc functions are doing (see [here](https://github.com/alexanderbaulin/Telegram/blob/e94a96ad068d5d2fe1a1a2c98629f2d083fa2b03/TMessagesProj/jni/exoplayer/libFLAC/include/share/alloc.h#L156)).

Yes, its another way of doing it. The ultimate question is:

- Do you want to completely terminate eveerything on memory failure? (your proposal)
- Do you want to keep state as-is and do not modify it, in case of failure

I am for nr.2, you are for nr.1

There are also 2 points to know:
- We are NOT changing realloc semantic, we are adding new feature
- Existing code anyway has a bug, so its a question, do we fix it with proposal nr.1 or nr.2 in previous question.

If we fix it by terminate everything, then we may need to delete widget too, so it means even additional work and flash usage.
Frankly, flash usage is one function call more, so like 10bytes of instructions on Cortex-M :)

X-Ryl669, 2022-02-04:
> There are also 2 points to know:
>    We are NOT changing realloc semantic, we are adding new feature

Yes I understood this. But exploiting this feature implies modifying the calling code (so every place where realloc was called).
And that's where the handling is either bloating the flash or not working anyway. 

>    Existing code anyway has a bug, so its a question, do we fix it with proposal nr.1 or nr.2 in previous question.

Yes exactly. Right now, the *effects* of the bug is that the application will likely crash. On an embedded system, the watchdog will trigger, the application will restart. No additional flash space is used here, that's just how it happens to work.

With nr.2, you'll need to unwrap, correct/fix the code behavior to either:
1. Restore to a working state (in your example, meaning restoring to the previous label button's text). This means remembering the working state (if you deleted the text, you can't restore it, so you need to copy the previous text somewhere, realloc, fail, copy the previous text back to the label's text). Then you'll return to the calling code. But then, you have to change the signature of the calling code to be able to return an error to the calling code. The calling code now needs more code to deal with the error (for example, by displaying a message box with "No more memory"). How you do that without heap, I don't know, but anyway, you see how this leads to a lot more code. 
2. Restore to an error state. This means however having to check for the error state in every function using the object. It's adding few bytes of code, but since it's multiplied by the number of functions using the object, it can lead to a lot more code. In that case, I consider the application as broken anyway, and restarting the application is the only sane way to recover. Understand: this extra code is just bloat compared to point nr.1

Honestly, I know that solution 2 and point 1 is the correct way to go, **if you had unlimited resources**. It's what C++ exception was promising to developer. But in reality, it failed to fulfill its promises because [writing exception safe code is hard](https://mortoray.com/2013/12/05/is-exception-safe-code-truly-possible/). Doing the same in C and in LVGL is prone to code bloat (demonstrated above) and in the end, if we weight the benefit versus the effort and resource consumption I don't think it's worth it.

> Frankly, flash usage is one function call more, so like 10bytes of instructions on Cortex-M :)

Yes for the safe_realloc function. But the bloat does not come from this function. It comes from fixing all the calling code to deal with the out-of-memory exception/error. See above.

MaJerle, 2022-02-04:
> Yes I understood this. But exploiting this feature implies modifying the calling code (so every place where realloc was called). And that's where the handling is either bloating the flash or not working anyway.

We anyway need to fix it as it has a bug :)

> Yes exactly. Right now, the _effects_ of the bug is that the application will likely crash. On an embedded system, the watchdog will trigger, the application will restart. No additional flash space [...]

If we go this way, lvgl is usable only for hobby, there is no way you pass any software certification in serious product with that

> With nr.2, you'll need to unwrap, correct/fix the code behavior to either:
>
> 1. Restore to a working state (in your example, meaning restoring to the previous label button's text). This means [...]

No arrogance in following text, but I honestly think you don't know how realloc works, so let me explain it to you.
If you have a piece of memory, that has been allocated previously by `malloc`, then realloc does (simplified thing):

- Shrinks memory size if you set new size to be lower than previous one, keeps memory as is
- Tries to enlarge existing block if possible (when new size is possible), if not tries to allocate new block, copy content there and free old block

Keep in mind that heap management system can use *first-fit*, *last-fit*, *best-fit*, etc, so it may happen that new block is allocated at new location even if new size is smaller than previous one.

But what realloc doesn't do is: It never modifies old memory if realloc failed because it couldn't do job that is supposed to do. So bottom-line, we don't do any modifications or unwrapping. We simply return false and that's it.

```c
ext->options = malloc(...); //whatever is the size
ext->options_len = some_len; //whateveer is the len..

size_t new_len = ext->options_len + new_added_len;

// Try to realloc block, if cannot be done, ext->options stays as is.
if (realloc_safe(&ext->options, new_len) == FALSE) {
    //Return and basta, new operations doesn't work anymore
    return FALSE;
}
ext->options_len = new_len; //Store new len (only if necessary).
```

What are we unwrapping here, bitte?

X-Ryl669, 2022-02-04:
> What are we unwrapping here, bitte? That code is actually smaller than existing one, that even has a bug at many places.

It's here: `return FALSE`. And it's where you haven't written in the example code, the calling code that now has to deal with the action not happening. And the calling code of the calling code.
Typically, if you look at current code you'll have:

```cpp
 void event_handler(...) {
     if (event == LV_CLICKED && event->target == stopNuclearPlantBtn) 
    {
        dropGraphiteBarsInReactor();
        shutdownHour.set_options("12:00\n14:00\n..."); // Just to call your function
        shutDownNuclearReactor();
    }
}
```
In the new version will need:
```cpp
  bool event_handler(...) {
     if (event == LV_CLICKED && event->target == stopNuclearPlantBtn) 
    {
        if (dropGraphiteBarsInReactor() == FALSE) {
          if (showErrorMsgBox("No more memory") == FALSE) {
             terminate();
          }
       }
       if (!shutdownHour.set_options("12:00\n14:00\n...")) {
          if (showErrorMsgBox("No more memory") == FALSE) {
             terminate();
          }
      }
      if (shutDownNuclearReactor() == FALSE) {
          if (showErrorMsgBox("No more memory") == FALSE) {
             terminate();
          }
      }
}
```
Do you see now what I call bloat here ?
And what I discussed above, that it's very likely that `showErrorMsgBox("No more memory")` will try to allocate memory too and will fail, finally requiring calling terminate() thus preventing calling the very last (and important) function.
If you want to "pass certification", you need a real unwrapping that'll look like this: (the code will become even more convoluted)
```cpp
  bool event_handler(...) {
     if (event == LV_CLICKED && event->target == stopNuclearPlantBtn) 
    {
        if (dropGraphiteBarsInReactor() == FALSE) {
          if (showErrorMsgBox("No more memory") == FALSE) {
             terminate();
          }
          // Houston, we have a problem here, let's the fire soldier solve this
          return false;
       }
       if (!shutdownHour.set_options("12:00\n14:00\n...")) {
          if (showErrorMsgBox("No more memory") == FALSE) {
             terminate();
          }
          undropGraphiteBarsFromReactor(); // WTF, is that even possible ?
         return false;
      }
      if (shutDownNuclearReactor() == FALSE) {
          if (showErrorMsgBox("No more memory") == FALSE) {
             terminate();
          }
          if (shutdownHour.set_options("What were the previous options... let me think about it... Dude, I forgot to save them") == FALSE) {
              terminate();
          }
          undropGraphiteBarsFromReactor(); // WTF, is that even possible ?
          return false;
      }
      return true;
}

```

BTW, I've worked in automotive sector and aerospace sector, and I know about MISRA and the various DAL level. I *know* about how to write actual safe code upon exceptions and it's not really something I like to do. If you expect to make LVGL certified for a MISRA C, you'll have to rewrite all code since malloc is discouraged in MISRA context. It means no realloc too.
There is no certified TLSF heap allocator that I'm aware of.

Said differently, I'm saying you should not do that. I'm saying that the actual code does not do that and getting it to do that, is something that'll require a huge amount of work. And for an hypothetical gain in safety of your code.

MaJerle, 2022-02-04:
Then we need to separate things.

If you as a user don't want to do error checks, you don't lose anything you had so far, because you didn't have this option anyway.
But what matters is the way library does the job. It must be protected against potential crazy user inputs. That's the whole point. What you do in your handler, lib doesn't care. If you want to trigger hard-fault and then watchdog, go ahead, but assumption saying "let's rely on watchdog" must not be priority. Btw, I am not talking about MISRA-C. For non-critical apps, malloc is OK, if you can properly handle failures. Here lib is doing it badly.

> BTW, I've worked in automotive sector and aerospace sector, and I know about MISRA and the various DAL level. I know about how to write actual safe code upon exceptions and it's not really something I like to do.

I am very sorry to underestimate you, but you have been unclear with your statements.
I will close the ticket right now and let @kisvegabor to decide as he is the leader.

closed 2022-02-04 by MaJerle

embeddedt, 2022-02-04:
In my humble opinion, what is needed here is the ability for the platform/LVGL/some layer to provide a second emergency heap that is used once the main heap runs out. The emergency heap should be large enough to carry out whatever task was needed, and inform the user that the system is out of memory and needs to be restarted. This does not eliminate the chance of an allocation failing, but it greatly reduces it.

Proper error codes on each function are the only true solution to the problem, but I don't think @kisvegabor is currently looking to go in that direction.