lvgl/lvgl #2198: why canvas can't show on screen?

MrsKan, 2021-04-15:
lv_canvas_set_buffer(,,,), when color formate set to LV_IMG_CF_TRUE_COLOR, canvas can't show on screen. 
But when set to LV_IMG_CF_TRUE_COLOR_ALPHA, it can show.
PC simulator is all right.
My device is based on stm32f429, mcu interface display,16bit width.
How can I fix this problem?
thx.

kisvegabor, 2021-04-19:
Please a code snippet to reproduce the issue. I can test it on STM32F429 Discovery.

MrsKan, 2021-04-20:
`#define CANVAS_WIDTH   100
#define CANVAS_HEIGHT  100
extern u8 picbuf[];
lv_color_t cbuf[LV_CANVAS_BUF_SIZE_TRUE_COLOR_ALPHA(CANVAS_WIDTH, CANVAS_HEIGHT)];
lv_obj_t * canvas;
void fp_page()
{
	fppage_scr = lv_obj_create(NULL, NULL);
	lv_obj_t* this_scr = fppage_scr;
	lv_obj_set_event_cb(this_scr, NULL);

	/*Create a canvas and initialize its the palette*/
	canvas = lv_canvas_create(this_scr, NULL);
	lv_canvas_set_buffer(canvas, cbuf, CANVAS_WIDTH, CANVAS_HEIGHT, LV_IMG_CF_TRUE_COLOR);
	lv_obj_align(canvas, NULL, LV_ALIGN_CENTER, 0, 0);
	
	lv_canvas_fill_bg(canvas, LV_COLOR_GREEN, LV_OPA_COVER);
	lv_obj_set_style_local_bg_color(canvas, LV_CANVAS_PART_MAIN, LV_STATE_DEFAULT, lv_color_make(193, 210, 240));
	lv_obj_set_style_local_bg_opa(canvas, LV_CANVAS_PART_MAIN, LV_STATE_DEFAULT, LV_OPA_COVER);	
}`
when use lv_scr_load_anim function , the canvas can show in the animation process,when animation finshed, canvas disapper.

kisvegabor, 2021-04-21:
I've tested it on STM32F429 Discovery but it works well here.

I've loaded the screen liek this:
```
    lv_scr_load_anim(this_scr, LV_SCR_LOAD_ANIM_MOVE_LEFT, 400, 0, false);
```

Are you sure your display driver works well?
Are other widgets displayed properly?

closed 2021-05-14 by kisvegabor