lvgl/lvgl #2201: I found a bug when using the Asan tool: stack-use-after-scope

jianjunxia, 2021-04-19:
Here is more information about this bug：
`==247937==ERROR: AddressSanitizer: stack-use-after-scope on address 0x7fffe3120c50 at pc 0x7ffff702bc53 bp 0x7fffe3120680 sp 0x7fffe3120670
READ of size 8 at 0x7fffe3120c50 thread T18 (lv_task)
    #0 0x7ffff702bc52 in lv_draw_mask_apply /home/light/project/test_version/voice_control/components/lvgl/src/lv_draw/lv_draw_mask.c:115
    #1 0x7ffff700d471 in lv_draw_map /home/light/project/test_version/voice_control/components/lvgl/src/lv_draw/lv_draw_img.c:603
    #2 0x7ffff700af80 in lv_img_draw_core /home/light/project/test_version/voice_control/components/lvgl/src/lv_draw/lv_draw_img.c:282
    #3 0x7ffff700a65e in lv_draw_img /home/light/project/test_version/voice_control/components/lvgl/src/lv_draw/lv_draw_img.c:91
    #4 0x7ffff701c499 in draw_pattern /home/light/project/test_version/voice_control/components/lvgl/src/lv_draw/lv_draw_rect.c:1284
    #5 0x7ffff700fede in lv_draw_rect /home/light/project/test_version/voice_control/components/lvgl/src/lv_draw/lv_draw_rect.c:112
    #6 0x7ffff6f46265 in lv_obj_design /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_obj.c:3906
    #7 0x7ffff6f2df4a in lv_refr_obj /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:677
    #8 0x7ffff6f2e17c in lv_refr_obj /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:717
    #9 0x7ffff6f2e17c in lv_refr_obj /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:717
    #10 0x7ffff6f2e17c in lv_refr_obj /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:717
    #11 0x7ffff6f2e17c in lv_refr_obj /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:717
    #12 0x7ffff6f2e17c in lv_refr_obj /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:717
    #13 0x7ffff6f2e17c in lv_refr_obj /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:717
    #14 0x7ffff6f2e17c in lv_refr_obj /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:717
    #15 0x7ffff6f2db23 in lv_refr_obj_and_children /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:619
    #16 0x7ffff6f2d826 in lv_refr_area_part /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:550
    #17 0x7ffff6f2ce0b in lv_refr_area /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:460
    #18 0x7ffff6f2c3e4 in lv_refr_areas /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:382
    #19 0x7ffff6f2b60c in _lv_disp_refr_task /home/light/project/test_version/voice_control/components/lvgl/src/lv_core/lv_refr.c:199
    #20 0x7ffff703efa7 in lv_task_exec /home/light/project/test_version/voice_control/components/lvgl/src/lv_misc/lv_task.c:409
    #21 0x7ffff703e382 in lv_task_handler /home/light/project/test_version/voice_control/components/lvgl/src/lv_misc/lv_task.c:142
    #22 0x7ffff6ed8914 in __lv_task_thrd /home/light/project/test_version/voice_control/components/gui_core/src/tuya_gui_core.c:21
    #23 0x5555555cfd2c in __WrapRunFunc /root/workspace_temp/EmbedSDKs/ty_iot_gw_wr_sdk/ty_iot_gw_wr_sdk/sdk/base_utilities/src/uni_thread.c:250
    #24 0x555555682bd2 in tuya_os_adapt_thread_wrap_func /root/workspace_temp/EmbedSDKs/ty_iot_gw_wr_sdk/ty_iot_gw_wr_sdk/sdk/base_os_adapter/OS_adapter_linux/src/system/tuya_os_adapt_thread.c:44
    #25 0x7ffff6a65608 in start_thread /build/glibc-ZN95T4/glibc-2.31/nptl/pthread_create.c:477
    #26 0x7ffff698c292 in __clone (/lib/x86_64-linux-gnu/libc.so.6+0x122292)

Address 0x7fffe3120c50 is located in stack of thread T18 (lv_task) at offset 256 in frame
    #0 0x7ffff701b210 in draw_pattern /home/light/project/test_version/voice_control/components/lvgl/src/lv_draw/lv_draw_rect.c:1187`

The following is my analysis：
in lv_draw_rect.c draw_pattern( )function, line_1239,you allocated a stack memory, and in line_1241 lv_draw_mask_add() function called, I think you used this stack exit address in another place

embeddedt, 2021-04-19:
Hi,

Could you share which commit of LVGL you are using? Lines 1239 and 1241 appear to be slightly off in `release/v7`.

jianjunxia, 2021-04-19:
> Hi,
> 
> Could you share which commit of LVGL you are using? Lines 1239 and 1241 appear to be slightly off in `release/v7`.

Hi,
The version I'm using is release/v7, But sorry, I forgot to take my computer home when I got off work, so I can't remember which specific commit it is. It's already 23:20 in China, I will tell you in the morning which commit I used.(about nine hours later),
I remember the code as follows:
in line 1239:   lv_draw_mask_radius_param_t radius_mask_param;
in line 1241:   int16_t radius_mask_id = lv_draw_mask_add(&radius_mask_param, NULL);
according to the Asan tool, the last error was located at lv_draw_mask.c:115,  code as follows:
res = dsc->cb(mask_buf, abs_x, abs_y, len, (void *)m->param);

jianjunxia, 2021-04-20:
> Hi,
> 
> Could you share which commit of LVGL you are using? Lines 1239 and 1241 appear to be slightly off in `release/v7`.

Hi,
The specific version I use is release/v7.7.1

The following is the latest printing information of Asan Tool：
==268023==ERROR: AddressSanitizer: stack-use-after-scope on address 0x7fffe311cc50 at pc 0x7ffff702baa9 bp 0x7fffe311c680 sp 0x7fffe311c670 READ of size 8 at 0x7fffe311cc50 thread T18 (lv_task)

#0 0x7ffff702baa8 in lv_draw_mask_apply  /lvgl/src/lv_draw/lv_draw_mask.c:115
#1 0x7ffff700d45f in lv_draw_map  /lvgl/src/lv_draw/lv_draw_img.c:595
#2 0x7ffff700af7d in lv_img_draw /lvgl/src/lv_draw/lv_draw_img.c:275
#3 0x7ffff700a65e in lv_draw_img /lvgl/src/lv_draw/lv_draw_img.c:84
#4 0x7ffff701c2ef in draw_pattern /lvgl/src/lv_draw/lv_draw_rect.c:1284
#5 0x7ffff700fd34 in lv_draw_rect /lvgl/src/lv_draw/lv_draw_rect.c:112
#6 0x7ffff6f46265 in lv_obj_design /lvgl/src/lv_core/lv_obj.c:3906
#7 0x7ffff6f2df4a in lv_refr_obj /lvgl/src/lv_core/lv_refr.c:677
#8 0x7ffff6f2e17c in lv_refr_obj /lvgl/src/lv_core/lv_refr.c:717
#9 0x7ffff6f2e17c in lv_refr_obj /lvgl/src/lv_core/lv_refr.c:717
#10 0x7ffff6f2e17c in lv_refr_obj /lvgl/src/lv_core/lv_refr.c:717
#11 0x7ffff6f2e17c in lv_refr_obj /lvgl/src/lv_core/lv_refr.c:717
#12 0x7ffff6f2e17c in lv_refr_obj /lvgl/src/lv_core/lv_refr.c:717
#13 0x7ffff6f2e17c in lv_refr_obj /lvgl/src/lv_core/lv_refr.c:717
#14 0x7ffff6f2e17c in lv_refr_obj /lvgl/src/lv_core/lv_refr.c:717
#15 0x7ffff6f2db23 in lv_refr_obj_and_children /lvgl/src/lv_core/lv_refr.c:619
#16 0x7ffff6f2d826 in lv_refr_area_part /lvgl/src/lv_core/lv_refr.c:550
#17 0x7ffff6f2ce0b in lv_refr_area /lvgl/src/lv_core/lv_refr.c:460
#18 0x7ffff6f2c3e4 in lv_refr_areas /lvgl/src/lv_core/lv_refr.c:382
#19 0x7ffff6f2b60c in _lv_disp_refr_task /lvgl/src/lv_core/lv_refr.c:199
#20 0x7ffff703edfd in lv_task_exec /lvgl/src/lv_misc/lv_task.c:409
#21 0x7ffff703e1d8 in lv_task_handler /lvgl/src/lv_misc/lv_task.c:142

Address 0x7fffe311cc50 is located in stack of thread T18 (lv_task) at offset 256 in frame
#0 0x7ffff701b066 in draw_pattern /lvgl/src/lv_draw/lv_draw_rect.c:1187

jianjunxia, 2021-04-20:
**> Hi,
> 
> Could you share which commit of LVGL you are using? Lines 1239 and 1241 appear to be slightly off in `release/v7`.

Hi,
I already know how to solve this problem, and I see that you have solved this problem in the updated version

Old code：
`static void draw_pattern(const lv_area_t * coords, const lv_area_t * clip, const lv_draw_rect_dsc_t * dsc)
{
    if(dsc->pattern_image == NULL) return;
    if(dsc->pattern_opa <= LV_OPA_MIN) return;

    lv_img_src_t src_type = lv_img_src_get_type(dsc->pattern_image);

    lv_draw_img_dsc_t img_dsc;
    lv_draw_label_dsc_t label_dsc;
    int32_t img_w;
    int32_t img_h;

    if(src_type == LV_IMG_SRC_FILE || src_type == LV_IMG_SRC_VARIABLE) {
        lv_img_header_t header;
        lv_res_t res = lv_img_decoder_get_info(dsc->pattern_image, &header);
        if(res != LV_RES_OK) {
            LV_LOG_WARN("draw_img: can't get image info");
            return;
        }

        img_w = header.w;
        img_h = header.h;

        lv_draw_img_dsc_init(&img_dsc);
        img_dsc.opa = dsc->pattern_opa;
        img_dsc.recolor_opa = dsc->pattern_recolor_opa;
        img_dsc.recolor = dsc->pattern_recolor;
    }
    else if(src_type == LV_IMG_SRC_SYMBOL) {
        lv_draw_label_dsc_init(&label_dsc);
        label_dsc.color = dsc->pattern_recolor;
        label_dsc.font = dsc->pattern_font;
        label_dsc.opa = dsc->pattern_opa;
        lv_point_t s;
        _lv_txt_get_size(&s, dsc->pattern_image, label_dsc.font, label_dsc.letter_space, label_dsc.line_space, LV_COORD_MAX,
                         LV_TXT_FLAG_NONE);
        img_w = s.x;
        img_h = s.y;

    }
    else {
        /*Trigger the error handler of image drawer*/
        LV_LOG_WARN("lv_img_design: image source type is unknown");
        lv_draw_img(coords, clip, NULL, NULL);
        return;
    }

    /*Can't draw zero sized images*/
    if(img_w == 0 || img_h == 0) return;

    lv_area_t coords_tmp;

    if(dsc->pattern_repeat) {
        lv_draw_mask_radius_param_t radius_mask_param;
        lv_draw_mask_radius_init(&radius_mask_param, coords, dsc->radius, false);
        int16_t radius_mask_id = lv_draw_mask_add(&radius_mask_param, NULL);

        /*Align the pattern to the middle*/
        int32_t ofs_x = (lv_area_get_width(coords) - (lv_area_get_width(coords) / img_w) * img_w) / 2;
        int32_t ofs_y = (lv_area_get_height(coords) - (lv_area_get_height(coords) / img_h) * img_h) / 2;

        coords_tmp.y1 = coords->y1 - ofs_y;
[... 121 lines cut ...]
        coords_tmp.x2 = coords_tmp.x1 + img_w - 1;
        coords_tmp.y2 = coords_tmp.y1 + img_h - 1;

        /* If the (obj_h - img_h) is odd there is a rounding error when divided by 2.
         * It's better round up in case of symbols because probably there is some extra space in the bottom
         * due to the base line of font*/
        if(src_type == LV_IMG_SRC_SYMBOL) {
            int32_t y_corr = (obj_h - img_h) & 0x1;
            coords_tmp.y1 += y_corr;
            coords_tmp.y2 += y_corr;
        }

        int16_t radius_mask_id = LV_MASK_ID_INV;
        if(_lv_area_is_in(&coords_tmp, coords, dsc->radius) == false) {
            //lv_draw_mask_radius_param_t radius_mask_param;
            lv_draw_mask_radius_init(&radius_mask_param, coords, dsc->radius, false);
            radius_mask_id = lv_draw_mask_add(&radius_mask_param, NULL);
        }

        if(src_type == LV_IMG_SRC_SYMBOL)  lv_draw_label(&coords_tmp, clip, &label_dsc, dsc->pattern_image, NULL);
        else lv_draw_img(&coords_tmp, clip, dsc->pattern_image, &img_dsc);

        lv_draw_mask_remove_id(radius_mask_id);
    }
}`

closed 2021-04-21 by jianjunxia

kisvegabor, 2021-04-22:
Thanks for the investigation.
It"d be great to have it fixed on the `release/v7` branch. Could you send a Pull request?

embeddedt, 2021-04-22:
@kisvegabor Unless I interpreted @jianjunxia's statement incorrectly, this bug is already fixed in newer v7 versions.

kisvegabor, 2021-04-22:
Oh, great :)