lvgl/lvgl #2203: Issues with examples

mirozmrzli, 2021-04-20:
Hi all,

I got the library working with esp32 that uses arduino as component. To properly build I had to modify component.mk to include extra themes, widgets and layouts, along with modifying two c files to correct the #include path. lv_conf.h file seems to be ignored and the library wants to use menuconfig instead. The issues I have are the following:

1. light/dark theme setting is ignored, it always uses light (white background, blue buttons)
2. message box example is showing only the right part of the message box (lv_example_msgbox_1.c)
3. text box/keyboard example shows text box way too big (lv_example_keyboard_1.c). 

other examples seem ok. I am 75% confident this is due to configuration and 3. might not even be an issue, but I'm not sure how to proceed. All ideas are welcome!

Cheers,
Miro

[image]

[image]

component.mk
```
COMPONENT_SRCDIRS := .
COMPONENT_SRCDIRS += src
COMPONENT_SRCDIRS += src/core
COMPONENT_SRCDIRS += src/draw
COMPONENT_SRCDIRS += src/font
COMPONENT_SRCDIRS += src/gpu
COMPONENT_SRCDIRS += src/hal
COMPONENT_SRCDIRS += src/misc
COMPONENT_SRCDIRS += src/widgets
COMPONENT_SRCDIRS += src/extra
COMPONENT_SRCDIRS += src/extra/themes/default
COMPONENT_SRCDIRS += src/extra/themes/basic
COMPONENT_SRCDIRS += src/extra/widgets/tabview
COMPONENT_SRCDIRS += src/extra/widgets/win
COMPONENT_SRCDIRS += src/extra/widgets/msgbox
COMPONENT_SRCDIRS += src/extra/widgets/calendar
COMPONENT_SRCDIRS += src/extra/widgets/keyboard
COMPONENT_SRCDIRS += src/extra/widgets/list
COMPONENT_SRCDIRS += src/extra/widgets/spinbox
COMPONENT_SRCDIRS += src/extra/widgets/spinner
COMPONENT_SRCDIRS += src/extra/widgets/tileview
COMPONENT_SRCDIRS += src/extra/widgets/colorwheel
COMPONENT_SRCDIRS += src/extra/widgets/led
COMPONENT_SRCDIRS += src/extra/layouts/flex
COMPONENT_SRCDIRS += src/extra/layouts/grid

COMPONENT_ADD_INCLUDEDIRS := .
COMPONENT_ADD_INCLUDEDIRS += src
COMPONENT_ADD_INCLUDEDIRS += src/core
COMPONENT_ADD_INCLUDEDIRS += src/draw
COMPONENT_ADD_INCLUDEDIRS += src/font
COMPONENT_ADD_INCLUDEDIRS += src/gpu
COMPONENT_ADD_INCLUDEDIRS += src/hal
COMPONENT_ADD_INCLUDEDIRS += src/misc
COMPONENT_ADD_INCLUDEDIRS += src/widgets
```

configuration
```
#
# Logging
#
CONFIG_LV_USE_LOG=

#
# Asserts
#
CONFIG_LV_USE_ASSERT_NULL=y
CONFIG_LV_USE_ASSERT_MALLOC=y
CONFIG_LV_USE_ASSERT_STYLE=y
CONFIG_LV_USE_ASSERT_MEM_INTEGRITY=
CONFIG_LV_USE_ASSERT_OBJ=

#
# Others
#
CONFIG_LV_USE_PERF_MONITOR=y
CONFIG_LV_USE_MEM_MONITOR=
CONFIG_LV_USE_REFR_DEBUG=
CONFIG_LV_SPRINTF_CUSTOM=
CONFIG_LV_SPRINTF_USE_FLOAT=
CONFIG_LV_USE_USER_DATA=y

#
# Compiler settings
#
CONFIG_LV_BIG_ENDIAN_SYSTEM=
CONFIG_LV_USE_LARGE_COORD=
CONFIG_LV_USE_USER_DATA_FREE=
CONFIG_LV_USE_API_EXTENSION_V6=y
CONFIG_LV_USE_API_EXTENSION_V7=y

#
# Font usage
#

#
# Enable built-in fonts
#
CONFIG_LV_FONT_MONTSERRAT_8=y
CONFIG_LV_FONT_MONTSERRAT_10=y
CONFIG_LV_FONT_MONTSERRAT_12=y
CONFIG_LV_FONT_MONTSERRAT_14=y
CONFIG_LV_FONT_MONTSERRAT_16=y
CONFIG_LV_FONT_MONTSERRAT_18=y
CONFIG_LV_FONT_MONTSERRAT_20=y
CONFIG_LV_FONT_MONTSERRAT_22=y
CONFIG_LV_FONT_MONTSERRAT_24=y
CONFIG_LV_FONT_MONTSERRAT_26=y
CONFIG_LV_FONT_MONTSERRAT_28=y
CONFIG_LV_FONT_MONTSERRAT_30=y
CONFIG_LV_FONT_MONTSERRAT_32=y
CONFIG_LV_FONT_MONTSERRAT_34=y
CONFIG_LV_FONT_MONTSERRAT_36=y
CONFIG_LV_FONT_MONTSERRAT_38=y
CONFIG_LV_FONT_MONTSERRAT_40=y
CONFIG_LV_FONT_MONTSERRAT_42=y
CONFIG_LV_FONT_MONTSERRAT_44=y
CONFIG_LV_FONT_MONTSERRAT_46=y
CONFIG_LV_FONT_MONTSERRAT_48=y
CONFIG_LV_FONT_UNSCII_8=y
CONFIG_LV_FONT_UNSCII_16=y
CONFIG_LV_FONT_MONTSERRAT12SUBPX=y
CONFIG_LV_FONT_MONTSERRAT28COMPRESSED=y
CONFIG_LV_FONT_DEJAVU_16_PERSIAN_HEBREW=y
CONFIG_LV_FONT_SIMSUN_16_CJK=y
CONFIG_LV_FONT_FMT_TXT_LARGE=
CONFIG_LV_USE_FONT_COMPRESSED=
CONFIG_LV_USE_FONT_SUBPX=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_8=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_12=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_14=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_16=y
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_18=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_20=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_22=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_24=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_26=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_28=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_30=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_32=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_34=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_36=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_38=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_40=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_42=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_44=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_46=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT_48=
CONFIG_LV_FONT_DEFAULT_UNSCII_8=
CONFIG_LV_FONT_DEFAULT_UNSCII_16=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT12SUBPX=
CONFIG_LV_FONT_DEFAULT_MONTSERRAT28COMPRESSED=
CONFIG_LV_FONT_DEFAULT_DEJAVU_16_PERSIAN_HEBREW=
CONFIG_LV_FONT_DEFAULT_SIMSUN_16_CJK=

#
# Themes
#
CONFIG_LV_USE_THEME_DEFAULT=y
CONFIG_LV_THEME_DEFAULT_PALETTE_LIGHT=
CONFIG_LV_THEME_DEFAULT_GROW=y
CONFIG_LV_THEME_DEFAULT_TRANSITION_TIME=80
CONFIG_LV_USE_THEME_BASIC=

#
# Text Settings
#
CONFIG_LV_TXT_ENC_UTF8=y
CONFIG_LV_TXT_ENC_ASCII=
CONFIG_LV_TXT_BREAK_CHARS=" ,.;:-_"
CONFIG_LV_TXT_LINE_BREAK_LONG_LEN=0
CONFIG_LV_TXT_COLOR_CMD="#"
CONFIG_LV_USE_BIDI=
CONFIG_LV_USE_ARABIC_PERSIAN_CHARS=

#
# Widget usage
#
CONFIG_LV_USE_ARC=y
CONFIG_LV_USE_BAR=y
CONFIG_LV_USE_BTN=y
CONFIG_LV_USE_BTNMATRIX=y
CONFIG_LV_USE_CANVAS=y
CONFIG_LV_USE_CHECKBOX=y
CONFIG_LV_USE_CHART=y
CONFIG_LV_USE_DROPDOWN=y
CONFIG_LV_USE_IMG=y
CONFIG_LV_USE_LABEL=y
CONFIG_LV_LABEL_TEXT_SEL=y
CONFIG_LV_LABEL_LONG_TXT_HINT=y
CONFIG_LV_USE_LINE=y
CONFIG_LV_USE_METER=y
CONFIG_LV_USE_ROLLER=y
CONFIG_LV_ROLLER_INF_PAGES=7
CONFIG_LV_USE_SLIDER=y
CONFIG_LV_USE_SWITCH=y
CONFIG_LV_USE_TEXTAREA=y
CONFIG_LV_TEXTAREA_DEF_PWN_SHOW_TIME=1500
CONFIG_LV_USE_TABLE=y

#
# Extra components
#
CONFIG_LV_USE_CALENDAR=y
CONFIG_LV_CALENDAR_WEEK_STARTS_MONDAY=y
CONFIG_LV_USE_CALENDAR_HEADER_ARROW=y
CONFIG_LV_USE_CALENDAR_HEADER_DROPDOWN=y
CONFIG_LV_USE_COLORWHEEL=y
CONFIG_LV_USE_IMGBTN=y
CONFIG_LV_USE_KEYBOARD=y
CONFIG_LV_USE_LED=y
CONFIG_LV_USE_LIST=y
CONFIG_LV_USE_MSGBOX=y
CONFIG_LV_USE_SPINBOX=y
CONFIG_LV_USE_SPINNER=y
CONFIG_LV_USE_TABVIEW=y
CONFIG_LV_USE_TILEVIEW=y
CONFIG_LV_USE_WIN=y

#
# Layouts
#
CONFIG_LV_USE_FLEX=y
CONFIG_LV_USE_GRID=y
```

kisvegabor, 2021-04-22:
Hi,

Thanks for the feedback!

Could you open a Pull requires to update components.mk?

> light/dark theme setting is ignored, it always uses light (white background, blue buttons)

I've just added dark theme support. You can enable it by
```c
lv_theme_default_init(NULL, LV_COLOR_PALETTE_LIGHT_BLUE, LV_COLOR_PALETTE_RED, true, &lv_font_montserrat_10, LV_FONT_DEFAULT, &lv_font_montserrat_18);
```
`true` switches to dark mode.

>  message box example is showing only the right part of the message box (lv_example_msgbox_1.c)
> text box/keyboard example shows text box way too big (lv_example_keyboard_1.c).

I think LVGL calculates with a different display size.
In v8 the display size needs be set in the display driver. E.g.
```c
disp_drv.hor_res = 320;
disp_drv.hor_res = 240;
```

BTW, what is the resolution of the display:

mirozmrzli, 2021-04-22:
Hi, thanks for the feedback! I think the resolution is set properly. Also, other examples render correctly:

```
static const uint32_t screenWidth = 320;
static uint32_t screenHeight = 240;

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[screenWidth * 10];

	lv_disp_draw_buf_init(&draw_buf, buf, NULL, screenWidth * 10);

	/*Initialize the display*/
	lv_disp_drv_t disp_drv;
	lv_disp_drv_init(&disp_drv);

	/*Change the following line to your display resolution*/
	disp_drv.hor_res = screenWidth;
	disp_drv.ver_res = screenHeight;
	disp_drv.flush_cb = my_disp_flush;
	disp_drv.draw_buf = &draw_buf;
	lv_disp_drv_register(&disp_drv);

	/*Initialize the (dummy) input device driver*/
	lv_indev_drv_t indev_drv;
	lv_indev_drv_init(&indev_drv);
	indev_drv.type = LV_INDEV_TYPE_POINTER;
	indev_drv.read_cb = my_touchpad_read;
	lv_indev_drv_register(&indev_drv);
```

mirozmrzli, 2021-04-22:
Also, stumbled on another issue with ```lv_example_checkbox_1```. All checkboxes are rendered on top of each other:

[image]

kisvegabor, 2021-04-23:
I've found a bug in LVGL and just fixed it. It solved the issue of the checkbox example for me.

kisvegabor, 2021-05-14:
As there was no activity here for a while  I suppose the issue is solved.

If it's not working feel free the comment here and reopen this issue.

closed 2021-05-14 by kisvegabor