lvgl/lvgl #2204: lv_disp_drv_init in calling error in lv_canvas_draw_rect function

linuxdrv, 2021-04-21:
hi all,

in **lv_canvas_draw_rect** function (_lv_canvas.c_), after calling   
    **lv_memset_00(&disp, sizeof(lv_disp_t));**
disp.driver = NULL;
   then calling **lv_disp_drv_init(disp.driver);** will cause an error.
same issues in function **lv_canvas_draw_text**, **lv_canvas_draw_img**,**lv_canvas_draw_line**,**lv_canvas_draw_polygon** and **lv_canvas_draw_arc**.

Cheers,

### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

embeddedt, 2021-04-21:
Oops; that looks like an oversight from the API change in v8. Thanks for reporting!

embeddedt, 2021-04-21:
This should be fixed now.

commit 980c18fa (2021-04-21): fix(canvas) allocate a fake driver on the stack

As of v8 drivers are no longer stored inside lv_disp_t, so a driver
needs to be allocated as well.

Fixes #2204
closed 2021-04-21 by embeddedt