lvgl/lvgl #2205: (Zephyr lvgl 7.6.1) checkbox in container not receiving events

thedjnK, 2021-04-21:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version - using latest version supported by zephyr, 7.6.1
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html) - not tested in a simulator

### Describe the bug
I'm trying to create a layout with a checkbox and have it so that checkbox events can be received and used, it works when there is no container but when there is a container the checkbox appears dead - it can't be ticked and it doesn't receive callbacks.

### To Reproduce
```c
lv_obj_t * cont;
lv_obj_t * cb;

static void event_handler(lv_obj_t * obj, lv_event_t event)
{
	printf("Obj %p, Event %d, State: %s\n", obj, event, lv_checkbox_is_checked(obj) ? "Checked" : "Unchecked");
}

void setup()
{
	const struct device *display_dev;
	lv_obj_t *hello_world_label;
	lv_obj_t *count_label;

	display_dev = device_get_binding(CONFIG_LVGL_DISPLAY_DEV_NAME);

	cont = lv_cont_create(lv_scr_act(), NULL);
	lv_obj_set_auto_realign(cont, true);/*Auto realign when the size changes*/

	count_label = lv_label_create(cont /*lv_scr_act()*/, NULL);
	lv_obj_align(count_label, NULL, LV_ALIGN_IN_BOTTOM_MID, 0, 0);

	lv_task_handler();
	display_blanking_off(display_dev);

	cb = lv_checkbox_create(cont /*lv_scr_act()*/, NULL);
	lv_checkbox_set_text(cb, "test");
	lv_obj_align(cb, NULL, LV_ALIGN_CENTER, 0, 0);
	lv_obj_set_event_cb(cb, event_handler);

	while (1) {
		if ((count % 100) == 0U) {
			sprintf(count_str, "%d", count/100U);
			lv_label_set_text(count_label, count_str);
		}
		lv_task_handler();
		k_sleep(K_MSEC(10));
		++count;
	}
}
```

### Expected behavior
The checkbox to tick and untick and receive the event

thedjnK, 2021-04-21:
So it is an issue in 7.6.x, I hacked together a basic version of 7.7 and despite the touch being 180 degrees out, the checkbox inside a container is working in that version, so it's waiting for zephyr to catch up.

embeddedt, 2021-04-21:
Unfortunately, we can only support the latest version of v7 (branch `release/v7` on GitHub).

I am surprised that Zephyr is still using 7.6 - it was released in October 2020. I'll have to take a look at their GitHub repo and see if there were issues upgrading.

thedjnK, 2021-04-21:
I did try just clearing the lvgl module and fetching it from here on master and it threw out of a lot of build errors, switched to 7.7 and it was able to build minus a single missing include. They've got a 7.8 work in progress patch from December but hasn't had any work since December 5th

closed 2021-04-21 by thedjnK

embeddedt, 2021-04-21:
They may just be holding out for v8, as otherwise they will have to rewrite their Kconfig twice in the space of a few months.