lvgl/lvgl #2214: Issues about the minimal platform dependency

MouriNaruto, 2021-04-24:
In the recent days, I tried to port the LVGL to UEFI platform with the SDK split from EDK II by myself. (I don't want to use EDK II because it's too big and I don't want to use the build system based on INI files. I also don't want to use gnu-efi because it doesn't contain all protocols mentioned in the UEFI specifications.)

I need to implement the minimum C runtime library by myself, and finally I have ported LVGL to UEFI platform with the SDK split from EDK II by myself.

[image]

You can read my work in https://github.com/ProjectMile/Mile.Uefi.

I found LVGL still need to include stdio.h, stdlib.h and string.h even it is an empty file, and LVGL still need memcmp, memcpy, memset, strcat, strcmp, strcpy, strlen and strncpy.

It looks strange because we all know LVGL implement the basic C runtime functions like sprintf, memcpy and memset etc. 

Note: I think it's OK to rely on stdarg.h, stddef.h, stdbool.h and stdint.h because it's platfrom specific.

kisvegabor, 2021-04-24:
Hi,

First of all, amazing project! It's so good to see that LVGL can be used in such special cases as a UEFI.
It'd be great to have a blog post about the basics of LVGL-UEFI. Nothing fancy just the steps to get started. Would you be interested in adding a [blog post](https://github.com/lvgl/blog) like this?

I've pushed some fixed to remove the use of `<stdlib.h>` and `<stdio.h>` and `<string.h>`.

MouriNaruto, 2021-04-24:
> Would you be interested in adding a blog post like this?

Yes, It's my pleasure. I will push an article to the repo you mentioned. 

> I've pushed some fixed to remove the use of <stdlib.h> and <stdio.h> and <string.h>.

Looks good. I also think it may be good for LVGL to move some reimplementation of the C runtime functions to a separate folder.

kisvegabor, 2021-04-24:
> Yes, It's my pleasure. I will push an article to the repo you mentioned.

Awesome! :slightly_smiling_face: 

> Looks good. I also think it may be good for LVGL to move some reimplementation of the C runtime functions to a separate folder.

Only a very small assorted set of functions is implemented. (`lv_mem_alloc`, `lv_realloc`, `lv_mem_free`, `lv_memcpy`, `lv_memset`, `lv_snprintf`).  I think it'd be misleading to suggest that it's C runtime reimplementation.

MouriNaruto, 2021-04-24:
> I think it'd be misleading to suggest that it's C runtime reimplementation.

I am agreed with you. 

P.S. I will continue to implement the UEFI demo. For example, I need to add the mouse and keyboard support.

kisvegabor, 2021-04-24:
> P.S. I will continue to implement the UEFI demo. For example, I need to add the mouse and keyboard support.

Great! Feel free to open issues if you have any questions or fine problematic parts in LVGL.

As the original issue is solved I close this.

closed 2021-04-24 by kisvegabor