lvgl/lvgl #2216: support rich text in LVGL

guoweilkd, 2021-04-25:
### support rich text in LVGL

problem link:<https://forum.lvgl.io/t/text-area-rich-text-or-simple-html-tags/4031>

### my solution

program base on v7. 

Reference to HTML, use `span` as the string descriptor. 
```c
typedef struct {
    char * txt;          /* save string */
    lv_style_t style;  /* save string style */
} lv_span_t;
```
I write a widget named `spangroup` to link and draw all `span`. maybe can instead `label` widget .
the spangroup widget of ext struct like this:
```c
typedef struct {
    ....
    lv_ll_t  child_ll; /* Link all span */
    ....
} lv_span_ext_t;
```

the following is draw `spangroup` function 
```c
static LV_ATTRIBUTE_FAST_MEM void lv_draw_span(lv_obj_t * spans, const lv_area_t * coords, const lv_area_t * mask)
{
    /* return if not draw area */
    lv_area_t clipped_area;
    bool clip_ok = _lv_area_intersect(&clipped_area, coords, mask);
    if(!clip_ok) return;

    lv_span_ext_t * ext = lv_obj_get_ext_attr(spans);
    if (_lv_ll_get_head(&ext->child_ll) == NULL) {
        return;
    }
    /* calculate and get max width */
    lv_coord_t max_width = lv_area_get_width(coords);
    lv_coord_t max_w = max_width;

    lv_point_t txt_pos;
    txt_pos.x = coords->x1;
    txt_pos.y = coords->y1;
    lv_coord_t max_line_h = 0;
    lv_coord_t use_width = 0;
    lv_txt_flag_t txt_flag = LV_TXT_FLAG_NONE;
    lv_coord_t line_space = lv_obj_get_style_text_line_space(spans, LV_SPAN_PART_MAIN);;
    txt_pos.x += ext->start_ofs.x;
    txt_pos.y += ext->start_ofs.y;

    lv_snippet_t snippet;
    lv_span_t *cur_span = _lv_ll_get_head(&ext->child_ll);
    char * cur_txt = cur_span->txt;
    uint32_t cur_txt_ofs = 0;

    while (cur_span) {
        /* find a line text and push the relevant span onto the stack  */
        while(1) {
            /* switch to the next span when current is end */
            if (cur_txt[cur_txt_ofs] == '\0') {
                cur_span = _lv_ll_get_next(&ext->child_ll, cur_span);
                if (cur_span == NULL) break;
                cur_txt = cur_span->txt;
                cur_txt_ofs = 0;
            }

            if (cur_txt_ofs == 0) {
                snippet.span = cur_span;
                snippet.font = lv_span_get_style_text_font(spans, cur_span);
                snippet.letter_space = lv_span_get_style_text_letter_space(spans, cur_span);
                snippet.line_h = lv_font_get_line_height(snippet.font) + line_space;
            }

            /* get current span text line info */
            uint32_t next_ofs = cur_txt_ofs;
            bool isfill = lv_txt_get_snippet(&cur_txt[cur_txt_ofs], snippet.font, snippet.letter_space,
                    max_w, txt_flag, &use_width, &next_ofs);
            
            /* break word deal width */
            if (isfill && next_ofs > 0 && lv_get_snippet_cnt() > 0) {
                uint32_t letter = (uint32_t)cur_txt[cur_txt_ofs + next_ofs - 1];
                if (!(letter == '\0' || letter == '\n' || letter == '\r' || is_break_char(letter))) {
                    letter = (uint32_t)cur_txt[cur_txt_ofs + next_ofs];
                    if (!(letter == '\0' || letter == '\n'  || letter == '\r' || is_break_char(letter))) {
                        break;
                    }
                }
                
            }

            snippet.txt = &cur_txt[cur_txt_ofs];
            snippet.bytes = next_ofs;
            snippet.txt_w = use_width;
            cur_txt_ofs += next_ofs;

            printf("cur_span:%p, isfill:%d, cur_txt_ofs:%d, max_w:%d\n", cur_span, isfill, cur_txt_ofs, max_w);

            if (max_line_h < snippet.line_h) {
                max_line_h = snippet.line_h;
            }
            lv_snippet_push(&snippet);
            if (isfill) {
                break;
            }
            max_w -= use_width;
        }

        printf("max_width:%d, max_line_h:%d, txt_pos(%d,%d)\n", max_width, max_line_h, txt_pos.x, txt_pos.y);

        /* break if stack is empty */
        uint16_t item_cnt = lv_get_snippet_cnt();
        if (item_cnt == 0) {
            break;
        }

        /*Go the first visible line*/
        if (txt_pos.y + max_line_h < mask->y1) {
            printf("---continue(txt_pos.y < mask->y1)\n");
            goto Next_line_init;
        }

        /* align deal with:todo */
        lv_coord_t txts_w = 0;
        for (int i = 0; i < item_cnt; i++) {
            lv_snippet_t *pinfo = lv_get_snippet(i);
            txts_w += pinfo->txt_w;
        }
        txt_pos.x += 0;

        printf("item_cnt:%d ", item_cnt);

        /* draw line letters */
        const char * bidi_txt;
        for (int i = 0; i < item_cnt; i++) {
            lv_snippet_t *pinfo = lv_get_snippet(i);
            printf("span:%p, bytes:%d, txt_w:%d,line_h:%d\n", pinfo->span, pinfo->bytes, pinfo->txt_w, pinfo->line_h);

            /* bidi deal with:todo */
            bidi_txt = pinfo->txt;
            lv_point_t pos;
            pos.y = txt_pos.y + max_line_h - pinfo->line_h;
            lv_color_t letter_color = lv_span_get_style_text_color(spans, pinfo->span);
            lv_opa_t letter_opa = lv_span_get_style_text_opa(spans, pinfo->span);
            lv_blend_mode_t blend_mode = lv_span_get_style_text_blend_mode(spans, pinfo->span);
            uint32_t j = 0;
            while (j < pinfo->bytes) {
                pos.x = txt_pos.x;
                uint32_t letter      = _lv_txt_encoded_next(bidi_txt, &j);
                uint32_t letter_next = _lv_txt_encoded_next(&bidi_txt[j], NULL);
                int32_t letter_w = lv_font_get_glyph_width(pinfo->font, letter, letter_next);
                _lv_draw_letter(&pos, &clipped_area, pinfo->font, letter, letter_color, letter_opa, blend_mode);
                if (letter_w > 0) {
                    txt_pos.x = txt_pos.x + letter_w + pinfo->letter_space;
                }
            }
        }

Next_line_init:
        lv_snippet_clear();
        /* next line init */
        txt_pos.x = coords->x1;
        txt_pos.y += max_line_h;
        printf("txt_pos(%d,%d), mask->y2:%d\n", txt_pos.x, txt_pos.y, mask->y2);
        if(txt_pos.y > mask->y2) {
            printf("---end(txt_pos.y > mask->y2)\n");
            return;
        }

        max_w = max_width;
        max_line_h = 0;

        printf("---end(max_line_h:%d)\n", max_line_h);
    }
}
```

the following is functions  by lv_draw_span() invoked
```c

static bool lv_txt_get_snippet(const char * txt, const lv_font_t * font,
    lv_coord_t letter_space, lv_coord_t max_width, lv_txt_flag_t flag, 
    lv_coord_t * use_width, uint32_t * end_ofs)
{
    uint32_t ofs = _lv_txt_get_next_line(txt, font, letter_space, max_width, flag);
    lv_coord_t width = _lv_txt_get_width(txt, ofs, font, letter_space, flag);

    *end_ofs = ofs;
    *use_width = width;

    printf("snippet:%s, max_width(%d), ,width:%d, ofs:%d\n",txt, max_width, width, ofs);

    if (txt[ofs] == '\0' && width < max_width) {
        return false;
    }
    else {
        return true;
    }
}

/* The information stack used to hold all the Spans that make up a line of text  */
#define LV_SPAN_SNIPPET_STACK_SIZE 64
struct _snippet_stack{
    lv_snippet_t stack[LV_SPAN_SNIPPET_STACK_SIZE];
    uint16_t index;
};
struct _snippet_stack snippet_stack;

static void lv_snippet_push(lv_snippet_t *item)
{
    memcpy(&snippet_stack.stack[snippet_stack.index], item, sizeof(lv_snippet_t));
    snippet_stack.index++;
}

/* get style functions */
const lv_font_t *lv_span_get_style_text_font(lv_obj_t * par, lv_span_t *span)
{
    const lv_font_t *font = NULL;
    int res = _lv_style_get_ptr(&span->style, LV_STYLE_TEXT_FONT, &font);
    if (res < 0) {
        font = lv_obj_get_style_text_font(par, LV_OBJ_PART_MAIN);
    }
    return font;
}
```

The result like this:

[image]

embeddedt, 2021-04-25:
I like it! It probably needs some tweaks to work with v8's layout system, but this could open up the possibility of proper inline layouts (useful for emojis, etc.).

kisvegabor, 2021-04-26:
I also like it!

Do you have time to port it to v8? See for example [lv_led](https://github.com/lvgl/lvgl/blob/master/src/extra/widgets/led/lv_led.c) to see how to create a new widget in v8.

guoweilkd, 2021-04-27:
The span widget based on the V7 is almost  already , I'm gonna PR this first ? and then i will port it to v8.

kisvegabor, 2021-04-27:
The problem is that there are no more minor releases for v7 (that is no new features are added).

I suggest hosting the v7 version in your LVGL v7 fork and write a post about it in the forum. I'm sure a lot of people will like it but we can't merge it to v7 :slightly_frowning_face:

closed 2021-05-10 by guoweilkd