lvgl/lvgl #2217: I found a bug that exists in all versions：This bug will cause Segmentation fault

jianjunxia, 2021-04-25:
Here is more information about this bug：
in lv_draw_label function，There is something wrong with this line of code： 
`    if(txt[0] == '\0')  return;`
problem causes：
if txt is null, then it will become:  ` if (null[0] == '\0')`, this will cause Segmentation fault
I suggest to modify it like this：   `if(txt != null ) { if (txt[0] == '\0') return;} else{return}`       
this way can avoid segmentation fault
I will initiate a merge request later

guoweilkd, 2021-04-25:
Why does this happen? Precautions have been taken in  lv_label_create() and lv_label_set_text() functions.

```c
lv_obj_t * lv_label_create(lv_obj_t * par, const lv_obj_t * copy)
{
    ...
    lv_label_set_text(new_label, "Text");
    ...
}
```

```c
void lv_label_set_text(lv_obj_t * label, const char * text)
{
    ....
    /*If text is NULL then refresh */
    if(text == NULL) {
        lv_label_refr_text(label);
        return;
    }
    ....
```

jianjunxia, 2021-04-25:
> ```c
> lv_label_set_text
> ```
in lv_obj_t * lv_label_create(lv_obj_t * par, const lv_obj_t * copy) function， 
when `copy != null && par->ext_arrt->text = null,`  then will result in a segment fault

jianjunxia, 2021-04-25:
> Why does this happen? Precautions have been taken in lv_label_create() and lv_label_set_text() functions.
>
> ```c
> lv_obj_t * lv_label_create(lv_obj_t * par, const lv_obj_t * copy)
> {
>     ... [...]

jianjunxia, 2021-04-25:
> Why does this happen? Precautions have been taken in lv_label_create() and lv_label_set_text() functions.
>
> ```c
> lv_obj_t * lv_label_create(lv_obj_t * par, const lv_obj_t * copy)
> {
>     ... [...]

in  void lv_label_set_text(lv_obj_t * label, const char * text) function 
` text != null && label->>ext_arrt->text = null`  this case may cause sementtation fault

kisvegabor, 2021-04-26:
> in void lv_label_set_text(lv_obj_t * label, const char * text) function
text != null && label->>ext_arrt->text = null this case may cause sementtation fault

Where is this line in `lv_label.c`? Please link it from GtiHub.

jianjunxia, 2021-04-26:
> > in void lv_label_set_text(lv_obj_t * label, const char * text) function
> > text != null && label->>ext_arrt->text = null this case may cause sementtation fault
> 
> Where is this line in `lv_label.c`? Please link it from GtiHub.

yes,  https://github.com/lvgl/lvgl/blob/release/v7/src/lv_widgets/lv_label.c
in lv_label_create() function, in line139
in lv_label_set_text() function, in line 214

kisvegabor, 2021-04-26:
https://github.com/lvgl/lvgl/pull/2218 is merged so I close this issue.

Thank you for the report and PR.

closed 2021-04-26 by kisvegabor