lvgl/lvgl #2225: Scroll tabview page fully covered by objects

pjanco, 2021-04-27:
### Describe the bug
Hi,
I have tabview page fully covered by buttons. I noticed that when I try to scroll the page by slide gesture, it is working really bad if the gesture is made over a buttons. Touchscreen is unresponsive and I have to make slide gesture multiple times to do slide 
action. It have the same bad behavior on other objects too, for example on containers.

Only clean tab page without objects or simple text labels without any parent container are working fine and is possible to slide over them.

### To Reproduce
```c
static void add_tabview_btn1(lv_obj_t * parent, const char *text)
{
    static uint16_t position_y = 24;

    lv_obj_t * tmp;

    tmp = lv_btn_create(parent, NULL);
    lv_obj_set_size(tmp, 400, 60);
    lv_obj_align(tmp, NULL, LV_ALIGN_IN_TOP_MID, 0, position_y);
    lv_obj_set_style_local_radius(tmp, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, 10);
    lv_obj_set_style_local_text_font(tmp, LV_LABEL_PART_MAIN, LV_STATE_DEFAULT, LV_THEME_DEFAULT_FONT_SUBTITLE);
    tmp = lv_label_create(tmp, NULL);
    lv_label_set_text_static(tmp, text);

    position_y += 60 + 4;
}

void lv_ex_tabview(void)
{
    lv_obj_t *tabview;
    tabview = lv_tabview_create(lv_scr_act(), NULL);

    lv_obj_t *tab1 = lv_tabview_add_tab(tabview, "Tab 1");
    lv_obj_t *tab2 = lv_tabview_add_tab(tabview, "Tab 2");

    add_tabview_btn1(tab1, "Action 1");
    add_tabview_btn1(tab1, "Action 2");
    add_tabview_btn1(tab1, "Action 3");
    add_tabview_btn1(tab1, "Action 4");
    add_tabview_btn1(tab1, "Action 5");
    add_tabview_btn1(tab1, "Action 6");
    add_tabview_btn1(tab1, "Action 7");
    add_tabview_btn1(tab1, "Action 8");
    add_tabview_btn1(tab1, "Action 9");
    add_tabview_btn1(tab1, "Action 1");
    add_tabview_btn1(tab1, "Action 2");
    add_tabview_btn1(tab1, "Action 3");
    add_tabview_btn1(tab1, "Action 4");
    add_tabview_btn1(tab1, "Action 5");
    add_tabview_btn1(tab1, "Action 6");
    add_tabview_btn1(tab1, "Action 7");
    add_tabview_btn1(tab1, "Action 8");
    add_tabview_btn1(tab1, "Action 9");
}
```

### Expected behavior
I expect behavior should be similar like in the slider object. There is also slide gesture over active buttons but it is working fine.

### Screenshots or video

https://user-images.githubusercontent.com/34516806/116206782-71441e80-a73f-11eb-9010-1eeec53be8d5.mp4

embeddedt, 2021-04-27:
You can try gluing the button to the tabview with `lv_page_glue_obj(btn, true)`. Part of this is a limitation of the scroll system in v7; scrolling can only propagate through page-like objects.

I will try the same demo in v8 just to confirm, but I think this will be less of an issue there.

kisvegabor, 2021-04-27:
I've just tried it out in v8. It works quite naturally there.

[image]

```c

static void add_tabview_btn1(lv_obj_t * parent, const char *text)
{
    static uint16_t position_y = 24;

    lv_obj_t * tmp;

    tmp = lv_btn_create(parent);
    lv_obj_set_size(tmp, 400, 60);
    lv_obj_align(tmp, LV_ALIGN_TOP_MID, 0, position_y);
    lv_obj_set_style_radius(tmp, 10, 0);
    tmp = lv_label_create(tmp);
    lv_label_set_text_static(tmp, text);
    lv_obj_center(tmp);

    position_y += 60 + 4;
}

void lv_ex_tabview(void)
{
    lv_obj_t *tabview;
    tabview = lv_tabview_create(lv_scr_act(), LV_DIR_TOP, 60);

    lv_obj_t *tab1 = lv_tabview_add_tab(tabview, "Tab 1");
    lv_obj_t *tab2 = lv_tabview_add_tab(tabview, "Tab 2");

    add_tabview_btn1(tab1, "Action 1");
    add_tabview_btn1(tab1, "Action 2");
    add_tabview_btn1(tab1, "Action 3");
    add_tabview_btn1(tab1, "Action 4");
    add_tabview_btn1(tab1, "Action 5");
    add_tabview_btn1(tab1, "Action 6");
    add_tabview_btn1(tab1, "Action 7");
    add_tabview_btn1(tab1, "Action 8");
    add_tabview_btn1(tab1, "Action 9");
    add_tabview_btn1(tab1, "Action 1");
    add_tabview_btn1(tab1, "Action 2");
    add_tabview_btn1(tab1, "Action 3");
    add_tabview_btn1(tab1, "Action 4");
    add_tabview_btn1(tab1, "Action 5");
    add_tabview_btn1(tab1, "Action 6");
    add_tabview_btn1(tab1, "Action 7");
    add_tabview_btn1(tab1, "Action 8");
    add_tabview_btn1(tab1, "Action 9");
}

```

pjanco, 2021-04-27:
> I've just tried it out in v8. It works quite naturally there.
>
> [image]
>
> ```c
> static void add_tabview_btn1(lv_obj_t * parent, const char *text)
> {
>     static uint16_t position_y = 24;
>
>     lv_obj_t * [...]

That's great, thank you. Before I asked, I should have tried latest dev branch myself...

closed 2021-04-27 by pjanco

kisvegabor, 2021-04-27:
Note that the current version of v7 is available from the master branch.