lvgl/lvgl #2226: Recent changes in V8 break simulator

uraich, 2021-04-27:
### Perform all steps below and tick them with [x]
- [x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [x ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
change of parameters in
lv_img_create
lv_theme_default_init
break the simulator
Same is true for the change LV_COLOR_PALETTE_LIGHT_BLUE -> LV_PALETTE_LIGHT_BLUE 

### To Reproduce
Run the simulator with the newest V8 incarnation

### Screenshots or video

kisvegabor, 2021-04-28:
Probably I missed that conversation but v8 is already ported to the online MixroPython simulator? On https://sim.lvgl.io I see only v7 and v6.

uraich, 2021-04-28:
Sorry, I expressed myself badly. I meant lv_sim_eclipse_sdl.

kisvegabor, 2021-04-28:
I've jsut upated lv_sim_eclipse_sdl.

uraich, 2021-04-28:
Thanks! Working fine again.
In order to translate the example programs from C to Python it is good to see the original C code in action.

kisvegabor, 2021-04-28:
Great! :)

closed 2021-04-28 by kisvegabor