lvgl/lvgl #2228: updated lv_label.c  moves label to wrong position

ataweg, 2021-04-28:
with my application code from yesterday and the latest lvgl code from today some labels were moved or removed (see image).
When I remove such lines from my code it's ok again.
   _lv_obj_set_height( temperature_unit, fontText_height );
   lv_obj_set_width( temperature_unit, text_width );_
I expect the text to be truncated at best, but not to disappear or be moved.
Something still seems to be wrong with lv_label.c.

[image]

kisvegabor, 2021-04-28:
What is the value of `text_width` and how do you calculate it?

ataweg, 2021-04-28:
I am using these styles:
```
void WeatherStationStyles_setup( void )
{
   // style_screen

   // style_wsView
   lv_style_init( &style_wsView );
   lv_style_set_border_width( &style_wsView, 0 );
   lv_style_set_radius( &style_wsView, 0 );
   lv_style_set_outline_pad( &style_wsView, 8 );
   lv_style_set_pad_all( &style_wsView, 0 );
   lv_style_set_bg_opa( &style_wsView, LV_OPA_TRANSP );

   // style_wsValue
   lv_style_init( &style_wsValue );
   lv_style_set_text_font( &style_wsValue, wsFont72 );
   lv_style_set_text_color( &style_wsValue, lv_palette_main( LV_PALETTE_YELLOW ) );

   // style_wsValueSmall
   lv_style_init( &style_wsValueSmall );
   lv_style_set_text_font( &style_wsValueSmall, wsFont36 );
   lv_style_set_text_color( &style_wsValueSmall, lv_palette_main( LV_PALETTE_YELLOW ) );

   // style_wsText
   lv_style_init( &style_wsText );
   lv_style_set_text_font( &style_wsText, wsFont22 );
   lv_style_set_text_color( &style_wsText, lv_palette_main( LV_PALETTE_YELLOW ) );

   // style_wsIcon
   lv_style_init( &style_wsIcon );
}
```
and with this I create my panel

```
lv_obj_t * TemperaturePanel_create( lv_obj_t * parent )
{
   lv_obj_t * temperature_panel = lv_obj_create( parent );
   lv_obj_add_style( temperature_panel,  &style_wsView, 0 );

   lv_obj_set_height( temperature_panel, LV_SIZE_CONTENT );
   lv_obj_set_width( temperature_panel, LV_SIZE_CONTENT );

   lv_obj_t * temperature_value       = lv_label_create( temperature_panel );   // child index = 0
   lv_obj_t * temperature_tenth_value = lv_label_create( temperature_panel );   // child index = 1
   lv_obj_t * temperature_unit        = lv_label_create( temperature_panel );   // child index = 2

   lv_obj_add_style( temperature_value, &style_wsValue, 0 );
   lv_obj_add_style( temperature_tenth_value, &style_wsValueSmall, 0 );
   lv_obj_add_style( temperature_unit, &style_wsText, 0 );

   lv_font_t *wsFontSmall;
   lv_font_t *wsFontLarge;
   lv_font_t *wsFontText;

   lv_style_get_prop( &style_wsValueSmall, LV_STYLE_TEXT_FONT, (lv_style_value_t * )&wsFontSmall );
   lv_style_get_prop( &style_wsValue, LV_STYLE_TEXT_FONT, (lv_style_value_t * )&wsFontLarge );
   lv_style_get_prop( &style_wsText, LV_STYLE_TEXT_FONT, (lv_style_value_t * )&wsFontText );

   // format of temperature "-12.03"
   lv_coord_t fontLarge_height = lv_font_get_line_height( wsFontLarge );
   lv_coord_t fontSmall_height = lv_font_get_line_height( wsFontSmall );
   lv_coord_t fontText_height = lv_font_get_line_height( wsFontText );

   lv_coord_t textLarge_width = _lv_txt_get_width( "-88", 3, wsFontLarge, 0, 0 );
   lv_coord_t textSmall_width = _lv_txt_get_width( ".88", 3, wsFontSmall, 0, 0 );
   lv_coord_t text_width = _lv_txt_get_width( " °C", 3, wsFontText, 0, 0 );

   lv_obj_set_height( temperature_value, fontLarge_height );
   lv_obj_set_width( temperature_value, textLarge_width );
   lv_obj_set_pos( temperature_value, 0, 0 );

   lv_obj_set_height( temperature_tenth_value, fontSmall_height );
   lv_obj_set_width( temperature_tenth_value, textSmall_width );
   lv_obj_set_pos( temperature_tenth_value, textLarge_width, fontLarge_height- fontSmall_height );

   lv_obj_set_height( temperature_unit, fontText_height );
   lv_obj_set_width( temperature_unit, text_width );
   lv_obj_set_pos( temperature_unit, textLarge_width, 0 );

   lv_label_set_text( temperature_value, "---" );
   lv_label_set_text( temperature_tenth_value, ".--" );
   lv_label_set_text( temperature_unit, " \u00B0""C" );

   lv_label_t * label = ( lv_label_t * )temperature_unit;
   ESP_LOGI( TAG, "Label text: %s", label->text );

   return temperature_panel;
}

```

kisvegabor, 2021-04-28:
Thanks! I've debugged and there are few things to highlight:
- You don't have to set the label's size manually. LVGL sets it automatically to the text size. Therefore in the "old" version, your size settings were simply overwritten by LVGL. But I've added an update to the label that lets you set the size and LVGL won't overwrite it but wrap/scroll/dot/clip the too long text. By default, the label's size is `LV_SIZE_CONTENT` to allow the label to set its size automatically.
-  In `lv_coord_t text_width = _lv_txt_get_width( " °C", 3, wsFontText, 0, 0 );`  instead of `3` you should write `4` because `°` is two characters long
- But once your write `3` you might expect to see at least`" °"`. But LVGL detected that the entire text won't fit in the given with and applied the default "long mode" which is "wrap". It saw that the first character is a space and a good candidate to wrap the text here. So moved `"°C"` to the next line but it was also too wide so LVGL broke it after `°` and finally you got a 3 line text.

There are good use cases when you need to set the label size manually but usually, only the width to make LVGL wrap/scroll/etc the long texts.

ataweg, 2021-04-28:
Thank you very much for your help. 
A note about " °C". The length of this string depends on how the editor saves the file. As a plain 8-bit ascii file its length is 3, as utf-8 or saved in some other format it is 4 or more.
What I have learned from this is ascii characters larger than 127 are to be avoided even if they are easier to read in the source code. 
From the above snipple I now have replaced
`lv_coord_t text_width =  _lv_txt_get_width( " °C", 3, wsFontText, 0, 0 ); `
with
`lv_coord_t text_width = _lv_txt_get_width( " \u00B0""C", strlen( " \u00B0""C" ), wsFontText, 0, 0 );`

embeddedt, 2021-04-28:
@kisvegabor What do you think about adding an `LV_TXT_CALCULATE_LEN` constant (`0xFFFFFFFF`) so that a duplicate `strlen` call is not necessary? It would also solve the issue of differing character formats, although UTF-8 is the only one LVGL understands properly.

kisvegabor, 2021-04-29:
@ataweg 
Other encodings might contain `\0` i in the middle of the text so `strlen` won't work on them.  E.g.
```
"hello" in UTF32
U+00000068U+00000065U+0000006cU+0000006cU+0000006f
```

@embeddedt 
`_lv_txt_get_width` is not intended to be used by the users so I think it's not necessary.

ataweg, 2021-04-29:
Knowing how big objects will be before they are created is quite helpful when arranging objects. Especially with text objects, for example, this can be used to determine the size of the font to be used. Thus the very useful function `_lv_txt_get_width() `should rather be made available to the user (without leading underscore).

embeddedt, 2021-04-29:
I would agree with @ataweg on this; this type of function is usually available for public use in other GUI libraries.

kisvegabor, 2021-04-29:
My only concern is that `_lv_txt_get_width` won't always give the result you expect. For example, it doesn't handle `\n` and line-wrapping.

`lv_txt_get_size()` does all this but you need to provide a `max_width` parameter that will be used for line-wraping.

embeddedt, 2021-04-29:
I wasn't aware of `lv_txt_get_size`. That should work fine because one can always provide `LV_COORD_MAX` as the maximum width.

ataweg, 2021-04-29:
Yes,` lv_txt_get_size( ) `can do that. But this is an elaborate function just to calculate the space needed by a simple line. Don't shoot with cannons at sparrows.

kisvegabor, 2021-05-04:
There is only 1 extra parameter in `lv_txt_get_size` (the `max_width`) and you get all the text wrapping features. I think it's a good deal. 

Anyway, I'm also ok with removing  the `_` from `_lv_txt_get_width` to make more user-friendly.

ataweg, 2021-05-04:
I agree with you.

kisvegabor, 2021-05-04:
Done in 79d9f29

closed 2021-05-04 by kisvegabor