lvgl/lvgl #2229: lv_example_get_started_2 not working

uraich, 2021-04-29:
### Perform all steps below and tick them with [x]
- [ x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [ x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
 lv_obj_remove_style_all breaks the program

### To Reproduce
run lv_example_get_started_2 in  lv_sim_eclipse_sdl

### Expected behavior
Get 2 buttons. This works if  lv_obj_remove_style_all is commented

### Screenshots or video
with  lv_obj_remove_style_all
[image]
without  lv_obj_remove_style_all
[image]

kisvegabor, 2021-04-29:
Thanks, fixed.

See the added comment for the explanation of the issue.

closed 2021-04-29 by kisvegabor

uraich, 2021-04-29:
Ok, understood. Thanks! The Python version works correctly as well now.

kisvegabor, 2021-04-29:
Greae! 
It's awesome that you create the PYthon examples! :heart:

commit 45a50f17 (2021-04-29): fix(example) correct the elements size and position
Fixes #2229