lvgl/lvgl #2230: I cannot print images via SD Card with ESP32

Miguel0101, 2021-04-29:
When I use `lv_img_set_src(img, "/sd/image.png");` I received a "No data" error.

embeddedt, 2021-04-29:
Which LVGL port are you using? Usually you also need to specify the drive letter in front of the path (e.g. `S:/sd/image.png`, but the drive letter depends on your specific platform).

Miguel0101, 2021-04-29:
There, but it shows nothing.

Miguel0101, 2021-04-29:
I tried to use lv_fs_if but to no avail.

embeddedt, 2021-04-29:
There's still not enough information to help. Which project are you using (`lv_port_esp32`, something custom based on ESP-IDF, etc.)? Does it provide any info about what drive letter to use? Can you open the file with its API (bypassing LVGL)? Please provide more details.

Miguel0101, 2021-04-29:
I don't use lv_port_esp32

Miguel0101, 2021-04-29:
I use lvgl / lvgl 7.11.0

Miguel0101, 2021-04-29:
> I don't use lv_port_esp32

Is that the problem?

embeddedt, 2021-04-29:
I don't know. :slightly_smiling_face: Without knowing more information there is really not much I can help with, as the filesystem implementation is project-specific. LVGL defines the API but the actual behavior is almost entirely decided by whatever glue code you are using.

Miguel0101, 2021-04-29:
I will try to use the port `lv_port esp32`

Miguel0101, 2021-04-29:
I used a 200x150x16 image.

Miguel0101, 2021-04-29:
RAM: 520 KB.
Flash: 4MB with Huge APP.

Library: lv_lib_png.

embeddedt, 2021-04-29:
Can you enable logging and post the resulting log?

Miguel0101, 2021-04-29:
> Can you enable logging and post the resulting log?

Yes, but I get "error 83: memory allocation failed"

Miguel0101, 2021-04-29:
I don't believe it's memory.

embeddedt, 2021-04-29:
I mean LVGL's logging (`LV_USE_LOG` in `lv_conf.h`).

Miguel0101, 2021-04-29:
```
/*1: Enable the log module*/
#define LV_USE_LOG      1
#if LV_USE_LOG
/* How important log should be added:
 * LV_LOG_LEVEL_TRACE       A lot of logs to give detailed information
 * LV_LOG_LEVEL_INFO        Log important events
 * LV_LOG_LEVEL_WARN        Log if something unwanted happened but didn't cause a problem
 * LV_LOG_LEVEL_ERROR       Only critical issue, when the system may fail
 * LV_LOG_LEVEL_NONE        Do not log anything
 */
#  define LV_LOG_LEVEL    LV_LOG_LEVEL_ERROR

/* 1: Print the log with 'printf';
 * 0: user need to register a callback with `lv_log_register_print_cb`*/
#  define LV_LOG_PRINTF   1
#endif  /*LV_USE_LOG*/
```

Miguel0101, 2021-04-29:
I'll compile

Miguel0101, 2021-04-29:
```
error 83: memory allocation failed
WARNING: File: C:\Users\migue\OneDrive\Documentos\Arduino\libraries\lvgl\src\lv_draw\lv_img_cache.c
#128: _lv_img_cache_open: Image draw cannot open the image resource
WARNING: File: C:\Users\migue\OneDrive\Documentos\Arduino\libraries\lvgl\src\lv_draw\lv_draw_img.c
#88: lv_draw_img: Image draw error
```

Miguel0101, 2021-04-29:
I received this LOG

Miguel0101, 2021-04-29:
Sorry for the delay.

Miguel0101, 2021-04-29:
I still think I'm missing something, but I don't know which one.

Miguel0101, 2021-04-29:
> ```
> error 83: memory allocation failed
> WARNING: File: C:\Users\migue\OneDrive\Documentos\Arduino\libraries\lvgl\src\lv_draw\lv_img_cache.c
> #128: _lv_img_cache_open: Image draw cannot open the image [...]

I think the stack is overflowing.

Miguel0101, 2021-04-29:
How many kilobytes does the `lv_lib_png` library use?

embeddedt, 2021-04-29:
Are you asking about stack or heap?

- Stack - typically 4-8KB is a good rule of thumb for most LVGL applications. You can always optimize later if you see that a portion is never touched.
- Heap - unfortunately lodepng needs at least as much RAM as the size of the PNG you're decoding. In your case that's 60K plus the overhead for decoding and LVGL, so perhaps try with a 96KB heap.

Miguel0101, 2021-04-29:
> Are you asking about stack or heap?
>
> * Stack - typically 4-8KB is a good rule of thumb for most LVGL applications. You can always optimize later if you see that a portion is never touched.
> * Heap - [...]

Okay, thanks.

Miguel0101, 2021-04-29:
How to change the heap?

embeddedt, 2021-04-29:
That will again depend on your target platform and configuration. If you are using LVGL's heap (`LV_MEM_CUSTOM == 0`), it can be adjusted by changing `LV_MEM_SIZE` in `lv_conf.h`.

Miguel0101, 2021-04-29:
I used (`LV_MEM_CUSTOM == 1`)

embeddedt, 2021-04-29:
You will have to look it up for your project then - `LV_MEM_CUSTOM == 1` means that LVGL is using your project's heap implementation, not its own.

Miguel0101, 2021-04-29:
> You will have to look it up for your project then - `LV_MEM_CUSTOM == 1` means that LVGL is using your project's heap implementation, not its own.

I will change to (LV_MEM_CUSTOM = 0)

Miguel0101, 2021-04-29:
`#  define LV_MEM_SIZE    (32U * 1024U)`

Miguel0101, 2021-04-29:
> `# define LV_MEM_SIZE (32U * 1024U)`

Do I need to put more RAM?

embeddedt, 2021-04-29:
The current heap size you have in there is 32K.

Miguel0101, 2021-04-29:
> The current heap size you have in there is 32K. That won't be big enough.

Ok, I put # define LV_MEM_SIZE (64U * 1024U)

Miguel0101, 2021-04-29:
Do I need to put more RAM?

embeddedt, 2021-04-29:
I recommended trying 96K in my earlier post.

Miguel0101, 2021-04-29:
Ok

Miguel0101, 2021-04-29:
Oops, on compilation I got this error "```
c: /users/migue/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/. ./lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: C: \ Users \ migrue \ AppData \ Local \ Temp \ arduino_build_831633 / userPage.ino.elf section `.dram0.bss 'will not fit in region` dram0_0_seg'
c: /users/migue/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin /../ lib / gcc / xtensa- esp32-elf / 5.2.0 /../../../../ xtensa-esp32-elf / bin / ld.exe: DRAM segment data does not fit.
c: /users/migue/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin /../ lib / gcc / xtensa- esp32-elf / 5.2.0 /../../../../ xtensa-esp32-elf / bin / ld.exe: region `dram0_0_seg 'overflowed by 29224 bytes
collect2.exe: error: ld returned 1 exit status 
```"

Miguel0101, 2021-04-29:
> Oops, on compilation I got this error "```
> c: /users/migue/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/. [...]

This means that ESP32 does not have enough memory.

Miguel0101, 2021-04-29:
Even with 520 KB of RAM, ESP32 partitions use a lot of RAM.

embeddedt, 2021-04-29:
Does it have a partition for its heap? You could reduce that since LVGL won't be needing it with LV_MEM_CUSTOM disabled.

Miguel0101, 2021-04-29:
> Does it have a partition for its heap? You could reduce that since LVGL won't be needing it with LV_MEM_CUSTOM disabled.

Yes, it has a partition, but I am trying to shrink it. Thanks.

Miguel0101, 2021-04-29:
I managed to reduce the ESP32 partition, compiled (and loaded) successfully, but still displays "No data".

embeddedt, 2021-04-29:
Does the log show the same memory allocation error? If so, if you have a debugger, it would probably be worth stepping through `lv_img_design` (probably [this section](https://github.com/lvgl/lvgl/blob/e08ecffca94623727ef7a8d7ae9f620e15cf7784/src/lv_widgets/lv_img.c#L662-L716)).

Miguel0101, 2021-04-29:
`#include <lvgl.h>
#include "lv_png.h"
#include <TFT_eSPI.h>
#include <TouchScreen.h>
#include <SD.h>

const int XP = 12, XM = 15, YP = 33, YM = 13;
const int TS_LEFT = 167, TS_RT = 927, TS_TOP = 988, TS_BOT = 105;

#define MINPRESSURE 1
#define MAXPRESSURE 1000

TouchScreen ts = TouchScreen(XP, YP, XM, YM, 300);
TSPoint tp;

TFT_eSPI tft = TFT_eSPI(); /* TFT instance */

static lv_disp_buf_t disp_buf;
static lv_color_t buf[LV_HOR_RES_MAX * 5];

LV_IMG_DECLARE(user_icon);

void my_log_cb(lv_log_level_t level, const char * file, uint32_t line, const char * fn_name, const char * dsc)
{
  /*Send the logs via serial port*/
  if(level == LV_LOG_LEVEL_ERROR) Serial.print("ERROR: ");
  if(level == LV_LOG_LEVEL_WARN)  Serial.print("WARNING: ");
  if(level == LV_LOG_LEVEL_INFO)  Serial.print("INFO: ");
  if(level == LV_LOG_LEVEL_TRACE) Serial.print("TRACE: ");

  Serial.print("File: ");
  Serial.println(file);

  char line_str[8];
  sprintf(line_str,"%d", line);
  Serial.print("#");
  Serial.print(line_str);

  Serial.print(": ");
  Serial.print(fn_name);
  Serial.print(": ");
  Serial.print(dsc);
  Serial.println("");
}

void my_print(lv_log_level_t level, const char * file, uint32_t line, const char * dsc)
{

  Serial.printf("%s@%d->%s\r\n", file, line, dsc);
  Serial.flush();
}

/* Display flushing */
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint64_t w = (area->x2 - area->x1 + 1);
  uint64_t h = (area->y2 - area->y1 + 1);

  tft.startWrite();
  tft.setAddrWindow(area->x1, area->y1, w, h);
  tft.pushColors(&color_p->full, w * h, true);
  tft.endWrite();

  lv_disp_flush_ready(disp);
}

/*Read the touchpad*/
bool my_touchpad_read(lv_indev_drv_t * indev_driver, lv_indev_data_t * data)
{
  uint64_t touchX, touchY;
[... 118 lines cut ...]
    lv_keyboard_set_textarea(keyboard, NULL); /*De-assign the text area to hide it cursor if needed*/
    lv_obj_del(keyboard);
    pwd();
  } else if (event == LV_EVENT_CANCEL) {
    lv_obj_align(user_img_block, NULL, LV_ALIGN_CENTER, 0, -45);
    lv_obj_align(user_name, NULL, LV_ALIGN_CENTER, 0, 20);
    lv_obj_align(pwd_ta, NULL, LV_ALIGN_CENTER, 0, 55);
    lv_keyboard_set_textarea(keyboard, NULL); /*De-assign the text area to hide it cursor if needed*/
    lv_obj_del(keyboard);
  }
}

void pwd() {
  if (!strcmp(lv_textarea_get_text(pwd_ta), password) == 0) {
    lv_textarea_set_text(pwd_ta, "Incorrect password!");
    lv_textarea_set_pwd_mode(pwd_ta, false);
    lv_obj_set_style_local_text_color(pwd_ta, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, LV_COLOR_RED);
  } else if (strcmp(lv_textarea_get_text(pwd_ta), password) == 0) {
    lv_textarea_set_text(pwd_ta, "Correct password!");
    lv_textarea_set_pwd_mode(pwd_ta, false);
    lv_obj_set_style_local_text_color(pwd_ta, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT, LV_COLOR_GREEN);
  }
}`

This is the code.

Miguel0101, 2021-04-29:
> Does the log show the same memory allocation error? If so, if you have a debugger, it would probably be worth stepping through `lv_img_design` (probably [this section](https://github.com/lvgl/lvgl/blob/e08ecffca94623727ef7a8d7ae9f620e15cf7784/src/lv_widgets/lv_img.c#L662-L716)).

Ok

Miguel0101, 2021-04-29:
How to use `lv_img_design`?

Miguel0101, 2021-04-29:
@embeddedt Can I try to use the lodepng library?

Miguel0101, 2021-04-29:
Does the "lodepng" library work?

kisvegabor, 2021-04-29:
I recommend trying it out with a very small file, e.g. 10x10 resolution.

Miguel0101, 2021-04-29:
> I recommend trying it out with a very small file, e.g. 10x10 resolution.

Ok.

Miguel0101, 2021-04-29:
it worked, thanks @kisvegabor and @embeddedt

Miguel0101, 2021-04-30:
How do I get more memory? Should I add an external PSRAM?

Miguel0101, 2021-04-30:
[](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html)

ESP32 supports external memory.

Miguel0101, 2021-04-30:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html

kisvegabor, 2021-04-30:
`lodepng` uses the standard `malloc` so it's an ESP-related question of how to use the external RAM for memory allocation.

All I can help with that if you have PSRAM you can use `ps_malloc/free/realloc` in [lodepng.c](https://github.com/lvgl/lv_lib_png/blob/master/lodepng.c#L73-L92).

Miguel0101, 2021-04-30:
> `lodepng` uses the standard `malloc` so it's an ESP-related question of how to use the external RAM for memory allocation.
> 
> All I can help with that if you have PSRAM you can use `ps_malloc/free/realloc` in [lodepng.c](https://github.com/lvgl/lv_lib_png/blob/master/lodepng.c#L73-L92).

Okay, is LVGL compatible with external memories?

embeddedt, 2021-04-30:
LVGL is platform-independent. If your external RAM is mapped in address space normally, it works.

embeddedt, 2021-04-30:
Note that, while I've never used ESP32, I have heard that the performance of external RAM can be noticeably worse compared to internal RAM. If I recall correctly it's something to do with ESP32 having too small of a cache size compared to the size of the RAM.

Miguel0101, 2021-04-30:
> Note that, while I've never used ESP32, I have heard that the performance of external RAM can be noticeably worse compared to internal RAM. If I recall correctly it's something to do with ESP32 having too small of a cache size compared to the size of the RAM.

Okay, thanks.