lvgl/lvgl #2231: Do you has the plan to complete api_map for v7 migrate to v8?

Hpbbs, 2021-04-30:
### Introduce the problem

### Examples and cases

### Suggested solution

I validated the availability the top layer interface(widgets style event) adapter for use macro and incline function

kisvegabor, 2021-04-30:
It's not possible in many cases. I'd rather make a "clean start" instead of a bunch of hacks to support old methods. 

Note that v7 will be supported for a year so you don't have to migrate your project to v8 but it's recommended to start the new ones with it.

embeddedt, 2021-05-07:
To add to @kisvegabor's point - older major releases of LVGL (5.3, 6.1) are usable and stable even after support ends. It's just that we don't have the resources as a project to continue to answer questions and fix bugs regarding them. So if someone's hypothetical project works well with 6.1 - no need to upgrade. :slightly_smiling_face:

kisvegabor, 2021-05-14:
As the questions answered I close this issue.

closed 2021-05-14 by kisvegabor