lvgl/lvgl #2233: [ESP32-IDF] LVGL Image: No Chroma for Indexed Colors

sieren, 2021-05-01:
### Introduce the problem
I am unable to use Chroma Keying on ESP32 IDF due to the dependency on `LV_COLOR_SCREEN_TRANSP` which is dependent on `LV_COLOR_DEPTH_32` in `KConfig`.
However, removing the first dependency (`LV_COLOR_TRANSP_HEX` on `LV_COLOR_SCREEN_TRANSP`) solves the issue and chroma keying works with images with indexed colors.

### Examples and cases
See above.

### Suggested solution
I'm happy to do a PR (one-liner) that removes this dependency from KConfig. I wonder why this is there in the first place, but I suppose it might have been an oversight when merging the ESP32 work into master here?

kisvegabor, 2021-05-03:
`LV_COLOR_SCREEN_TRANSP` is really not required for Chroma keying. Could you send a PR with the removed dependency?

embeddedt, 2021-05-03:
@kisvegabor A PR was already opened: #2234

kisvegabor, 2021-05-03:
Oh shame on me. I could have seen the link above my comment :smile: 

Thank you!

closed 2021-05-05 by sieren