lvgl/lvgl #2237: Memory leak when deleting table rows

nazar01, 2021-05-04:
### Describe the bug
It looks like on deleting table rows the memory they are using isn't freed.
To demonstrate the bug I created a simple screen with label displaying free memory and table.

### Reproduce
To reproduce in lv_sim_eclipse_sdl use the following code in main.c:
```c
int main(int argc, char **argv)
{
  (void)argc; /*Unused*/
  (void)argv; /*Unused*/

  /*Initialize LVGL*/
  lv_init();

  /*Initialize the HAL (display, input devices, tick) for LVGL*/
  hal_init();

  lv_obj_t *free_mem_l = lv_label_create(lv_scr_act(), NULL);
  lv_obj_t *table = lv_table_create(lv_scr_act(), NULL);
  lv_obj_align(table, free_mem_l, LV_ALIGN_OUT_BOTTOM_LEFT, 0, 10);
  lv_table_set_col_cnt(table, 1);
//  lv_demo_widgets();
//  lv_demo_printer();

  while (1) {
    /* Periodically call the lv_task handler.
     * It could be done in a timer interrupt or an OS task too.*/
    lv_task_handler();
    usleep(5 * 1000);
    lv_table_set_row_cnt(table, 1);
    lv_table_set_row_cnt(table, 2);
    lv_table_set_cell_value(table, 0, 0, "HELLO WORLD");
    lv_table_set_cell_value(table, 1, 0, "HELLO WORLD");
    lv_mem_monitor_t monitor;
    lv_mem_monitor(&monitor);
    lv_label_set_text_fmt(free_mem_l, "free mem %d", monitor.free_size);
  }

  return 0;
}
```

### Expected behavior
Free memory doesn't run to 0

embeddedt, 2021-05-04:
Are you trying with 7.11 or `release/v7`? We recently [fixed a memory leak with the table](https://github.com/lvgl/lvgl/commit/0ce7795b7a7f0a3f7cb2f33828c739a7f9db9ea0) in the latter.

nazar01, 2021-05-04:
Oh, I was using 7.11 (default with lv_sim_eclipse_sdl). I've just checked release/v7, it doesn't have the leak. The bug you provided link to looks like exactly my bug. Looks like the problem is solved :)

closed 2021-05-04 by nazar01