lvgl/lvgl #2239: The text in the selected area of the roller is not centered

alec-2018, 2021-05-06:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

### To Reproduce

### Expected behavior

### Screenshots or video

The text in the selected area of the roller is not centered，
It can be reproduced using the following lv_ex_roller_1() program
[image]

void lv_ex_roller_1(void)
{

      lv_obj_t *roller1 = lv_roller_create(lv_scr_act(), NULL);

      lv_roller_set_options(roller1,
                        "January\n"
                        "February\n"
                        "March\n"
                        "April\n"
                        "May\n"
                        "June\n"
                        "July\n"
                        "August\n"
                        "September\n"
                        "October\n"
                        "November\n"
                        "December",
                        LV_ROLLER_MODE_INFINITE);

    lv_roller_set_visible_row_count(roller1, 4);
    lv_obj_align(roller1, NULL, LV_ALIGN_CENTER, 0, 0);
    lv_obj_set_style_local_text_font(roller1, LV_ROLLER_PART_BG, LV_STATE_DEFAULT, &lv_font_montserrat_12);
    lv_obj_set_style_local_text_font(roller1, LV_ROLLER_PART_SELECTED, LV_STATE_DEFAULT, &lv_font_montserrat_30);
    lv_obj_set_event_cb(roller1, event_handler);
}

Solution：
I think this is because the roller_part obtained by mode LV_DESIGN_DRAW_POST in lv_roller_design() is LV_ROLLER_PART_SELECTED instead of LV_ROLLER_PART_BG

[image]
[image]

Do you think it can be modified like this? @kisvegabor

embeddedt, 2021-05-06:
This looks like the correct solution to me; would you like to open a PR for `release/v7`?

alec-2018, 2021-05-11:
I am very happy, but I don’t have the energy to maintain it now, but I will feed back the problems and solutions I currently encounter, hoping to contribute to the open source project

embeddedt, 2021-05-11:
Sounds good; I've applied your fix here: https://github.com/lvgl/lvgl/commit/708272c37cacc190e327373f1972dadccbf137e9