lvgl/lvgl #2240: Out of memory after continues screen update

MyVanitar, 2021-05-06:
Hello,

I have built a screen that contains a few variables which should be updated continuously, for example, it shows the real-time temperature on the screen every 2 seconds or so. the program works well, however after some time, I receiver the below error. and the MCU hangs. Increasing the `LV_MEM_Size` does not help and I will receive this error just a bit later. it is a matter of time. it seems the buffer does not get refreshed or something like that and each time it just consumes the memory.

```
Warn: Couldn't allocate memory 	(lv_mem.c #199 lv_mem_alloc())
Warn: Couldn't allocate memory 	(lv_mem.c #306 lv_mem_realloc())
Error: _lv_mem_buf_get 	(lv_mem.c #511 _lv_mem_buf_get())
Error: Out of memory, can't allocate a new buffer (increase your LV_MEM_SIZE/heap size) (0x00000000) 	(lv_debug.c #127 lv_debug_log_error())
```

This is my code in the loop (the loop gets executed every 3 seconds for example):

```
lv_obj_clean(lv_scr_act());
setup_scr_screen(&ui, temp, humidity, 25, 17, 45);
lv_scr_load(ui.screen);
```

embeddedt, 2021-05-06:
It's definitely a memory leak, but we'll need more info to help. Which LVGL version are you using?

MyVanitar, 2021-05-06:
> It's definitely a memory leak, but we'll need more info to help. Which LVGL version are you using?

7.1.1, I don't remember which commit. I had downloaded it from here around 2 weeks ago or so.

embeddedt, 2021-05-06:
Based on the code provided, I can see one problem - you never delete `lv_scr_act()`. However, you can't do that until **after** the new screen is loaded. Here is an example:

```c
lv_obj_t * old_scr = lv_scr_act();
lv_obj_clean(old_scr);
setup_scr_screen(&ui, temp, humidity, 25, 17, 45);
lv_scr_load(ui.screen);
lv_obj_del(old_scr);
```

MyVanitar, 2021-05-06:
I think I also found a reason and it is my mistake, not the library. for updating the label values, I was calling a function that also contains the initialization of the objects, such as 

```
lv_obj_t *label;
label = lv_label_create(btn1, NULL);
lv_obj_t *btn1 = lv_btn_create(lv_scr_act(), NULL);
```

and when this function is being called frequently, it is like we are generating many samples of all objects on the same area of the memory. We must initialize once and just update the values (such as text, size .. etc) afterward. and this was the advantage of not refreshing the whole screen for an update in a object somewhere.

MyVanitar, 2021-05-06:
> Based on the code provided, I can see one problem - you never delete `lv_scr_act()`. However, you can't do that until **after** the new screen is loaded. Here is an example:
>
> ```c
> lv_obj_t * old_scr [...]

I tested your code and it throws the same error. 
I initialized the objects once and run the updates in a loop. it worked around 10 minutes before I close the debugger. These objects have not been declared globally so I don't know if the assigned memory should be released automatically after calling the function or not. For normal non-global variables, the compiler releases the memory. anyway, this method solved the problem so far.

MyVanitar, 2021-05-07:
I have an update. 
I had the same problem with objects `style` also. I had to update the `lv_style_set_image_recolor` and `lv_style_set_image_recolor_opa` properties with some given input data in the loop. I had to declared and initialize the style so it was causing the same memory leak problem, however after passing a few minutes in my case. by reducing `LV_MEM_SIZE`, I could replicate this problem faster.

I could solve it using `lv_style_reset` after applying the desired styles. Maybe the same concept can be used for the objects (widgets) if the same function is available for the objects. however, I think the library should automatically take care of releasing the memory.

kisvegabor, 2021-05-10:
HAve you called `lv_style_init()` multiple times?

MyVanitar, 2021-05-10:
> Have you called `lv_style_init()` multiple times?

Not really. Only once for each object. You can test and replicate yourself. Just put the code in the loop. only calling `lv_style_reset` solved it. I have also initialized the widgets once.

kisvegabor, 2021-05-10:
I mean haw you called `lv_style_init()` periodically too? because it doesn't free the memory allocated in the styles.

MyVanitar, 2021-05-10:
> I mean haw you called `lv_style_init()` periodically too? because it doesn't free the memory allocated in the styles.

Yes, `lv_style_init()` is inside the function that I should call periodically. 
Besides I have to initialize the Style each time that I want to play with them. Even the Style object can not be accessed in the same function block if I just simply declare an `if statement`. 

for example, in the code below, I have no access to the `style_label` inside the `if (something is true)` statement. 

However I'm not sure if this applies to the widgets or not. because there is no memory release function for the widgets. Now I initialized once when the screen is loaded and then play with the widget properties, such as the label `text` , for example to build a clock using a `Label` that should be updated every second. for the Styles I release the memory using `lv_style_reset`

```
void myfunction (...............)
{
   static lv_style_t style_label;
   lv_style_init(&style_label);

if  (something is true)
{
   ....
   ....
   ....
}
.
.
```

MyVanitar, 2021-05-12:
another update. after some testing, I realized that calling the `lv_style_reset` causes the styles (or updates) only show once, and not later, even by re-initialization. So what I did was not calling the `lv_style_init()` again in the loop (as you said) and therefore I did not use `lv_style_reset`. However `static lv_style_t object_name` must be called each time when we decide to update the Styles.

kisvegabor, 2021-05-15:
> I release the memory using lv_style_reset

That's correct.

> I  realized that calling the lv_style_reset causes the styles (or updates) only show once, and not later, even by re-initialization.

You need to tell LVGL that the style is changed. See the docs [here](http://docs_compiled.lvgl.io/8.0/overview/style.html#add-and-remove-styles-to-a-widget).

MyVanitar, 2021-05-15:
> You need to tell LVGL that the style is changed. See the docs here.

I think there are some typo mistakes in the **Remove styles** section. The compiler says references to the functions are not defined

kisvegabor, 2021-05-15:
Thanks! It's still only the iteration of the new docs so it's likely that there are some typos or non-sense sentences. :sweat_smile: 

Anyway, I fixed the `remove` function names here 26366ad2

MyVanitar, 2021-05-15:
Are these functions added recently? because I still get the same error for `lv_obj_remove_style_all`. if you have added them recently I should download the repo again and replace it.

embeddedt, 2021-05-15:
@kisvegabor I think @MyVanitar is using v7, not v8.

kisvegabor, 2021-05-15:
Of sorry, I forgot that you use v7.  This docs is for v8. űHere is the relevant port of the v7 docs:
https://docs.lvgl.io/latest/en/html/overview/style.html#managing-style-list

MyVanitar, 2021-05-15:
Thank you for the support. You are very professional in the field

closed 2021-05-15 by MyVanitar

kisvegabor, 2021-05-16:
You are welcome :slightly_smiling_face:

MyVanitar, 2023-02-19:
Hello,

I want to update the LVGL from 7 to 8.3.5. So I would appreciate it if you inform me about the Style memory release options in this version. Also releasing the memory in screen swap.

MyVanitar, 2023-02-20:
> After screen change you can delete all the previous screen.

with the same method that you provided above?

kisvegabor, 2023-02-21:
> with the same method that you provided above?

With `lv_obj_del(prev_screen);`