lvgl/lvgl #2242: Is v8 api design to compatible CSSDOM DOM and Box Model from W3C? Why there have no margin attribute

Hpbbs, 2021-05-07:
### Introduce the problem

### Examples and cases

### Suggested solution

kisvegabor, 2021-05-10:
Right now we differ from `border-box` in two points:
1. border has no "size". I.e. its size is not added to the padding to move the children further from the edge of the parent. Do you think the standard `border-box` is better? 
2. no margin, but it can be added later if v8 becomes stable

Hpbbs, 2021-05-11:
Hi, thanks for attention

I think standard `border-box` have some advantages:
1. front-end and Android developer use this model，It`s economy for other developer to learn and understand lvgl layout design
2. I learned that the lvgl style and theme is designed to compatible CSS, use CSS box model is very natrual and fluent
3. for embedded situation, lvgl should has our own optimise just like android can use bitmap to fill border etc. but standard can deduce many cost when standard principle did not differ from you design goal

kisvegabor, 2021-05-11:
@FASTSHIFT 
> Will v8 add stretch support in align-items?

It's supported on grid and planned on flexbox too after v8 release. Is there anything else where we can use it?

closed 2021-05-12 by Hpbbs