lvgl/lvgl #2243: [v7] lv_dropdown options cannot drag with large font

niexingithub, 2021-05-08:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

 lv_dropdown options cannot drag with large font.

I used two fonts, one 14px and the other 30px.

The drop-down menu of 14px can drag the scroll bar normally.

I can only click the drop-down menu of 30px.

I use touchpad device.

### To Reproduce

```c
lv_obj_t *ddlist = lv_dropdown_create(cont, NULL);
lv_dropdown_set_static_option(ddlist, "aaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa\naaa");
lv_obj_t *ddlist1 = lv_dropdown_create(cont, ddlist);
lv_obj_set_style_local_text_font(ddlist1, LV_DROPDOWN_PART_MAIN, LV_STATE_DEFAULT, &lv_font_monstserrta_30);
lv_obj_set_style_local_text_font(ddlist1, LV_DROPDOWN_PART_LIST, LV_STATE_DEFAULT, &lv_font_monstserrta_30);
```
### Expected behavior

30px can't drag scroll bar.

### Screenshots or video

https://user-images.githubusercontent.com/8086937/117532234-18b33200-b019-11eb-9730-e20f706ca472.mp4

kisvegabor, 2021-05-10:
Are you sure you are using the latest version?
`lv_dropdown_set_static_option` not exsits but `lv_dropdown_set_static_options` (`s` on the end).

Can you reproduce it in the simulator?

niexin-dev, 2021-05-11:
Thank you for your attention.
simulator can't reproduce it，the simulator is working fine.
I think it may be caused by the poor performance of my equipment.
If the options are too long,  lv_dropdown will not be able to drag.

niexin-dev, 2021-05-11:
On my device.
set 20 options to  lv_dropdown_set_static_options.
if a option length is "aaa", lv_dropdown is working fine.
if a option length is "aaaaaaaaaa", drag lv_dropdown will select and close dropdown.

kisvegabor, 2021-05-11:
Very strange. If it were an input device issue the problem would happen with smaller fonts too.

I don't think it's due to the low performance. If no release happens on the list it shouldn't be closed.
I've tried to simulted low performance in the simulator like this but it worked well:
```c
  while (1) {
    lv_task_handler();
    usleep(500 * 1000); //500 ms
}
```

Anyway, please print all coordinates and states (pressed, released) in your indev read task. I believe that the issue should be there. 

BTW can you see the issue if the dropdown is placed on an empty screen?

niexin-dev, 2021-05-11:
I add some printf.

```
[16:46:33] x = 248 y = 650 input_w = 95 sw = 3 finger_id = 0
[16:46:33]Niexin Debug : indev_proc_press : 886 : indev_obj_act == 0x4786d0
[16:46:33]Niexin Debug : indev_proc_press : 909 : indev_obj_act == 0x4786d0
[16:46:33]Niexin Debug : indev_proc_press : 998 : indev_obj_act == 0x4786d0
[16:46:34] x = 248 y = 652 input_w = 98 sw = 3 finger_id = 0
[16:46:34]Niexin Debug : indev_proc_press : 901 : indev_obj_act == 0x4786d0
[16:46:34]Niexin Debug : indev_proc_press : 998 : indev_obj_act == 0x4786d0
[16:46:34] x = 248 y = 652 input_w = 99 sw = 3 finger_id = 0
[16:46:34]Niexin Debug : indev_proc_press : 901 : indev_obj_act == 0x4786d0
[16:46:34]Niexin Debug : indev_proc_press : 998 : indev_obj_act == 0x4786d0
[16:46:34] x = 249 y = 657 input_w = 101 sw = 3 finger_id = 0
[16:46:34]Niexin Debug : indev_proc_press : 901 : indev_obj_act == 0x4786d0
[16:46:34]Niexin Debug : indev_proc_press : 998 : indev_obj_act == 0x4786d0
[16:46:34] x = 249 y = 657 input_w = 101 sw = 3 finger_id = 0
[16:46:34]Niexin Debug : indev_proc_press : 901 : indev_obj_act == 0x4786d0
[16:46:34]Niexin Debug : indev_proc_press : 998 : indev_obj_act == 0x4786d0
[16:46:34] x = 250 y = 670 input_w = 101 sw = 3 finger_id = 0
[16:46:34]Niexin Debug : indev_proc_press : 901 : indev_obj_act == 0x4786d0
[16:46:34]Niexin Debug : indev_proc_press : 998 : indev_obj_act == 0x4786d0
[16:46:34]NIexin Debug : indev_drag : 1407 : drag_in_prog == 1

```
The logs above are logs when the drop-down box is working properly.
I clicked drop-down box, and drag options.

if I add a print to lv_dropdown_page_scrl_signal(), the lv_dropdown can't working fine.
I drag the drop-down box, I can't find any coordinates.
The logs are as follows：
```

[16:52:07] x = 246 y = 517 input_w = 92 sw = 3 finger_id = 0
[16:52:07]Niexin Debug : indev_proc_press : 886 : indev_obj_act == 0x47866c
[16:52:07]Niexin Debug : indev_proc_press : 909 : indev_obj_act == 0x47866c
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=11
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : indev_proc_press : 998 : indev_obj_act == 0x47866c
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=12
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=14
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=8
[16:52:07]Niexin Debug : lv_dropdown_page_scrl_signal : 1085 : scrl == 0x47866c, sign=0

```

`BTW can you see the issue if the dropdown is placed on an empty screen?`
I tested, if a option length is short, lv_dropdown is working fine.

embeddedt, 2021-05-15:
> if I add a print to lv_dropdown_page_scrl_signal(), the lv_dropdown can't working fine.

If the behavior is changing just from a `printf` being added, this sounds like an issue with your platform, not LVGL - potentially a stack overflow or some other type of memory corruption.

closed 2021-06-07 by lvgl-bot