lvgl/lvgl #2253: One line text area with left padding fails under certain conditions

jupeos, 2021-05-13:
When using an `lv_textarea` in single line mode the left padding style is not honoured if the call to `lv_textarea_set_one_line` occurs before the padding is set.

### To Reproduce
Using the simulator...
This sets the correct padding.
```c
lv_obj_t* passwordText = lv_textarea_create( cont, NULL );
lv_textarea_set_text( passwordText, "SomePW" );
lv_obj_set_style_local_pad_left( passwordText, LV_TEXTAREA_PART_CURSOR, LV_STATE_DEFAULT, 10 );
lv_textarea_set_one_line( passwordText, true );
```
This set sets 0 left padding
```c
lv_obj_t* passwordText = lv_textarea_create( cont, NULL );
lv_textarea_set_text( passwordText, "SomePW" );
lv_textarea_set_one_line( passwordText, true );
lv_obj_set_style_local_pad_left( passwordText, LV_TEXTAREA_PART_CURSOR, LV_STATE_DEFAULT, 10 );
```
-->

### Expected behavior
I consider this fragile dependency ordering something that needs to be fixed.
If this is considered not a bug then at the very least the documentation around the use of `lv_textarea_set_one_line` needs bold red lettering stating that it must only be called **AFTER** styling of the object has occurred. Burnt a number of hours on this today.

kisvegabor, 2021-05-17:
Is it intentional that you want to set the padding of the CURSOR? Fr me both adds left padding to the cursor which is ugly with the default theme but works.

jupeos, 2021-05-18:
My mistake. This should be `LV_TEXTAREA_PART_BG`. I've updated the original post.

kisvegabor, 2021-05-19:
Thanks, it was really a bug. I've pushed a fix to `release/v7`.

commit b16e7f10 (2021-05-19): fix(textarea) fix setting padding in one line mode

Fixes #2253
closed 2021-05-19 by kisvegabor