lvgl/lvgl #2255: Color font support

MouriNaruto, 2021-05-13:
For more deeper thinking from https://github.com/lvgl/lvgl/issues/2144#issuecomment-840734543.

I also wonder to know how to make the LVGL to support the color font. (I wonder to add support for the color glyphs for Unicode Emoji characters in LVGL and make it looks as good as the app with the modern UI technology.)

The introduction of the color font: https://docs.microsoft.com/en-us/windows/win32/directwrite/color-fonts

kisvegabor, 2021-05-17:
I really like the idea. I'd also allow using color fonts like [these](https://www.fontspace.com/category/color).

I think the largest barrier is finding a suitable rendered that can produce an LVGL-digestible simple format. AFAIK FreeType (the rendered we currently use) doesn't support it. :slightly_frowning_face: 

[This comment](https://stackoverflow.com/a/54783701/1999969) says FreeType supports only RGB images and vector color fonts.
 
[This gist](https://gist.github.com/jokertarot/7583938) shows how to load bitmap fonts.

embeddedt, 2021-05-26:
@kisvegabor It appears that FreeType 2.10 (which was released a month after that answer) does in fact support rendering the "Segoe UI Emoji" font, so I assume that it will work for any other emoji fonts as well.

kisvegabor, 2021-06-09:
Ups, sorry, I forgot to comment here.

Is there someone who has the linking to make some tests with it? 
First first just see the output from FreeType.

guoweilkd, 2021-09-22:
These two links may be helpful:
https://www.freetype.org/freetype2/docs/reference/ft2-color_management.html
https://docs.microsoft.com/zh-cn/typography/opentype/spec/cpal

kisvegabor, 2021-09-23:
Good to know, thanks!

guoweilkd, 2021-09-26:
Simple support experiments :

add draw color font code  in deps/lvgl/src/draw/lv_draw_label.c, like this :
```c
LV_ATTRIBUTE_FAST_MEM static void draw_letter_normal(lv_coord_t pos_x, lv_coord_t pos_y, lv_font_glyph_dsc_t * g,
                                                     const lv_area_t * clip_area,
                                                     const uint8_t * map_p, lv_color_t color, lv_opa_t opa, lv_blend_mode_t blend_mode)
{
    const uint8_t * bpp_opa_table_p;
    uint32_t bitmask_init;
    uint32_t bitmask;
    uint32_t bpp = g->bpp;
    uint32_t shades;
    if(bpp == 3) bpp = 4;

    //draw color font 
    if (bpp == 32) {
        lv_area_t fill_area;
        fill_area.x1 = pos_x;
        fill_area.y1 = pos_y;
        fill_area.x2 = pos_x + g->box_w - 1;
        fill_area.y2 = pos_y + g->box_h - 1;
        _lv_blend_map(clip_area, &fill_area, (lv_color_t*)map_p, NULL, LV_DRAW_MASK_RES_FULL_COVER, opa, blend_mode);
        return;
    }
```

in get_glyph_dsc_cb_nocache of lv_lib_freetype, add `FT_LOAD_COLOR`  macro for FT_Load_Glyph(), and Mark color use bpp = 32:
```c
    error = FT_Load_Glyph(face, glyph_index, FT_LOAD_DEFAULT | FT_LOAD_COLOR);
   .....
    dsc_out->bpp = face->glyph->bitmap.pixel_mode == FT_PIXEL_MODE_BGRA ? 32 : 8;         /*Bit per pixel: 1/2/4/8*/

```
result:
[image]

kisvegabor, 2021-09-30:
Wow, it was very simple in LVGL. One thing is that we should support 16 bit color depth too.  

https://github.com/lvgl/lvgl/pull/2569 will merge lv_lib_freetype into lvgl. After that a PR for it would be very very welcome :slightly_smiling_face:

rabbitsaviola, 2022-01-17:
Hi @kisvegabor ,
  do you have plan to upgrade `lv_font_conv` to support this kind of color font also?

kisvegabor, 2022-01-19:
I'm still interested in this feature but the format needs to be designed and it needs to implemented in `lv_font_conv`. 

I'd happily help with the design but I can't implement it properly in JS. Can you contribute with it?

rabbitsaviola, 2022-01-20:
Sorry, I cannot assure to complete this task as I'm not familiar with node.js either. But will let you know when it's availabe. Our team is working on emoji recently and may try to update the tool.

kisvegabor, 2022-01-20:
I think ultimately the win-win situation would be be to make the LVGL upstream support color fonts. This way you could avoid maintaining  you fork.

So feel free to open issues or comment here to think to together how to manage these things in LVGL.

guoweilkd, 2022-03-11:
#3160,  This PR can solve the problem of using label to display picture emoji. However, otf color font display is not currently supported.

kisvegabor, 2022-03-17:
#3160 is merged so I close this issue.

Thank you @guoweilkd!

closed 2022-03-17 by kisvegabor