lvgl/lvgl #2256: Memory leaking in changing the screens in LVGL 7.11

MyVanitar, 2021-05-15:
Hello,

When I change the screens, it works, however after a few screen changes, I receive this error in the debugger. it is easy to replicate the error. Just reduce the  `#define LV_MEM_SIZE` and try to change the screens. if `MEM_Size` is big, you have to changes the screens many times and this may lead to the thought that there is no memory leaking.

```
Warn: Couldn't allocate memory 	(lv_mem.c #199 lv_mem_alloc())
Warn: Couldn't allocate memory 	(lv_mem.c #306 lv_mem_realloc())
Error: _lv_mem_buf_get 	(lv_mem.c #511 _lv_mem_buf_get())
Error: Out of memory, can't allocate a new buffer (increase your LV_MEM_SIZE/heap size) (0x00000000) 	(lv_debug.c #127 lv_debug_log_error())
```

What I tested to change screens were (the previous screen was "welcome" and the new screen is "clock")

```c
lv_obj_clean(ui.welcome);
setup_scr_clock(&ui, 0);
lv_scr_load(ui.clock);
lv_obj_del(ui.welcome);
```

also tested something like this

```c
lv_obj_t * old_scr = lv_scr_act();
lv_obj_clean(old_scr);
setup_scr_screen(&ui, 0);
lv_scr_load(ui.clock);
lv_obj_del(old_scr);
```

also tested this:

```c
lv_obj_clean(lv_scr_act());
setup_scr_clock(&ui, 0);
lv_scr_load(ui.clock);
```

**NONE Helped**

-----------------------------------------------------------------

Example code of an screen inside the source C file:

```c
void setup_scr_welcome(lv_ui *ui){

	//Write codes welcome
	ui->welcome = lv_obj_create(NULL, NULL);

	//Write codes welcome_cont5
	ui->welcome_cont5 = lv_cont_create(ui->welcome, NULL);
	
	//Write codes welcome_img_logo_welcome
	ui->welcome_img_logo_welcome = lv_img_create(ui->welcome, NULL);

	//Write style LV_IMG_PART_MAIN for welcome_img_logo_welcome
	static lv_style_t style_welcome_img_logo_welcome_main;
	lv_style_init(&style_welcome_img_logo_welcome_main);

	//Write style state: LV_STATE_DEFAULT for style_welcome_img_logo_welcome_main
	lv_style_set_image_recolor(&style_welcome_img_logo_welcome_main, LV_STATE_DEFAULT, lv_color_make(0xff, 0xff, 0xff));
	lv_style_set_image_recolor_opa(&style_welcome_img_logo_welcome_main, LV_STATE_DEFAULT, 0);
	lv_style_set_image_opa(&style_welcome_img_logo_welcome_main, LV_STATE_DEFAULT, 255);
	lv_obj_add_style(ui->welcome_img_logo_welcome, LV_IMG_PART_MAIN, &style_welcome_img_logo_welcome_main);
	lv_obj_set_pos(ui->welcome_img_logo_welcome, 84, 61);
	lv_obj_set_size(ui->welcome_img_logo_welcome, 143, 78);
	lv_obj_set_click(ui->welcome_img_logo_welcome, true);
	lv_img_set_src(ui->welcome_img_logo_welcome,&_logo_143x78);
	lv_img_set_pivot(ui->welcome_img_logo_welcome, 0,0);
	lv_img_set_angle(ui->welcome_img_logo_welcome, 0);

	//Write codes welcome_label_firmware
	ui->welcome_label_firmware = lv_label_create(ui->welcome, NULL);
	lv_label_set_text(ui->welcome_label_firmware, "Firmware: 2021-05-V1.0");
	lv_label_set_long_mode(ui->welcome_label_firmware, LV_LABEL_LONG_BREAK);
	lv_label_set_align(ui->welcome_label_firmware, LV_LABEL_ALIGN_LEFT);

	//Write style LV_LABEL_PART_MAIN for welcome_label_firmware
	static lv_style_t style_welcome_label_firmware_main;
	lv_style_init(&style_welcome_label_firmware_main);

	//Write style state: LV_STATE_DEFAULT for style_welcome_label_firmware_main
	lv_style_set_radius(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, 0);
	lv_style_set_bg_color(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, lv_color_make(0x00, 0x00, 0x00));
	lv_style_set_bg_grad_color(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, lv_color_make(0x00, 0x00, 0x00));
	lv_style_set_bg_grad_dir(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, LV_GRAD_DIR_VER);
	lv_style_set_bg_opa(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, 255);
	lv_style_set_text_color(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, lv_color_make(0xff, 0xff, 0xff));
	lv_style_set_text_font(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, &lv_font_montserrat_16);
	lv_style_set_text_letter_space(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, 2);
	lv_style_set_pad_left(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, 0);
	lv_style_set_pad_right(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, 0);
	lv_style_set_pad_top(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, 0);
	lv_style_set_pad_bottom(&style_welcome_label_firmware_main, LV_STATE_DEFAULT, 0);
	lv_obj_add_style(ui->welcome_label_firmware, LV_LABEL_PART_MAIN, &style_welcome_label_firmware_main);
	lv_obj_set_pos(ui->welcome_label_firmware, 12, 211);
	lv_obj_set_size(ui->welcome_label_firmware, 230, 0);
}
```

Header file:

```c
typedef struct {
	lv_obj_t *welcome;	
	lv_obj_t *welcome_img_logo_welcome;
	lv_obj_t *welcome_label_firmware;
}lv_ui;
void setup_scr_welcome(lv_ui *ui);
```

embeddedt, 2021-05-15:
You shouldn't call `lv_style_init` more than once on those styles; you should use `lv_style_reset` for reinitialization.

MyVanitar, 2021-05-15:
That's it. The problem solved. Thanks.

closed 2021-05-15 by MyVanitar