lvgl/lvgl #2264: How to improve the speed of your textarea

codingFoxy, 2021-05-20:
### Introduce the problem
When working with a lot of text inside a textarea or many, many `lv_textarea_add_text()` calls at a time, you might stumble upon some timing problems. Not only does the output become slow with increasing text size, I also noticed the text being printed in the wrong order. I couldn't find a fix in previous issues here, so I had to figure out a little workaround myself and actually came up with something that works for my project.

### Examples and cases
I work with a servo motor controller that is connected to my development board via LPUART.  When starting an operation on the controller (e. g. a bootloader), a _lot_ of text is sent via LPUART. My program receives the text and prints it onto my terminal.

My **terminal** consists of a plain textarea and a keyboard.

So far, I've used a very simple function for my terminal that receives text and prints it to the textarea.

```C
void TGUI_printToTerminal( const char * aStr )
{
    lv_textarea_add_text( pMyTextarea, aStr );
}
```

This works out just fine when the function is not called too often, but when it is, things start getting messed up. I controlled my input by writing it into an array and it looked all fine. If you'd print it via `printf()` it would look like this:

```
This is line 1.
This is line 2.
This is line 3.
```

But what I instead started seeing on my terminal would be:

```
This is line 1.
This is line 3.
This is line 2.
```

Reading through other people's timing issues with textarea, I realized that using `lv_textarea_add_text()` can become very inefficient for long texts because the function has to read the old text again and again every time it's being called. To speed things up a little, I wrote a very simple workaround.

### Suggested solution

```C
#define SCROLLBACK  4096
static char scrollbackTxt[SCROLLBACK]    //buffer for the terminal text
static int scrollbackPtr = 0;

void TGUI_printToTerminal( const char * aStr )
{
    terminalAddText( aStr );
}

static void terminalAddText( const char * aStr )
{
    int freeSpaces = SCROLLBACK - scrollbackPtr;   //What is the space I have left?
    int newTxtLen = strlen( aStr );
    if (newTxtLen >= freeSpaces)     //Does the string fit into the free space?
    {
        /*
        * The buffer would overrun, so we have to delete the oldest text
        * 
        * To be honest, I haven't tested this part of the code yet, so feel free to correct me
        * if i'm wrong somewhere.
        */
        int keepLen = SCROLLBACK - newTxtLen - (SCROLLBACK - scrollbackPtr);    //length of the text to keep
        memcpy( scrollbackTxt, scrollbackTxt + newTxtLen, keepLen );
        scrollbackPtr -= newTxtLen;
    }
    memcpy( scrollbackTxt + scrollbackPtr, aStr, newTxtLen );
    scrollbackPtr += newTxtLen;
    lv_textarea_set_text( pMyTextarea, scrollbackTxt );
}
```
Using `lv_textarea_set_text` instead of `lv_textarea_add_text` makes the print operation much faster and ensures everything is printed in the right order. This migh use more memory on your device, but if your main focus is on speed, this works very well.
It might not be a super pretty or super innovative solution, but it does the trick and I'm glad if I could help you with it.

kisvegabor, 2021-05-24:
Thanks for sharing this. Could you add it as an example here? 
BTW, if you use only `set_text` it'd be possible to use a label instead of a text area. Especially in v8:
```c
lv_obj_t * cont = lv_obj_create(lv_scr_act());
lv_obj_set_size(cont, 200, 100);

lv_obj_t * label = lv_label_create(cont);
lv_obj_set_width(label, lv_pct(100));
lv_label_set_text(label, "This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! ");

/*Scroll to the bottom*/
lv_obj_update_layout(cont);         /*Be sure the coordinates are updated*/
lv_obj_scroll_to_y(cont, 10000, LV_ANIM_ON);    /*Use a large number to reach the bottom*/
```
[image]

closed 2021-06-16 by lvgl-bot

lbxhacker, 2023-04-10:
> 感谢您分享这个。你能[在这里](https://github.com/lvgl/lvgl/tree/master/examples/widgets)添加它作为一个例子吗？顺便说一句，如果您只使用，则可以使用标签而不是文本区域。特别是在 v8 中：`set_text`
> 
> ```c
> lv_obj_t * cont = lv_obj_create(lv_scr_act());
> lv_obj_set_size(cont, 200, 100);
> 
> lv_obj_t * label = lv_label_create(cont);
> lv_obj_set_width(label, lv_pct(100));
> lv_label_set_text(label, "This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! This is a simple text area! ");
> 
> /*Scroll to the bottom*/
> lv_obj_update_layout(cont);         /*Be sure the coordinates are updated*/
> lv_obj_scroll_to_y(cont, 10000, LV_ANIM_ON);    /*Use a large number to reach the bottom*/
> ```
> 
> [image] [ [image] ](https://user-images.githubusercontent.com/7599318/119343946-5d84cd00-bc97-11eb-81b7-66ff671c404d.gif) [ ](https://user-images.githubusercontent.com/7599318/119343946-5d84cd00-bc97-11eb-81b7-66ff671c404d.gif)

hi!Label are not appropriate because a lot of the time you're adding text, and Label don't do that