lvgl/lvgl #2272: Doxygen Formatting

ryanpowers, 2021-05-26:
Just as an up-front note, this is a really minor documentation issue, but I noticed it and I figured I'd *document* it by posting it here.

### Introduce the problem
I noticed an issue with some of the doxygen generated documentation and found there was a formatting error with the comments.

### Examples and cases
I saw the issue in the documentation for the [Bar widget](https://docs.lvgl.io/7.11/widgets/bar.html#api), but haven't found other cases yet. The enumeration section for the bar parts has the descriptions of the values offset by one, and as a result, incorrectly described them. The code for it is as follows (lv_bar.h):
```c
/** Bar parts */
enum {
    LV_BAR_PART_BG, /** Bar background style. */
    LV_BAR_PART_INDIC, /** Bar fill area style. */
    _LV_BAR_PART_VIRTUAL_LAST
};
typedef uint8_t lv_bar_part_t;
```
The enumeration is documented on the line after the value, as is commonly done. Since it is *after* the definition, doxygen considers this a comment describing the *next* definition, causing the off-by-one error.

### Suggested solution
Because the comment appears after the definition, it needs to be backwards referencing comment:
```c
/** Bar parts */
enum {
    LV_BAR_PART_BG, /**< Bar background style. */
    LV_BAR_PART_INDIC, /**< Bar fill area style. */
    _LV_BAR_PART_VIRTUAL_LAST
};
typedef uint8_t lv_bar_part_t;
```

I checked a few files for other occurrences while writing this, and found [Calendar](https://docs.lvgl.io/7.11/widgets/calendar.html#api) has a mix on it's parts enumeration, with one entry having the backwards referencing comment, and the rest being normal doxygen.

I also checked the coding style document and it does not specify this sort of doxygen comments for enumerations.

I can't think of a good way to go through all the code to find occurrences like this, but perhaps a regexp could come up with something. I would suggest adding this documentation comment form to the style guide to ensure correct usage in the future.

kisvegabor, 2021-05-26:
Thanks for pointing it out. I'm afraid we really need to look into the files manually. 

Let's leave this issue open for new contributors.

himanshu007-creator, 2021-05-26:
Hi, i would like to work on this issue.

kisvegabor, 2021-05-28:
@himanshu007-creator 
Amazing! Are you familiar with [Pull requests](https://docs.lvgl.io/8.0/CONTRIBUTING.html#pull-request)?

himanshu007-creator, 2021-05-28:
> @himanshu007-creator
> Amazing! Are you familiar with [Pull requests](https://docs.lvgl.io/8.0/CONTRIBUTING.html#pull-request)?

 i LOVE green squares :)

kisvegabor, 2021-05-28:
What green squares? :smile:

closed 2021-06-19 by lvgl-bot