lvgl/lvgl #2273: Underflow in memset caused by draw_letter_subpx

deltabeard, 2021-05-26:
# Description

When using a font with sub-pixel rendering, and modifying the lv_theme_material theme to use less padding, the function `lv_area_get_width` can return a negative integer which is then past straight into `memset` which takes in an unsigned integer, causing an underflow.

This error occurs on Linux in my project at https://github.com/deltabeard/3ds-lvgl/tree/04ad1ecebfc7c3dd391e9708029887be2dbf79ba when changing tabs (it may take a few times for the bug to occur).

This is with LVGL v7 b16e7f1076963f123b36dec9ce1dd5950057ca1a

# Trace

```c
#0  0x00007f82aa4db981 in __memset_avx2_erms () from /usr/lib/libc.so.6
#1  0x00005643f1230f36 in _lv_memset_00 (dst=0x5643f235ee88, len=18446744073709551612) at inc/lvgl/src/lv_draw/../lv_misc/lv_mem.h:185
#2  0x00005643f1232f5f in draw_letter_subpx (pos_x=-12, pos_y=230, g=0x7ffc9356fe2c, clip_area=0x7ffc93570060, map_p=0x5643f255143c "", color=..., opa=255 '\377', blend_mode=0 '\000') at inc/lvgl/src/lv_draw/lv_draw_label.c:775
#3  0x00005643f123215c in lv_draw_letter (pos_p=0x7ffc9356ff14, clip_area=0x7ffc93570060, font_p=0x5643f13041e0 <noto_sans_14_common>, letter=103, color=..., opa=255 '\377', blend_mode=0 '\000') at inc/lvgl/src/lv_draw/lv_draw_label.c:443
#4  0x00005643f1231ba0 in lv_draw_label (coords=0x7ffc93570050, mask=0x7ffc93570060, dsc=0x7ffc93570070, txt=0x5643f2585628 "", hint=0x0) at inc/lvgl/src/lv_draw/lv_draw_label.c:325
#5  0x00005643f126337f in lv_label_design (label=0x5643f2585668, clip_area=0x7ffc93570160, mode=0 '\000') at inc/lvgl/src/lv_widgets/lv_label.c:1347
#6  0x00005643f1229522 in lv_refr_obj (obj=0x5643f2585668, mask_ori_p=0x7ffc93570210) at inc/lvgl/src/lv_core/lv_refr.c:694
#7  0x00005643f12295ea in lv_refr_obj (obj=0x5643f2585468, mask_ori_p=0x7ffc935702a0) at inc/lvgl/src/lv_core/lv_refr.c:734
#8  0x00005643f12295ea in lv_refr_obj (obj=0x5643f2593868, mask_ori_p=0x7ffc93570330) at inc/lvgl/src/lv_core/lv_refr.c:734
#9  0x00005643f12295ea in lv_refr_obj (obj=0x5643f2593768, mask_ori_p=0x7ffc935703c0) at inc/lvgl/src/lv_core/lv_refr.c:734
#10 0x00005643f12295ea in lv_refr_obj (obj=0x5643f257fbe8, mask_ori_p=0x7ffc93570450) at inc/lvgl/src/lv_core/lv_refr.c:734
#11 0x00005643f12295ea in lv_refr_obj (obj=0x5643f257fb28, mask_ori_p=0x7ffc935704e0) at inc/lvgl/src/lv_core/lv_refr.c:734
#12 0x00005643f12295ea in lv_refr_obj (obj=0x5643f254ebb8, mask_ori_p=0x7ffc93570570) at inc/lvgl/src/lv_core/lv_refr.c:734
#13 0x00005643f12295ea in lv_refr_obj (obj=0x5643f258d4c8, mask_ori_p=0x7ffc93570600) at inc/lvgl/src/lv_core/lv_refr.c:734
#14 0x00005643f12295ea in lv_refr_obj (obj=0x5643f25924e8, mask_ori_p=0x7ffc935706b0) at inc/lvgl/src/lv_core/lv_refr.c:734
#15 0x00005643f122939a in lv_refr_obj_and_children (top_p=0x5643f25924e8, mask_p=0x7ffc935706b0) at inc/lvgl/src/lv_core/lv_refr.c:636
#16 0x00005643f12291e7 in lv_refr_area_part (area_p=0x5643f258c474) at inc/lvgl/src/lv_core/lv_refr.c:567
#17 0x00005643f1228ec5 in lv_refr_area (area_p=0x5643f258c474) at inc/lvgl/src/lv_core/lv_refr.c:490
#18 0x00005643f1228b6c in lv_refr_areas () at inc/lvgl/src/lv_core/lv_refr.c:400
#19 0x00005643f1228552 in _lv_disp_refr_task (task=0x5643f2592448) at inc/lvgl/src/lv_core/lv_refr.c:199
#20 0x00005643f1244454 in lv_task_exec (task=0x5643f2592448) at inc/lvgl/src/lv_misc/lv_task.c:386
#21 0x00005643f1243dc7 in lv_task_handler () at inc/lvgl/src/lv_misc/lv_task.c:134
#22 0x00005643f1216c9f in main (argc=1, argv=0x7ffc93570b78) at src/main.c:520
```

Note in frame 1 whereby len is a very large number. This is because map_area is `{x1 = 0, y1 = 230, x2 = -5, y2 = 239}`, which produces a result of -5 from `lv_area_get_width`, causing the bug.

[dump.zip](https://github.com/lvgl/lvgl/files/6549986/dump.zip)

# Patch

The proposed patch simply performs the memset if the width is a positive integer.

```diff
diff --git a/inc/lvgl/src/lv_draw/lv_draw_label.c b/inc/lvgl/src/lv_draw/lv_draw_label.c
index 970791f..282d114 100644
--- a/inc/lvgl/src/lv_draw/lv_draw_label.c
+++ b/inc/lvgl/src/lv_draw/lv_draw_label.c
@@ -772,7 +772,10 @@ static void draw_letter_subpx(lv_coord_t pos_x, lv_coord_t pos_y, lv_font_glyph_
lv_draw_mask_res_t mask_res = lv_draw_mask_apply(mask_buf + mask_p_start, map_area.x1, map_area.y2,
lv_area_get_width(&map_area));
if(mask_res == LV_DRAW_MASK_RES_TRANSP) {
-                _lv_memset_00(mask_buf + mask_p_start, lv_area_get_width(&map_area));
+               lv_coord_t w = lv_area_get_width(&map_area);
+               if(w > 0) {
+                   _lv_memset_00(mask_buf + mask_p_start, w);
+               }
}
}
```

kisvegabor, 2021-05-28:
Hi,

Thanks for the report. It seems we can simply return if the width of `map_area` is <= 0. 

Could you test with this?
```c
    //Line 676
    lv_area_t map_area;
    map_area.x1 = col_start / 3 + pos_x;
    map_area.x2 = col_end / 3  + pos_x - 1;
    map_area.y1 = row_start + pos_y;
    map_area.y2 = map_area.y1;

  if(map_area.x2 < map_area.x1) return;

```

deltabeard, 2021-05-28:
Hello, thanks for you comment.

After reverting my patch and making your suggested changes, the program freezes after printing the messages below.
```
ERROR inc/lvgl/src/lv_misc/lv_mem.c +521 _lv_mem_buf_get(): _lv_mem_buf_get
ERROR inc/lvgl/src/lv_misc/lv_debug.c +127 lv_debug_log_error(): No free buffer. Increase LV_MEM_BUF_MAX_NUM. (0x0000000000000000)
```

I did a SIGTRAP and used gdb to generate the stack trace below.
```c
Program terminated with signal SIGTRAP, Trace/breakpoint trap.
#0  0x0000555756b6b499 in _lv_mem_buf_get (size=size@entry=150) at inc/lvgl/src/lv_misc/lv_mem.c:521
521         LV_DEBUG_ASSERT(false, "No free buffer. Increase LV_MEM_BUF_MAX_NUM.", 0x00);
[Current thread is 1 (Thread 0x7f227a5db240 (LWP 17234))]
(gdb) bt
#0  0x0000555756b6b499 in _lv_mem_buf_get (size=size@entry=150) at inc/lvgl/src/lv_misc/lv_mem.c:521
#1  0x0000555756b5f791 in draw_letter_subpx (pos_x=pos_x@entry=-11, pos_y=pos_y@entry=71, g=g@entry=0x7ffd292d25ec, clip_area=clip_area@entry=0x7ffd292d27e0,
map_p=0x555757e1f9c3 "\317\352@", color=..., color@entry=..., opa=255 '\377', blend_mode=0 '\000') at inc/lvgl/src/lv_draw/lv_draw_label.c:659
#2  0x0000555756b602cc in lv_draw_letter (pos_p=pos_p@entry=0x7ffd292d26b4, clip_area=clip_area@entry=0x7ffd292d27e0,
font_p=font_p@entry=0x555756c19240 <noto_sans_14_common>, letter=letter@entry=49, color=..., color@entry=..., opa=opa@entry=255 '\377', blend_mode=0 '\000')
at inc/lvgl/src/lv_draw/lv_draw_label.c:443
#3  0x0000555756b6082f in lv_draw_label (coords=coords@entry=0x7ffd292d27d0, mask=mask@entry=0x7ffd292d27e0, dsc=dsc@entry=0x7ffd292d27f0,
txt=0x7f2244001ba8 "01df93dc38eb5fef919636eede59c87841c860dbd91c49d5c53eda3c3baf90113a7c638840b6a998c1793ae9fa7ee02fb768caddcff06e13668f732f385e019c",
hint=hint@entry=0x0) at inc/lvgl/src/lv_draw/lv_draw_label.c:325
#4  0x0000555756b7ef79 in lv_label_design (label=0x7f2244001a48, clip_area=<optimized out>, mode=<optimized out>) at inc/lvgl/src/lv_widgets/lv_label.c:1347
#5  0x0000555756b5928a in lv_refr_obj (obj=obj@entry=0x7f2244001a48, mask_ori_p=mask_ori_p@entry=0x7ffd292d2980) at inc/lvgl/src/lv_core/lv_refr.c:694
#6  0x0000555756b59335 in lv_refr_obj (obj=obj@entry=0x7f2244008808, mask_ori_p=0x7ffd292d2990, mask_ori_p@entry=0x7ffd292d2a10) at inc/lvgl/src/lv_core/lv_refr.c:734
#7  0x0000555756b59335 in lv_refr_obj (obj=obj@entry=0x555757e1aed8, mask_ori_p=0x7ffd292d2a20, mask_ori_p@entry=0x7ffd292d2aa0) at inc/lvgl/src/lv_core/lv_refr.c:734
#8  0x0000555756b59335 in lv_refr_obj (obj=obj@entry=0x555757e1add8, mask_ori_p=0x7ffd292d2ab0, mask_ori_p@entry=0x7ffd292d2b30) at inc/lvgl/src/lv_core/lv_refr.c:734
#9  0x0000555756b59335 in lv_refr_obj (obj=obj@entry=0x555757e0ff68, mask_ori_p=0x7ffd292d2b40, mask_ori_p@entry=0x7ffd292d2bc0) at inc/lvgl/src/lv_core/lv_refr.c:734
#10 0x0000555756b59335 in lv_refr_obj (obj=obj@entry=0x555757e0fea8, mask_ori_p=0x7ffd292d2bd0, mask_ori_p@entry=0x7ffd292d2c50) at inc/lvgl/src/lv_core/lv_refr.c:734
#11 0x0000555756b59335 in lv_refr_obj (obj=obj@entry=0x555757e375f8, mask_ori_p=0x7ffd292d2c60, mask_ori_p@entry=0x7ffd292d2ce0) at inc/lvgl/src/lv_core/lv_refr.c:734
#12 0x0000555756b59335 in lv_refr_obj (obj=obj@entry=0x555757e33e38, mask_ori_p=0x7ffd292d2cf0, mask_ori_p@entry=0x7ffd292d2d70) at inc/lvgl/src/lv_core/lv_refr.c:734
#13 0x0000555756b59335 in lv_refr_obj (obj=obj@entry=0x555757e304c8, mask_ori_p=0x7ffd292d2d80, mask_ori_p@entry=0x7ffd292d2e10) at inc/lvgl/src/lv_core/lv_refr.c:734
#14 0x0000555756b59365 in lv_refr_obj_and_children (top_p=top_p@entry=0x555757e304c8, mask_p=mask_p@entry=0x7ffd292d2e10) at inc/lvgl/src/lv_core/lv_refr.c:636
#15 0x0000555756b597c1 in lv_refr_area_part (area_p=area_p@entry=0x555757e33c04) at inc/lvgl/src/lv_core/lv_refr.c:567
#16 0x0000555756b59ab4 in lv_refr_area (area_p=0x555757e33c04) at inc/lvgl/src/lv_core/lv_refr.c:478
#17 0x0000555756b59bce in lv_refr_areas () at inc/lvgl/src/lv_core/lv_refr.c:400
#18 0x0000555756b59c7d in _lv_disp_refr_task (task=0x555757e30178) at inc/lvgl/src/lv_core/lv_refr.c:199
#19 0x0000555756b6b84a in lv_task_exec (task=0x555757e30178) at inc/lvgl/src/lv_misc/lv_task.c:386
#20 0x0000555756b6b99f in lv_task_handler () at inc/lvgl/src/lv_misc/lv_task.c:134
#21 0x0000555756b4d8f4 in main (argc=<optimized out>, argv=<optimized out>) at src/main.c:744
```

kisvegabor, 2021-05-29:
Ah, the buffer is not released.

The initialization of `map_area` can be done before calling `lv_mem_buf_get` so no buffer will be allocated if it returns.

```diff
diff --git a/src/draw/lv_draw_label.c b/src/draw/lv_draw_label.c
index 79659c7d..1c686e1e 100644
--- a/src/draw/lv_draw_label.c
+++ b/src/draw/lv_draw_label.c
@@ -664,6 +664,14 @@ static void draw_letter_subpx(lv_coord_t pos_x, lv_coord_t pos_y, lv_font_glyph_
     int32_t col_bit;
     col_bit = bit_ofs & 0x7; /*"& 0x7" equals to "% 8" just faster*/
 
+    lv_area_t map_area;
+    map_area.x1 = col_start / 3 + pos_x;
+    map_area.x2 = col_end / 3  + pos_x - 1;
+    map_area.y1 = row_start + pos_y;
+    map_area.y2 = map_area.y1;
+
+    if(map_area.x2 <= map_area.x1) return;
+
     int32_t mask_buf_size = box_w * box_h > _LV_MASK_BUF_MAX_SIZE ? _LV_MASK_BUF_MAX_SIZE : g->box_w * g->box_h;
     lv_opa_t * mask_buf = lv_mem_buf_get(mask_buf_size);
     int32_t mask_p = 0;
@@ -682,12 +690,6 @@ static void draw_letter_subpx(lv_coord_t pos_x, lv_coord_t pos_y, lv_font_glyph_
     /*If the letter is partially out of mask the move there on draw_buf*/
     disp_buf_buf_tmp += (row_start * disp_buf_width) + col_start / 3;
 
-    lv_area_t map_area;
-    map_area.x1 = col_start / 3 + pos_x;
-    map_area.x2 = col_end / 3  + pos_x - 1;
-    map_area.y1 = row_start + pos_y;
-    map_area.y2 = map_area.y1;
-
     uint8_t other_mask_cnt = lv_draw_mask_get_cnt();
 
     uint8_t font_rgb[3];
```

embeddedt, 2021-06-13:
@kisvegabor Does your patch need to be applied to v7?

deltabeard, 2021-06-13:
Apologies for the delay. I have just tested the patch and can confirm that this fix works.

closed 2021-06-15 by kisvegabor

kisvegabor, 2021-06-15:
No worries. I've fixed it in v8 (master) and v7 (release/v7).

commit 6d5ac702 (2021-06-15): fix(draw) underflow in subpixel font drawing

Fixes: #2273
commit e04a8d29 (2021-06-15): fix(draw) underflow in subpixel font drawing

Fixes: #2273
commit 502ca5b8 (2021-07-16): fix(draw) underflow in subpixel font drawing

Fixes: #2273