lvgl/lvgl #2274: keyboard is too Sensitive.. how to slow down?

acer1204, 2021-05-30:
When i input like a 'a' something the textfield will add two or more 'a'.
it looks a bounce jump, how to avoid this problem? 

this is my keyboard event code, has any ideal?

`void kb_event_cb(lv_obj_t *keyboard, lv_event_t e)`
`{
   ` txt = lv_btnmatrix_get_active_btn_text(keyboard);`
    `lv_keyboard_def_event_cb(keyboardLayout, e);`
    `switch (e)`
    `{`
       ` case LV_EVENT_RELEASED:`
           ` Serial.println("LV_EVENT_RELEASED\n");`
            `if(isInputKey)`
            `{`
                `if (strcmp(txt, LV_SYMBOL_OK) == 0 || strcmp(txt, LV_SYMBOL_CLOSE) == 0)`
                `{
                   ` lv_keyboard_set_textarea(keyboardLayout, NULL);`
                    `lv_obj_del(keyboardLayout);`
                    `keyboardLayout = NULL;`
                `}`
                `isInputKey=false;`
            `}`
           ` break;`
        `case LV_EVENT_PRESSING:`
           ` Serial.println("LV_EVENT_PRESSING\n");`
            `break;`
        `case LV_EVENT_PRESSED:`
            `isInputKey=true;`
           ` Serial.println("LV_EVENT_PRESSED\n");`
            `break;`
        `case LV_EVENT_CLICKED:`
           ` Serial.println("LV_EVENT_CLICKED\n");`
        `break;`
    `default:`
        `break;`
    `}`
`}`

embeddedt, 2021-05-30:
It sounds like an issue with your touch driver. You probably need to add some debouncing logic to it.

acer1204, 2021-05-31:
> It sounds like an issue with your touch driver. You probably need to add some debouncing logic to it.

Is add debouncing in lv_keyboard_def_event_cb(...)?

embeddedt, 2021-05-31:
No, in your touchscreen driver (`read_cb`).

closed 2021-06-23 by lvgl-bot

Satnet, 2022-07-18:
Everyone misinterpretedthis issue. When using a custom keyboard you need to remove the old event handler. Otherwise both event handler will be called, thus inputing 2 characters each time you click once.

Do this:
```
lv_obj_remove_event_cb(keyboardObject, lv_keyboard_def_event_cb); // THIS IS THE OLD HANDLER
lv_obj_add_event_cb(keyboardObject, new_event_handler, LV_EVENT_VALUE_CHANGED, NULL);
```