lvgl/lvgl #2276: [v8.1] Planning

kisvegabor, 2021-06-01:
Hi,

v8.0 is released so let's move forward and start to think about v8.1.

In the list below I collected the ideas I found useful and reasonable for v8.1. It's not written into stone, so feel free to make suggestions.

If you are interested in contributing in v8.1, just comment here! :slightly_smiling_face: 

### Features
- Unit testing (gtest?). See #1658
- Benchmarking (gem5 or qemu?). See #1660  
- lv_snapshot: buffer a widget and all of its children into an image. The source widget can be on a different screen too. The resulting image can be transformed.
- 9-patch support for lv_imgbtn.
- Optimize line and circle drawing and masking
- High level GPU support. See #2058

### Examples
#### Missing examples
  - color examples
  - coordinate examples
  - draw (mask and event) examples
  - style color filter and const examples
  - at least 2 examples for every widget (basic + styling)

#### New features
- merge MP examples
- add a "Try out yourself" button to the Micropython examples
  
### Discuss
- CPP binding  
- Plugins. In v8 core and extra widgets are separated. With the new flexible events, the behavior of the widgets can be modified in a modular way. E.g. a plugin to add faded area to a line chart (as in the widgets demo)

### Docs
- Display the Micropytohn examples too.
- Add a link to the example C and py files
- List of all examples on a page. All in iframes grouped by category (e.g. flex, style, button)

### Others
- Add automatic rebuild to get binary directly. Similarly to [STM32F746 project](https://github.com/lvgl/lv_port_stm32f746_disco#try-it-with-just-a-few-clicks).

qinshijing, 2021-06-01:
hi, I think it is a dazzling feature to support obj zoom.

Miguel0101, 2021-06-01:
I think it would be nice to offer support for a video player.

MouriNaruto, 2021-06-01:
Reference: https://github.com/lvgl/lvgl/issues/2144#issuecomment-814763625, you typed the following paragraph:

> It'd require the add the widgets into a "default group" when they are created and that default group can be attached to a keyboard and/or encoder. Actually, it seems doable. You can set a default group to which interactive widgets are added automatically. **I'll try it today.**

But even in today, I still wait for that.

MouriNaruto, 2021-06-01:
Reference: https://github.com/lvgl/lvgl/issues/2144#issuecomment-814763625, you typed the following paragraph:

> It'd require the add the widgets into a "default group" when they are created and that default group can be attached to a keyboard and/or encoder. Actually, it seems doable. You can set a default group to which interactive widgets are added automatically. **I'll try it today.**

But even in to, I still wait for that.

kisvegabor, 2021-06-01:
> hi, I think it is a dazzling feature to support obj zoom.

It will be possible with `lv_snapshot`. (see in the first comment)

> I think it would be nice to offer support for a video player. For example lv_video

I agree. From LVGL's point of view, it only rendering to a canvas. So the task is rather to find a suitable 3rd party video player lib.
I someone already did something like this, the contribution is very welcome. :slightly_smiling_face: 

> Reference: #2144 (comment), you typed the following paragraph on 7th April 2021:

See this comment from 8th April: https://github.com/lvgl/lvgl/issues/2144#issuecomment-815664186
I really forgot to add it to the docs, but added it just now: https://docs.lvgl.io/8/overview/indev.html#default-group

MouriNaruto, 2021-06-01:
>I really forgot to add it to the docs, but added it just now: https://docs.lvgl.io/8/overview/indev.html#default-group

It will be nice if the LVGL's widget demo can use it. It's necessary for me to continue improve the input experience for the Windows port. (I think some people will complain me that it's not universal if it's only supported in non-LVGL official demo.)

I will try to add IME support recently, and I need some help.
- I need a way to send the utf-32 char to the LVGL input framework.
- I need to know how to get the position which I need to make IME to follow that.

kisvegabor, 2021-06-01:
> It will be nice if the LVGL's widget demo can use it. It's necessary for me to continue improve the input experience for the Windows port. (I think some people will complain me that it's not universal if it's only supported in non-LVGL official demo.)

It's independent of the demo your run because the default group needs should be created and set before creating the demo.

> I will try to add IME support recently, and I need some help.

Please open a new issue for it to discuss the details.

MouriNaruto, 2021-06-01:
> It's independent of the demo your run because the default group needs should be created and set before creating the demo.

I have added the support in the Windows port just now. But I found some issues.

- The virtual keyboard always shows when I use the mouse button or wheel to navigate to the something like text widget.
- How to use the mouse wheel to scroll up or down the content instead of focus to the next widget?

MouriNaruto, 2021-06-01:
> Please open a new issue for it to discuss the details.

https://github.com/lvgl/lvgl/issues/2279

kisvegabor, 2021-06-01:
> The virtual keyboard always shown when I use the mouse button or wheel to navigate to the something like text widget.

It's the intended behaviour. Note that the virtual keyboard is not a built-in accessibility feature (like on Android) but it's created manually in the demo if the text area is clicked. Then the demo checks if the click was coming from a "non-keyboard" input device and open the keyboard.

> How to use the mouse wheel to scroll up or down the content instead of changing the focus to the next widget

The mouse-wheel is emulating an encoder and works differently than it usually does on desktop. However, as your project also shows that LVGL tends to be a viable option for desktop apps too so considering this target seems reasonable. 

Anyway, it needs to be decided that the mouse-wheel is used for encoder emulation or for scrolling.
In the latter case, it still needs to be implemented in LVGL.

Miguel0101, 2021-06-01:
The video player may look like `lv_gif`

kisvegabor, 2021-06-01:
> The video player may look like lv_gif

I also imageined it like that (an external library)

Miguel0101, 2021-06-01:
Later I try to create a video decoder as a third party library.

kisvegabor, 2021-06-01:
> Later I try to create a video decoder as a third party library.

Amazing! Feel free to open a new issue to track if you got started with it.

ajsb85, 2021-06-02:
Hi, @kisvegabor 

Do you have a plan to integrate the rlottie animation widget?

References:
https://github.com/lvgl/lv_lib_rlottie
https://github.com/Samsung/rlottie/issues/484

kisvegabor, 2021-06-02:
Hi @ajsb85,
I've answered [here](https://github.com/lvgl/lv_lib_rlottie/issues/2#issuecomment-853367413).

Widgets that require such a large 3rd party library have their own repo. E.g. [lv_lib_png](https://github.com/lvgl/lv_lib_png).

kisvegabor, 2021-06-03:
I've updated teh following libs to v8:
- freetype
- gif
- qrcode
- bmp
- rlottie
- lv_fs_if

incity, 2021-06-04:
refactor the lv_dropdown to use the compose of lv_obj or lv_btn so that user can custom the user interface of dropdown items

Miguel0101, 2021-06-06:
I'm having problems in version 8.0, it renders, but it crashes and the screen is cluttered.

[image]

kisvegabor, 2021-06-07:
@incity 
A more general dropdown list would be really great. I can imagine it as an ["extra" widget](https://github.com/lvgl/lvgl/tree/master/src/extra/widgets).

@Miguel0101 
Let's discuss it in ontinue in https://forum.lvgl.io/t/screen-freezes-using-lvgl-v8-0/5723

kisvegabor, 2021-06-07:
@embeddedt 
Once we have `docs/8` is there any problem with adding `docs/master` too?

embeddedt, 2021-06-07:
I think it's doable.

kisvegabor, 2021-06-07:
And can you do it if you have some free time for it? :slightly_smiling_face:

embeddedt, 2021-06-07:
I've tried to do it [here](https://github.com/lvgl/lvgl/commit/6d05692d7820a2b833751d6881704b283f1fe618); let's see if it works.

kisvegabor, 2021-06-07:
Works well! I've updated the JS of the dropdown as [you saw](https://github.com/lvgl/lvgl/commit/3cf5751461d6a85974da4e5c66593736ae140a1a). :slightly_smiling_face: 

I had 3 other things for docs in my mind:
>  Display the Micropytohn examples too.

We have mentioned that you are planning to work on it. Shall we open an issue to track it?

> Add a link to the example C and py files

I think it'd be useful to have a show a link to the source files of the example. It'd show from where the examples come from.

> List of all examples on a page. All in iframes grouped by category (e.g. flex, style, button)

If I were new to LVGL I'd love to see a lot of examples in one place. If you already know how it works could you put together a script the puts all example iframes on html page with headers containing the directory name? I'd handle the rest (styling, and othesr) but I have no idea how to put everything on a single page. I meant something like:
```html
<h2>examples/widgets/arc</h2>
<h3>Simple Arc</h3>
<iframe>...</iframe>

<h3>An other arc</h3>
<iframe>...</iframe>

<h2>examples/widgets/btn</h2>
<h3>Simple button</h3>
<iframe>...</iframe>

<h3>An other button</h3>
<iframe>...</iframe>

embeddedt, 2021-06-07:
> We have mentioned that you are planning to work on it. Shall we open an issue to track it?

Sure.

> I think it'd be useful to have a show a link to the source files of the example. It'd show from where the examples come from.

Added. It should be visible in a few minutes.

> If I were new to LVGL I'd love to see a lot of examples in one place. If you already know how it works could you put together a script the puts all example iframes on html page with headers containing the directory name? 

Let me think about it for a bit. I can get the `lv_example` directive to output some HTML to another file (independently of Sphinx) but I'm not sure how to do the grouping automatically.

Another option is to include the `examples/widgets/<widget>/index.rst` files on that central page. That might be easier to group. We'd only have to change it when a widget gets added/removed. If someone added a new example to a widget (and updated the appropriate `.rst` file) it would appear on the central page without needing any manual changes.

kisvegabor, 2021-06-07:
> Another option is to include the examples/widgets/<widget>/index.rst files on that central page. That might be easier to group. We'd only have to change it when a widget gets added/removed. If [...]

My naive approach would be to build the HTML file with a python scrip (list `index.rst` files, search for example titles and file names, etc) in a "post build" event (if there is any) and output a html file.

kisvegabor, 2021-06-07:
> Added. It should be visible in a few minutes.

Great, thanks! I've just added some space before and in the links.

C47D, 2021-06-08:
Is there any update on the Kconfig file expected for v8.1?

FASTSHIFT, 2021-06-08:
I implemented an animation timeline, which can realize the forward and reverse playback of the entire group of animations, simplify the compilation of combined animations, and use them like this:

Demo :

https://user-images.githubusercontent.com/26767803/121167150-3ed02a00-c884-11eb-84e9-b56fb125335e.mp4

```C
static lv_obj_t* obj1 = NULL;
static lv_obj_t* obj2 = NULL;
static lv_obj_t* obj3 = NULL;

static void btn_event_handler(lv_event_t* e)
{
    lv_obj_t* btn = lv_event_get_target(e);

    lv_anim_timeline_t anim_timeline[] = {
        {0,   obj1, (lv_anim_exec_xcb_t)lv_obj_set_width,  0, 200, 300, lv_anim_path_overshoot},
        {0,   obj1, (lv_anim_exec_xcb_t)lv_obj_set_height, 0, 200, 300, lv_anim_path_ease_out},
        {200, obj2, (lv_anim_exec_xcb_t)lv_obj_set_width,  0, 200, 300, lv_anim_path_overshoot},
        {200, obj2, (lv_anim_exec_xcb_t)lv_obj_set_height, 0, 200, 300, lv_anim_path_ease_out},
        {400, obj3, (lv_anim_exec_xcb_t)lv_obj_set_width,  0, 200, 300, lv_anim_path_overshoot},
        {400, obj3, (lv_anim_exec_xcb_t)lv_obj_set_height, 0, 200, 300, lv_anim_path_ease_out},
        LV_ANIM_TIMELINE_END
    };

    bool playback = lv_obj_has_state(btn, LV_STATE_CHECKED);

    lv_anim_timeline_start(anim_timeline, playback);
}

void lv_anim_timeline_demo(void)
{
    lv_obj_t* btn = lv_btn_create(lv_scr_act());
    lv_obj_add_event_cb(btn, btn_event_handler, LV_EVENT_VALUE_CHANGED, NULL);
    lv_obj_add_flag(btn, LV_OBJ_FLAG_IGNORE_LAYOUT);
    lv_obj_add_flag(btn, LV_OBJ_FLAG_CHECKABLE);
    lv_obj_align(btn, LV_ALIGN_BOTTOM_MID, 0, -20);

    lv_obj_set_flex_flow(lv_scr_act(), LV_FLEX_FLOW_ROW);
    lv_obj_set_flex_align(
        lv_scr_act(),
        LV_FLEX_ALIGN_SPACE_AROUND,
        LV_FLEX_ALIGN_CENTER,
        LV_FLEX_ALIGN_CENTER
    );

    obj1 = lv_obj_create(lv_scr_act());
    obj2 = lv_obj_create(lv_scr_act());
    obj3 = lv_obj_create(lv_scr_act());
}
```

Code **lv_anim_timeline.h**:
```C
#ifndef __LV_ANIM_TIMELINE_H
#define __LV_ANIM_TIMELINE_H

#include "lvgl/lvgl.h"

#ifdef __cplusplus
extern "C" {
#endif

typedef struct {
    uint32_t start_time;
    lv_obj_t* obj;

    lv_anim_exec_xcb_t exec_cb;
    int32_t start;
    int32_t end;
    uint16_t duration;
    lv_anim_path_cb_t path_cb;
}lv_anim_timeline_t;

#define LV_ANIM_TIMELINE_END {0, NULL, NULL, 0, 0, 0, NULL}

uint32_t lv_anim_timeline_start(const lv_anim_timeline_t* anim_timeline, bool playback);
void     lv_anim_timeline_del(const lv_anim_timeline_t* anim_timeline);
uint32_t lv_anim_timeline_get_playtime(const lv_anim_timeline_t* anim_timeline);

#ifdef __cplusplus
}
#endif

#endif // ! __LV_ANIM_TIMELINE_H
```

Code **lv_anim_timeline.c**:
```C
#include "lv_anim_timeline.h"

static void lv_anim_timeline_add(const lv_anim_timeline_t* anim_timeline, bool playback, uint32_t playtime)
{
    lv_obj_t* obj = anim_timeline->obj;
    uint32_t delay = anim_timeline->start_time;

    lv_anim_exec_xcb_t exec_cb = anim_timeline->exec_cb;
    uint16_t time = anim_timeline->duration;

    int32_t start = anim_timeline->start;
    int32_t end = anim_timeline->end;
    if (playback)
    {
        int32_t temp = start;
        start = end;
        end = temp;

        delay = playtime - (delay + time);
    }

    lv_anim_t a;
    lv_anim_init(&a);
    lv_anim_set_var(&a, obj);
    lv_anim_set_delay(&a, delay);
    lv_anim_set_exec_cb(&a, exec_cb);
    lv_anim_set_time(&a, time);
    lv_anim_set_values(&a, start, end);
    lv_anim_set_path_cb(&a, anim_timeline->path_cb);

    lv_anim_start(&a);
}

uint32_t lv_anim_timeline_start(const lv_anim_timeline_t* anim_timeline, bool playback)
{
    uint32_t playtime = lv_anim_timeline_get_playtime(anim_timeline);
    uint32_t i = 0;
    while(anim_timeline[i].obj != NULL)
    {
        lv_anim_timeline_add(&(anim_timeline[i]), playback, playtime);
        i++;
    }
    return playtime;
}

void lv_anim_timeline_del(const lv_anim_timeline_t* anim_timeline)
{
    uint32_t i = 0;
    while (anim_timeline[i].obj != NULL)
    {
        lv_obj_t* obj = anim_timeline[i].obj;
        lv_anim_exec_xcb_t exec_xcb = anim_timeline[i].exec_cb;
        lv_anim_del(obj, exec_xcb);
        i++;
    }
}

uint32_t lv_anim_timeline_get_playtime(const lv_anim_timeline_t* anim_timeline)
{
    uint32_t playtime = 0;
    uint32_t i = 0;
    while (anim_timeline[i].obj != NULL)
    {
        uint32_t end = anim_timeline[i].start_time + anim_timeline[i].duration;
        if (end > playtime)
        {
            playtime = end;
        }
        i++;
    }
    return playtime;
}
```

@kisvegabor 
Are you interested in adding this feature to v8.1?

amirgon, 2021-06-08:
> I need to ask @amirgon if it's compatible with the MP binding in the current form.

It's not.

Micropython requirements are [listed here](https://github.com/lvgl/lvgl/issues/1763#issue-689089014).
Some of the problems I see are:
- Missing "user_data" in `lv_anim_timeline_t`. It's required because this struct holds callbacks 
- `exec_cb` is copied around without the accompanying `user_data`.
- lv_anim_start is called without storing its return value. That means these animations cannot be deleted, see [this discussion](https://github.com/lvgl/lvgl/issues/2144#issuecomment-851703533). `lv_anim_del` is incompatible because all callbacks are the same callback in Micropyton, only `user_data` is different, so `lv_anim_timeline_del` would probably not work.
- `LV_ANIM_TIMELINE_END` macro is inaccessible in Micropython

There may be other stuff I missed, need to actually convert the example to Micropython and test it (after the above issues are resolved)

kisvegabor, 2021-06-09:
@FASTSHIFT 
Could you send a PR to see it together and continue the discussion there?

XiaAndragon, 2021-06-10:
I want a new special msg_widget like this,which was group with a obj、icon、text、[close button]、animation

[image]

The website in iviewui.com(https://www.iviewui.com/components/message)

FASTSHIFT, 2021-06-11:
> @FASTSHIFT
> Could you send a PR to see it together and continue the discussion there?

Created here https://github.com/lvgl/lvgl/pull/2309

F5Key, 2021-06-11:
> > @FASTSHIFT
> > Could you send a PR to see it together and continue the discussion there?
> 
> Created here #2309

You can clone the project with 
```shell
git clone https://github.com/lvgl/lv_port_windows.git
```
The info_img is 
[image]

Maybe the gif picture I uploaded was too big so that it's animation cann't be show,you had better visit iviewui.com(https://www.iviewui.com/components/message) to view the actual animation

FASTSHIFT, 2021-06-11:
@xiaAnd

> I want a new special msg_widget like this,which was group with a obj、icon、text、[close button]、animation.
> But animation have not be achieved.
> 
> ```c
> #pragma region msg_widget
>   lv_obj_t *msg_info = lv_obj_create(lv_layer_sys());
>   lv_obj_set_style_border_width(msg_info,0,LV_PART_MAIN);
>   lv_obj_set_style_shadow_width(msg_info,15,LV_PART_MAIN);
>   lv_obj_set_style_shadow_opa(msg_info,LV_OPA_50,LV_PART_MAIN);
>   lv_obj_set_style_pad_all(msg_info,0,LV_PART_MAIN);
>   lv_obj_set_size(msg_info,216,42);
>   lv_obj_center(msg_info);
> 
>   lv_obj_t *info_img = lv_img_create(msg_info);
>   lv_img_set_src(info_img,"./png/info.png");
>   lv_obj_align(info_img,LV_ALIGN_LEFT_MID,10,0);
> 
>   lv_obj_t *info = lv_label_create(msg_info);
>   lv_obj_set_style_text_font(info,dengxian20.font,LV_PART_MAIN);
>   lv_label_set_text(info,"操作成功！");
>   lv_obj_center(info);
> #pragma endregion
> ```
> 
> [image]
> 
> [image]
> 
> The website in iviewui.com(https://www.iviewui.com/components/message)

Can refer to lv_msgbox

FASTSHIFT, 2021-06-11:
> I want a new special msg_widget like this,which was group with a obj、icon、text、[close button]、animation.
> But animation have not be achieved.
>
> ```c
> #pragma region msg_widget
>   lv_obj_t *msg_info = [...]

You can refer to [lv_msgbox](https://github.com/lvgl/lvgl/blob/master/examples/widgets/msgbox/lv_example_msgbox_1.c)

F5Key, 2021-06-11:
> You can refer to [lv_msgbox](https://github.com/lvgl/lvgl/blob/master/examples/widgets/msgbox/lv_example_msgbox_1.c)

## It is diferent from lv_msgbox ##
[image]

FASTSHIFT, 2021-06-11:
> ## It is diferent from lv_msgbox
> [image]

You may need to implement a **msgbox manager** to automatically manage the life cycle of the message box.

fahmimsh, 2021-06-11:
I hope the like a edgeline or nxps application can be used and generated a C program. I'm having trouble creating a ui ux manually
[https://github.com/fahmimsh/ESP32-Satpamtracker](url)

FASTSHIFT, 2021-06-11:
Does lv_switch support animation effects in v8.1?

wangxiao-star, 2021-06-11:
> Does lv_switch support animation effects in v8.1?

Can you speak English ？

embeddedt, 2021-06-11:
> Does lv_switch support animation effects in v8.1?

@FASTSHIFT No, the animation was removed because it added quite a bit of complexity and didn't offer much functional benefit.

> I want a new special msg_widget like this,which was group with a obj、icon、text、[close button]、animation.

@F5Key It sounds like what you're trying to implement is a snackbar/toast widget, which would be a great contribution for the `extras` folder. You will definitely need some type of management logic to track the snackbars and close them automatically.

-------------------------------

Heads up: please try to avoid quoting large screenshots in your replies as it adds more scrolling and makes it easier to miss new comments. :slightly_smiling_face:

F5Key, 2021-06-11:
> > > You can refer to [lv_msgbox](https://github.com/lvgl/lvgl/blob/master/examples/widgets/msgbox/lv_example_msgbox_1.c)
> > 
> > 
> > It is diferent from lv_msgbox
> 
> You may need to implement a **msgbox manager** to automatically manage the life cycle of the message box.

Seems like that.

F5Key, 2021-06-11:
> @F5Key It sounds like what you're trying to implement is a snackbar/toast widget, which would be a great contribution for the `extras` folder. You will definitely need some type of management logic [...]

I'm sorry for that.Thanks for you reply.I think that it need a task to manage message list and the  time they show.Beside, it have a appear animation and disappear animation.The animation most times just callback lv_obj_set_y(obj,v)

FASTSHIFT, 2021-06-11:
> @FASTSHIFT No, the animation was removed because it added quite a bit of complexity and didn't offer much functional benefit.
>
>
> @F5Key It sounds like what you're trying to implement is a [...]

Why is it complicated? Where is the difficulty.

embeddedt, 2021-06-13:
@FASTSHIFT 

In v7, the switch was essentially a slider with some custom toggling logic added. Sliders inherit from bars, which have the ability to be animated the way you describe. However, in v8, the switch was reimplemented as a standalone widget, and thus no longer has the animation feature included.

@kisvegabor 

I experimented with using [`conventional-changelog-cli`](https://github.com/conventional-changelog/conventional-changelog/tree/master/packages/conventional-changelog-cli) to automatically generate a changelog for releases. It does work (see the result [here](https://gist.github.com/embeddedt/554a108070d7cadc35ed1dd1f3d247ea)), but unfortunately, it's not a very concise changelog as it includes every single commit. By the same token, v8 was an extremely large release, so that is contributing to the mess. [7.11.0](https://gist.github.com/embeddedt/554a108070d7cadc35ed1dd1f3d247ea#7110-2021-03-16) looks a lot neater.

It's possible that for major releases we could run this tool first and then manually condense the entries afterward; that might give the best balance of accuracy and simplicity.

kisvegabor, 2021-06-14:
@F5Key 
I like the idea of this kind of message box. Can we call it and the whole system that stacking them as "notifications"? 
I can imagine it as an "extra" widget. Would you interested in implementing it? If you feel free to open a new issue for it.

@embeddedt 
The result of `conventional-changelog-cli` looks quite usable. We need to be more strict to follow the comment message rules (fix, feat, chore, etc). 

I added v8.0.1 locally but with `conventional-changelog -p angular -r 1` returned only 
`# [](https://github.com/lvgl/lv_sim_eclipse_sdl/compare/v8.0.1...v) (2021-06-14)`

So I still don't know how to get the changelog of the last tag.

kisvegabor, 2021-06-14:
For the v8.0.1 release today I'll create the CHANGELOG manually.

kisvegabor, 2021-06-17:
I got a few ideas about examples in the docs:
1. Use a common function to build the examples' html [large example page](https://docs.lvgl.io/master/examples.html) and [examples per topic](https://docs.lvgl.io/master/widgets/core/arc.html).
2. Add a short description to the examples.
3. The large example page seems difficult to manage (slow, out of memory error, etc). I think it'd be better to split it by topic (arc, bar, scroll, etc).
4. Update the design of the example "cards" to something like this:

I can manage all of these except using a common example card creator function and adding an extra description argument to the examples. @embeddedt Could you help with these?

embeddedt, 2021-06-17:
> Use a common function to build the examples' html

If you are referring to the logic which injects the title and iframe, it should be possible to use the `lv_example` RST directive for both. `example_list.py` just needs a minor modification to output directives like [these](https://raw.githubusercontent.com/lvgl/lvgl/master/examples/scroll/index.rst) inside an `eval_rst` code block.

> Add a short description to the examples.

I'll add the ability to specify a description. Do you think it should appear under the title but before the iframe?

> I think it'd be better to split it by topic (arc, bar, scroll, etc).

I'm a bit confused; isn't this the way they are structured now? On the [scroll documentation page](https://docs.lvgl.io/master/overview/scroll.html#examples) we have an examples section, and on [individual widgets' pages](https://docs.lvgl.io/master/widgets/core/arc.html#example) we also have an examples section.

> Update the design of the example "cards" to something like this:

I think your screenshot got lost. :slightly_smiling_face:

kisvegabor, 2021-06-18:
> If you are referring to the logic which injects the title and iframe, it should be possible to use the lv_example RST directive for both. example_list.py just needs a minor modification to output directives like these inside an eval_rst code block.

Good idea!

> I'll add the ability to specify a description. Do you think it should appear under the title but before the iframe?
I think your screenshot got lost. slightly_smiling_face

So the missing screenshot :slightly_smiling_face: 
[image]

> I'm a bit confused; isn't this the way they are structured now? On the scroll documentation page we have an examples section, and on individual widgets' pages we also have an examples section.

Sorry, I meant splitting the l large example page into shorter pages to avoid out-of-memory errors and slow loading. Too many pages would beet the purpose of having the examples in one place but 20-30 examples per page would be a good compromise IMO. However I don't know how to name pages ("Examples 1", "Examples 2", ...) Or pagination?

Miguel0101, 2021-06-18:
Is lvgl based on HTML?

kisvegabor, 2021-06-18:
> Is lvgl based on HTML?

How do you mean it?

fvanroie, 2021-06-18:
@Miguel0101 The *live* C and MicroPython examples in the documentation are based on WASM (WebAssembly). It's a way to run compiled C code in a web browser frame. What @embeddedt and @kisvegabor are discussing is a way to conveniently include that iframe in the documentation by adding meta data to the code to render the docs, I believe...

kisvegabor, 2021-06-18:
>  What @embeddedt and @kisvegabor are discussing is a way to conveniently include that iframe in the documentation by adding meta data to the code to render the docs, I believe...

Yes, that' it :slightly_smiling_face:

Miguel0101, 2021-06-18:
I think it would be nice to add a multiple choice widget.

embeddedt, 2021-06-18:
> So the missing screenshot

That's much nicer than what I managed to come up with. :laughing: That should be easy to implement.

embeddedt, 2021-06-18:
@kisvegabor Could you please test the performance [here](https://docs.lvgl.io/8.1/examples.html) and let me know if it's improved? The out of memory error seems to be fixed.

Descriptions can also be specified [in the RST file](https://github.com/lvgl/lvgl/blob/master/examples/widgets/arc/index.rst) now: https://docs.lvgl.io/8.1/widgets/core/arc.html#simple-arc

kisvegabor, 2021-06-18:
@embeddedt 
Wow, that's amazing :heart_eyes: 
Only a minor thing: would it be possible to "preload" the next 2-3 examples and keep the previous 2-3?

In the `rst` files the order of parameters are fixed? (`:language` and `:description`)

[62 more replies]