lvgl/lvgl #2278: lvgl does not build with cmake

embexus, 2021-06-01:
lvgl does not build with cmake :
```
$ cmake .
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.20)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
-- Build files have been written to: /home/zaki/Workspace/Playground/lvgl

$ make
[  0%] Building C object CMakeFiles/lvgl.dir/src/core/lv_disp.o
In file included from /home/zaki/Workspace/Playground/lvgl/src/core/../hal/../misc/lv_color.h:16:0,
                 from /home/zaki/Workspace/Playground/lvgl/src/core/../hal/lv_hal_disp.h:21,
                 from /home/zaki/Workspace/Playground/lvgl/src/core/../hal/lv_hal.h:16,
                 from /home/zaki/Workspace/Playground/lvgl/src/core/lv_disp.h:16,
                 from /home/zaki/Workspace/Playground/lvgl/src/core/lv_disp.c:9:
/home/zaki/Workspace/Playground/lvgl/src/core/../hal/../misc/../lv_conf_internal.h:41:14: fatal error: ../../lv_conf.h: No such file or directory
 #    include "../../lv_conf.h"                 /*Else assume lv_conf.h is next to the lvgl folder*/
              ^~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/lvgl.dir/build.make:75: recipe for target 'CMakeFiles/lvgl.dir/src/core/lv_disp.o' failed
make[2]: *** [CMakeFiles/lvgl.dir/src/core/lv_disp.o] Error 1
CMakeFiles/Makefile2:82: recipe for target 'CMakeFiles/lvgl.dir/all' failed
make[1]: *** [CMakeFiles/lvgl.dir/all] Error 2
Makefile:90: recipe for target 'all' failed
make: *** [all] Error 2
```
would be also possible to add automated build on github ?

kisvegabor, 2021-06-01:
Hi,

See this project for a CMake example: https://github.com/lvgl/lv_sim_eclipse_sdl
I've just rested and it works well.

Also, see this porting guide: https://docs.lvgl.io/8.0/get-started/quick-overview.html#add-lvgl-into-your-project

There is no build guides for lvgl itself because lvgl can't be built on its own. It should be added into your project and build it along with the other source files of your poject.

kisvegabor, 2021-06-03:
> wouldn't be possible to build and provide lvgl as library [ static and | or shared] ?

The question is what targets would be important. I think there are too many. Linux, Windows, Mac, ARM variants, NXP, STM, PIC, ATMEL, Raspberry related, etc.

embexus, 2021-06-03:
I don't mean to provide it as binaries but to make the build able to generate shared | static  library an headers to be installed and used as most libraries do.

kisvegabor, 2021-06-03:
Maybe I know something worng but `.a` and `.lib` files are platform-specific. (See the [StackOverflow answer](https://stackoverflow.com/a/3120328))

embeddedt, 2021-06-03:
I think @embexus is suggesting that the CMakeLists file should include the ability to produce a `.a` file, not that we should provide binaries.

This feature is already present:

https://github.com/lvgl/lvgl/blob/38ebcd8154459f58ccc0e6b44bd8185372a14417/CMakeLists.txt#L62-L67

The issue is that there needs to be an `lv_conf.h` somewhere in the compile path, as the way you want the library configured will depend on your project. So it's not really possible to build a 100% standalone .a file in all cases.

closed 2021-06-26 by lvgl-bot