lvgl/lvgl #2279: IME support for LVGL

MouriNaruto, 2021-06-01:
References
- https://github.com/lvgl/lvgl/issues/2276#issuecomment-852338512
- https://github.com/lvgl/lvgl/issues/2276#issuecomment-852342841

I will try to add IME support recently, and I need some help.
- I need a way to send the utf-32 char to the LVGL input framework.
- I need to know how to get the position which I need to make IME to follow that.

kisvegabor, 2021-06-01:
LVGL supports only UTF-8 but converting from UTF-32 to UTF-8 is quite straightforward.

> I need a way to send the utf-32 char to the LVGL input framework.

From the keyboard's input device you can simply return the character in UTF-8. But a minor fix is required in text area [here](https://github.com/lvgl/lvgl/blob/master/src/widgets/lv_textarea.c#L894). `lv_textarea_add_char` needs to be changed to `lv_textarea_add_text` and the key (the characters) needs to be converted to a string. In the most simply case it's:
```c
uint32_t c2[2] {c, 0};
lv_textarea_add_text(obj, (const char *)c2);
```

> I need to know how to get the position which I need to make IME to follow that.

How do you mean it?

MouriNaruto, 2021-06-02:
> How do you mean it?

I need to make the IME candidate box positioned at the text caret instead of the top left of the screen. So I need to know the position of the text caret.

[image]

MouriNaruto, 2021-06-02:
Just add the simple IME support.

https://user-images.githubusercontent.com/10867563/120468758-27042c00-c3d4-11eb-8d43-ea690912a55b.mp4

kisvegabor, 2021-06-02:
Cool!

There is no API to get cursor positon but it should work:
```c
    lv_obj_t * obj = lv_textarea_create(lv_scr_act());
    lv_textarea_t * ta = (lv_textarea_t *) obj;
    lv_obj_t * label = lv_textarea_get_label(obj);
    lv_coord_t cx = label->coords.x1 + ta->cursor.area.x1;
    lv_coord_t cy = label->coords.y1 + ta->cursor.area.y1;
    printf("%d,%d\n", cx, cy);
```

The "IME box" is created by Windows or you with LVGL?

MouriNaruto, 2021-06-02:
> The "IME box" is created by Windows or you with LVGL?

The "IME box" is created by Windows. But we can set the position for that.

MouriNaruto, 2021-06-02:
And I wonder to know how to get the active text area.

kisvegabor, 2021-06-03:
There is no explicit way t get the active text area but you can do this:
```c
  lv_group_t * g = lv_group_get_default();
  lv_obj_t * obj = lv_group_get_focused(g);
  lv_obj_t * ta = NULL;

  if(obj) {
      if(lv_obj_get_class(obj) == &lv_textarea_class) ta = obj;
      else if(lv_obj_get_class(obj) == &lv_keyboard_class) ta = lv_keyboard_get_textarea(obj);
  }
```

WShiBin, 2021-06-09:
I have the same feature request with that. the final is change some source code of keyboard,

demo:

https://user-images.githubusercontent.com/18117928/121283735-345a7280-c90e-11eb-9bde-49da21cd3a46.mp4

MouriNaruto, 2021-06-09:
@WShiBin Looks really nice about that.

@kisvegabor I have implement the Windows IME support for the Windows port. But I met some issues when I send the char to LVGL via keypad indev driver.

Here is my keypad indev callback. https://github.com/lvgl/lv_port_windows/blob/b54aef380187698e8e26f75941312e40cba60ea1/LVGL.Windows.Desktop/LVGL.Windows.Desktop.cpp#L156

kisvegabor, 2021-06-10:
@WShiBin 
That's great! Would you be interested in sending a PR with the new keyboard? As a new widget next to to the normal keyboard.

@MouriNaruto 
>  But I met some issues when I send the char to LVGL via keypad indev driver.

What's the issue?

MouriNaruto, 2021-06-10:
@kisvegabor 

> What's the issue?

I don't which is the proper value of `data->state` in the indev keypad read callback I should been sent after I got the CJK character. (I have used the indev because I don't know `lv_textarea_add_text` is or isn't designed for all places which can accept CJK characters in LVGL.)

kisvegabor, 2021-06-10:
Basically, all you need is to set `data->key` and `data->state = LV_INDEV_STATE_PRESSED` and in the next call `data->state = LV_INDEV_STATE_RELEASED`.

You can set `data->continue_reading = true` to make LVGL call read_cb again. It can be useful when setting `LV_INDEV_STATE_PRESSED` to allow setting `LV_INDEV_STATE_RELEASED` immediatelly.

kisvegabor, 2021-06-17:
@MouriNaruto Are there any new issues with IME support?

MouriNaruto, 2021-06-17:
@kisvegabor 

> Basically, all you need is to set data->key and data->state = LV_INDEV_STATE_PRESSED and in the next call data->state = LV_INDEV_STATE_RELEASED.

I have used the queue to achieve that.

> Are there any new issues with IME support?

Yes. It's OK to type something with IME. But when I insert something, Something weird happened. I need to check that and try to fix or post here.

MouriNaruto, 2021-06-17:
@kisvegabor I have found the issue. So I need to know something.

How to make the indev know that we need to skip the current result in the indev read callback?

kisvegabor, 2021-06-17:
>  I have achieved that via “don't modify the last status”

Great!

MouriNaruto, 2021-06-17:
@kisvegabor The implementation of the IME support for LVGL's Windows port has been completed basically. I hope you will like it.

https://user-images.githubusercontent.com/10867563/122409735-f6b4b400-cfb5-11eb-95e9-160a0e7cb32d.mp4

[image]
P.S. The implementation supports UTF-16's supplementary planes.

I have pushed these implementations to lv_port_windows repo.

kisvegabor, 2021-06-18:
I do like it! :slightly_smiling_face: 
I'll share it on Linked In as well.

kisvegabor, 2021-06-29:
Sorry for the delay, I just shared it: https://www.linkedin.com/feed/update/urn:li:activity:6815548036668764160/

Is there anything else to do with the IME support or can it be considered ready?

MouriNaruto, 2021-06-29:
@kisvegabor 

IME support for Windows port can be considered ready. But @FASTSHIFT suggests me to implement the builtin IME for LVGL for bring the CJK input in all scenarios. So this issue still need to be opened.

Feng573795502, 2021-06-29:
> 我有相同的功能请求。最后是更改键盘的一些源代码，
> 
> 演示：
> 
>  VID_20210408_101813_1.mp4

Hello this has a general idea? Or demo, I also want to implement a T9 input method

kisvegabor, 2021-06-29:
> But @FASTSHIFT suggests me to implement the builtin IME for LVGL for bring the CJK input in all scenarios. So this issue still need to be opened.

Can you describe the technical requirements? E.g. does it require a large (how large) dictionary of ASCII letters-CJK characters?

MouriNaruto, 2021-06-30:
@kisvegabor 

> Can you describe the technical requirements?

It needs really large space to storage CJK font and the dictionary of ASCII letters to CJK characters.
I am planning to use FreeType to render the CJK characters.

kisvegabor, 2021-07-01:
> It needs really large space to storage CJK font and the dictionary of ASCII letters to CJK characters.
I am planning to use FreeType to render the CJK characters.

Once I collected the most common CJK characters. The `LV_FONT_SIMSUN_16_CJK` is built from these characters.
It can be also useful.

> Also, I need the way to extend (add input mode switch button, show the candidate box and etc) the LVGL virtual keyboard.

I think it's 2 different thing:
1. Build an extension to the keyboard where any words can be displayed (T9, IME, etc)
2. Create the "IME engine" that feeds the words into that area

The input method switch button is just an other button on the keyboard?

MouriNaruto, 2021-07-02:
> The input method switch button is just an other button on the keyboard?

Yes, I think it belongs to the `extension to the keyboard` your mentioned.

> Once I collected the most common CJK characters. The LV_FONT_SIMSUN_16_CJK is built from these characters. It can be also useful.

I will try it because its size is really small.

kisvegabor, 2021-07-19:
It seems the IME support itself is finished so I close this issue. 

For the keyboard extension feel free to open a new issue to discuss it separately.

closed 2021-07-19 by kisvegabor