lvgl/lvgl #2282: Test engine inegration

kisvegabor, 2021-06-02:
I made some experiments with the Unity test engine [here](https://github.com/lvgl/lvgl/tree/feat/test-unity).

It works like this:
- tests are craeted in `test_<name>.c` files. Like [this](https://github.com/lvgl/lvgl/blob/feat/test-unity/tests/test_obj_tree.c).
- a python script collects the `test_<name>.c` files
- a ruby script creates `test_<name>_Runner.c` files that actually calls the test functions
- a python script builds lvgl with `test_<name>_Runner.c` 
- run the executable and see the test result

You can try it out by running `main.py` in the `lvgl/test` fodler.

embeddedt, 2021-06-13:
Looks promising! I have a few suggestions:

* Can the compiled object files be put into a separate `objs` folder? Currently they clutter the main folder and make it hard to find the actual test case files.
* Perhaps the test cases should be put into their own folder so they are separated from the test runner itself.

It's probably worth looking into something like [gcov](https://gcc.gnu.org/onlinedocs/gcc/Gcov.html) for code coverage, so that we can see exactly how much of the library is being tested.

I use a combination of unit tests and coverage for the backend portion of a web project I work on. it has saved me many hours of debugging because if the tests pass, I know that the backend will continue working the way it did before, and I can run the tests many times during the development of a new feature to confirm that my most recent change didn't break anything.

kisvegabor, 2021-06-17:
> Can the compiled object files be put into a separate objs folder? Currently they clutter the main folder and make it hard to find the actual test case files.

I've just tried but somehow `$(OBJDIR)/%.o: %.c` is not working. 

> Perhaps the test cases should be put into their own folder so they are separated from the test runner itself.

Done. They are moved to `src/test_cases` and `src/test_runners` folders.

> It's probably worth looking into something like gcov for code coverage, so that we can see exactly how much of the library is being tested.

I've tried it out and it worked very well. `--coverage` needs to be added C and LD flags and then `gcov lv_obj.h` show the statistics. I don't know how to aggregate the results from many files and many tests into a single percentage value yet.

kisvegabor, 2021-06-17:
I've found gcovr that can collect the result into text, xml, and html formats.
This is the detailed version of the html output.  [coverage.zip](https://github.com/lvgl/lvgl/files/6672372/coverage.zip)

I've also found [codecov](https://github.com/codecov/codecov-action) which is a GitHub action to collect the test result and publish them in some form. (Not fully clear yet). But with this, we can have badges from [here](https://shields.io/category/coverage). :slightly_smiling_face:

embeddedt, 2021-06-17:
Amazing! That's pretty much exactly what I was thinking of. It makes it easy to see what parts of the library are still untested.

kisvegabor, 2021-06-18:
I've pushed the changes to `feat/test-unity`. You can try it with `./main.py report`

kisvegabor, 2021-06-18:
The next step is to add an indev emulator with features like:
- move_to(x, y, time);
- press(time)
- release (time)
- wait(time)

embeddedt, 2021-06-18:
To fix the object directory issue, perhaps we should just move to using CMake. I can make a CMakeLists file for the tests if you like.

kisvegabor, 2021-06-20:
> I can make a CMakeLists file for the tests if you like.

It would be great! :slightly_smiling_face:

embeddedt, 2021-06-23:
I'm still working on the CMake file. It's not behaving exactly the same way that Make does when passing source file names and defines through variables from `build.py`, which is causing issues.

Part of me is wondering if I should just fix the Makefile instead. :laughing: Using CMake brings other advantages, though, like being able to compile LVGL using the CMakeLists file in the root directory.

kisvegabor, 2021-06-24:
> I'm still working on the CMake file. It's not behaving exactly the same way that Make does when passing source file names and defines through variables from build.py, which is causing issues.

What is the issue? 

Meanwhile, I made it work with the CI too. See https://github.com/lvgl/lvgl/tree/feat/test-unity
All the tests running currently on master are working and I added a unit test for the dropdown as proof of concept. 

If you agree we can merge it to master.

embeddedt, 2021-06-26:
> What is the issue?

When there's multiple source file paths in a variable, it seems to be interpreting the entire variable as one file with spaces in its name, instead of treating each chunk as a separate file.

> If you agree we can merge it to master.

Yes; let's merge it and we can make further improvements on `master` directly.

kisvegabor, 2021-06-28:
> When there's multiple source file paths in a variable, it seems to be interpreting the entire variable as one file with spaces in its name, instead of treating each chunk as a separate file.

Just an idea: "new line" separation instead of space?

> Yes; let's merge it and we can make further improvements on master directly.

Done

embeddedt, 2021-07-04:
I've looked at it again; there is quite a bit that would need to be changed to work with the CMake directory layout. Since Make works well enough for our needs, let's just stick with it for now.

The next step is probably to figure out a way of getting the coverage report uploaded to a service like Coveralls or Codecov. I think they are free for open-source repositories.

kisvegabor, 2021-07-05:
> I've looked at it again; there is quite a bit that would need to be changed to work with the CMake directory layout. Since Make works well enough for our needs, let's just stick with it for now.

Okay.

> The next step is probably to figure out a way of getting the coverage report uploaded to a service like Coveralls or Codecov. I think they are free for open-source repositories.

It seems like some GitHub action magic. Do you plan to give it a try? If you don't have time for it I can try it too.

embeddedt, 2021-07-05:
> It seems like some GitHub action magic. Do you plan to give it a try?

embeddedt, 2021-07-05:
Sure; I can give it a try. I just wanted to make sure you were fine with it before I set it all up. :slightly_smiling_face:

embeddedt, 2021-07-05:
@kisvegabor Could you please check for me whether you can see the contents of `lv_checkbox.c` [at this link](https://codecov.io/gh/lvgl/lvgl/src/master/src/widgets/lv_checkbox.c) when you are not logged in to Codecov? I get a "GitHub API: Forbidden" error.

kisvegabor, 2021-07-06:
I get this error. (Altough the path looks good) :slightly_frowning_face: 
[image]

kisvegabor, 2021-07-06:
Here it works: https://codecov.io/gh/lvgl/lvgl/src/e3024032dc5de2ece4fa17059ebad4189a5fa670/src/widgets/lv_checkbox.c

I got here from the "Details" of the failed actions:
[image]

kisvegabor, 2021-07-06:
Ah, the second was working because meanwhile I was logged in.

kisvegabor, 2021-07-06:
But it's working in incognito window too:
https://codecov.io/gh/lvgl/lvgl/branch/master/graph/badge.svg

kisvegabor, 2021-07-06:
Now [your link](https://codecov.io/gh/lvgl/lvgl/src/master/src/widgets/lv_checkbox.c) is working even if I logged out, or in an incognito window, or on a colleague machine.

kisvegabor, 2021-07-06:
Wow, what a [nice feature](https://github.com/lvgl/lvgl/pull/2353#issuecomment-874730002)!

embeddedt, 2021-07-06:
Thanks for the testing. I suspect that for anonymous users, there is a delay between a commit being pushed on our end, and their app being able to read the file contents of that commit.

I'm not sure why this is the case, given that as a human, you can instantly see source code on GitHub without needing to be logged in.

> Wow, what a nice feature!

Yes. :laughing: For now I'd like to turn it off, though, as I don't think we have enough coverage of existing features to be requiring tests for new ones.

kisvegabor, 2021-07-06:
> Yes. laughing For now I'd like to turn it off, though, as I don't think we have enough coverage of existing features to be requiring tests for new ones.

I agree and wouldn't advertise the badge with 26% coverage either :laughing:

kisvegabor, 2021-07-21:
I think we are good with this now. Unity is used and code coverage is reported. The next task is to [add tests](https://github.com/lvgl/lvgl/issues/2337).

Closing.

closed 2021-07-21 by kisvegabor