lvgl/lvgl #2283: Image zoom causes the widget size to be incorrect

FASTSHIFT, 2021-06-03:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When the zoom is adjusted, the size of the widget set does not match the actual size.

### To Reproduce

```c
void img_test()
{
    lv_obj_t* obj = lv_obj_create(lv_scr_act());
    lv_obj_set_size(obj, 300, 300);

    lv_obj_t* img = lv_img_create(lv_scr_act());
    lv_img_set_src(img, &img_clothes);

    lv_obj_set_style_bg_color(img, lv_color_black(), 0);
    lv_obj_set_style_bg_opa(img, LV_OPA_COVER, 0);
    lv_img_set_pivot(img, 0, 0);
    lv_img_set_zoom(img, 128);
    lv_obj_set_size(img, 300, 300);
}
```

### Expected behavior

When the zoom is adjusted, the size of the image widget set should be the same as the actual size.

### Screenshots or video

[image]

embeddedt, 2021-06-03:
As far as I know, this was originally intended behavior, though definitely not intuitive.

v8 uses a different sizing system, so it's possible that we can adjust the behavior, but it would likely need to be behind a style property.

XuNeo, 2021-06-03:
Hi @embeddedt,

I'm not sure if I understand correctly, but this intended behavior leads to some trouble.
For us, the situation is that three icons in a row, contained in a parent object to allow automatical layout. The zoom scale will change with its y coordinates to make a fancy look on round screen.

The default behavior returns wrong object size, thus makes auto layout failed. If we are going to set the object size manually without auto layout, then the image scale factor will affect the final object size --> not the size manually set.

The proposed solution is fully commited with meaning of LV_SIZE_CONTENT that if the image scale changes, the object size changes with its content. And if object size is set manually, then size stays. Although there are some changes i'm not fully sure if it's correct, thus needs a carefull review.

The tile behaviour is also changed that the image is tiled one after another using zoomed picture size, not the oject size as show in above picture.

Regards,
Neo

FASTSHIFT, 2021-06-06:
@kisvegabor 
What do you think about this?

kisvegabor, 2021-06-07:
Sorry for the late reply.

I agree that this part can be reconsidered. However, the current mechanism is quite similar to HTML/CSS in which scale [doesn't affect the document flow](https://stackoverflow.com/questions/14386948/how-does-css-scale-transform-affect-document-flow).

Let's say there are 3 images in a row and when you press one of them it gets 10% larger, If the elements were positioned by felx layout the other 2 images would "move" which is probably not desirable.

FASTSHIFT, 2021-06-08:
@kisvegabor 
Thanks for the reply, I think it should be automatically set the width and height of the widget according to the zoom when the user does not specify the width and height. If the user specifies the width and height, the widget will be subject to the set width and height.

kisvegabor, 2021-06-09:
I think in most cases the users don't set the size of the images but let the default size that is set to the size of the image source.  And in this case, zooming will result in probably unwanted layout recalculation. 

Maybe I don't understand the underlying problem well enough, but why isn't it causes issues on the web?

XuNeo, 2021-06-10:
Hi kisvegabor, 

> I think in most cases the users don't set the size of the images but let the default size that is set to the size of the image source. And in this case, zooming will result in probably unwanted layout recalculation.

Yes, in most cases user won't set image object size, but one they did, it results to unexpected object size.
Check the picture above, that the image object size is set to 300x300. Another object with white background is also set to 300x300 as a reference. So it's apparent that image object size looks like only 150x150.

To avoid unwanted layout recalculation, then the object size should be set to image size, not LV_SIZE_CONTENT if I understand correctly. Once size is manually set, then it stays.

Regards,
Neo

kisvegabor, 2021-06-10:
I see your point the issue I mentioned still persists. I created some demos to illustrate it.

Applying your modifications from #2284 and some minor fixes:
```c
LV_IMG_DECLARE(img_clothes);

lv_obj_t* obj = lv_obj_create(lv_scr_act());
lv_obj_set_size(obj, 300, 300);
lv_obj_set_flex_flow(obj, LV_FLEX_FLOW_ROW_WRAP);
lv_obj_set_flex_align(obj, LV_FLEX_ALIGN_SPACE_EVENLY, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);

lv_obj_t* img;
int i;
for(i = 0; i < 9; i++) {
  img = lv_img_create(obj);
  lv_img_set_src(img, &img_clothes);
  lv_obj_set_style_bg_color(img, lv_color_black(), 0);
  lv_obj_set_style_bg_opa(img, LV_OPA_COVER, 0);
  lv_img_set_pivot(img, 0, 0);
  lv_obj_set_style_transform_zoom(img, 400, LV_STATE_PRESSED);
  lv_obj_add_flag(img, LV_OBJ_FLAG_CLICKABLE);
  //lv_obj_set_size(img, 56, 56);
}
```

[image]

If I set the size explicitly: `lv_obj_set_size(img, 56, 56);`:
[image]

And this is how it works now. I commented out `lv_img_set_pivot(img, 0, 0);`:
[image]

The first 2 are probably not what the user wants if he creates a numpad.

I think we should keep the "soft size" to not readjust the layout but still set the size correctly as in your PR.

XuNeo, 2021-06-14:
Thanks kisvegabor, for creating all thoses tests. I agree that numpad usecase won't want behaviour in first 2 pictures.

However, the usecase below requires layout readjusted. Maybe the best method is to let user decides the behaviour? Or create a custom widget based on lv_img to manipulate object size. What do you think?

[image]

kisvegabor, 2021-06-14:
> Maybe the best method is to let user decides the behaviour?

As both methods are useful, I agree. What about adding a function like `lv_img_set_zoom_mode(img, LV_IMG_ZOOM_MODE_REAL/VIRTUAL)`? (or with better names)

kisvegabor, 2021-06-20:
There is a 3rd mode too when the zoom is set automatically to the object's width and height. E.g. if you set  a 100x100 image to 200x200 the zoom is automatically set to 512.

XuNeo, 2021-06-22:
I will try to implement those zoom modes next week.
I'm not really sure, but If there is need for the 3rd mode, do we need to add 4th mode that images should be tiled if object size is larger than original image size. Does that make sense to add all those complicated modes? Is there any other solution?

kisvegabor, 2021-06-22:
So the questions are what to do when the size of the object is different from the size of the image, and what should be a relation between the object size and zoom.

I've checked the related properties in CSS:
- [background-repeat](https://css-tricks.com/almanac/properties/b/background-repeat/)
- [object-fit](https://css-tricks.com/almanac/properties/o/object-fit/)

I think it's a little bit incontinent mainly because the `<img>` element can not be tiled.

So here is my proposal based on CSS and the ideas in the issue.

Add an `image_size_mode` attribute (need a better name) that describes how the image should behave if the image size and object size is different:
- `none` just draw the image as it is in the top left corner
- `center` same as `none` just center the image
- `repeat` tile the image on the set size
- `zoom` zoom the image to the set object size

This way you can zoom as described in the issue by setting the object size instead of `lv_img_set_zoom` and selecting `image_size_mode = zoom`.

`lv_img_set_zoom` would still change the size only "virtually" (backward compatible with LVGL and compatible with CSS)

It also maintains backward compatibility if `image_size_mode = repeat` by default.

XuNeo, 2021-07-05:
Hi @kisvegabor ,
Sorry for the late reply, I have updated the pull request, but using a different mode called obj_size_mode.
- LV_IMG_OBJ_SIZE_MODE_ORIGINAL is the default behavior, object SIZE_CONTENT means original image size.
- LV_IMG_OBJ_SIZE_MODE_ZOOMED is newly added mode, the object size changes with image transformed size if object size is set to SIZE_CONTENT. If object size is manually set, then object size stays.

The main reason I use this method is that lots of existing API won't work if use `image_size_mode`, such as `lv_obj_set_style_transform_zoom`  thus leads to more confusion.

I used below examples to test the various combinations. Do you think it's acceptable?

[image]

```c

LV_IMG_DECLARE(img_clothes);

void img_callback(lv_event_t* e)
{
    lv_obj_t * obj = lv_event_get_target(e);
    lv_event_send(obj->parent, LV_EVENT_SIZE_CHANGED, obj);
}

void img_test0(lv_obj_t* parent)
{
    lv_obj_t* obj = lv_obj_create(parent);
    lv_obj_set_size(obj, 180, 180);
    lv_obj_set_flex_flow(obj, LV_FLEX_FLOW_ROW_WRAP);
    lv_obj_set_flex_align(obj, LV_FLEX_ALIGN_SPACE_EVENLY, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);

    lv_obj_t* img;
    int i;
    for(i = 0; i < 4; i++) {
        img = lv_img_create(obj);
        lv_img_set_src(img, &img_clothes);
        lv_obj_set_style_bg_color(img, lv_color_black(), 0);
        lv_obj_set_style_bg_opa(img, LV_OPA_COVER, 0);
        // lv_img_set_pivot(img, 0, 0);
        lv_obj_set_style_transform_zoom(img, 400, LV_STATE_PRESSED);
        lv_obj_add_flag(img, LV_OBJ_FLAG_CLICKABLE);
        //lv_obj_set_size(img, 56, 56);
    }
}

void img_test1(lv_obj_t* parent)
{
    lv_obj_t* obj = lv_obj_create(parent);
    lv_obj_set_size(obj, 250, 250);
    lv_obj_set_flex_flow(obj, LV_FLEX_FLOW_ROW_WRAP);
    lv_obj_set_flex_align(obj, LV_FLEX_ALIGN_SPACE_EVENLY, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);

    lv_obj_t* img;
    int i;
    for(i = 0; i < 4; i++) {
        img = lv_img_create(obj);
        lv_img_set_src(img, &img_clothes);
        lv_obj_set_style_bg_color(img, lv_color_black(), 0);
        lv_obj_set_style_bg_opa(img, LV_OPA_COVER, 0);
        // lv_img_set_pivot(img, 0, 0);
        lv_obj_add_event_cb(img, img_callback, LV_EVENT_PRESSED, obj);
        lv_obj_add_event_cb(img, img_callback, LV_EVENT_RELEASED, obj);
        // lv_obj_add_event_cb(img, img_callback, LV_EVENT_RELEASED, obj);
        lv_obj_set_style_transform_zoom(img, 400, LV_STATE_PRESSED);
        lv_obj_add_flag(img, LV_OBJ_FLAG_CLICKABLE);
        lv_img_set_angle(img, 450);
        lv_img_set_content_size_mode(img, LV_IMG_OBJ_SIZE_MODE_ZOOMED);
    }
}

void img_test2(lv_obj_t* parent)
{
    lv_obj_t* obj = lv_obj_create(parent);
    lv_obj_set_size(obj, 300, 300);
    lv_obj_set_flex_flow(obj, LV_FLEX_FLOW_ROW_WRAP);
    lv_obj_set_flex_align(obj, LV_FLEX_ALIGN_SPACE_EVENLY, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);

    lv_obj_t* img;
    int i;
    for(i = 0; i < 4; i++) {
        img = lv_img_create(obj);
        lv_img_set_src(img, &img_clothes);
        lv_obj_set_style_bg_color(img, lv_color_black(), 0);
        lv_obj_set_style_bg_opa(img, LV_OPA_COVER, 0);
        // lv_img_set_pivot(img, 0, 0);
        lv_obj_add_event_cb(img, img_callback, LV_EVENT_PRESSED, obj);
        lv_obj_add_event_cb(img, img_callback, LV_EVENT_RELEASED, obj);
        // lv_obj_add_event_cb(img, img_callback, LV_EVENT_RELEASED, obj);
        lv_obj_set_style_transform_zoom(img, 128, LV_STATE_PRESSED);
        lv_obj_add_flag(img, LV_OBJ_FLAG_CLICKABLE);
        lv_img_set_angle(img, 450);
        lv_obj_set_size(img, 100, 100);
        lv_img_set_content_size_mode(img, LV_IMG_OBJ_SIZE_MODE_ZOOMED);
    }
}

void img_test()
{
    lv_obj_t* obj = lv_obj_create(lv_scr_act());
    lv_obj_set_size(obj, 800, 480);
    lv_obj_set_flex_flow(obj, LV_FLEX_FLOW_ROW_WRAP);
    lv_obj_set_flex_align(obj, LV_FLEX_ALIGN_SPACE_EVENLY, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);

    img_test0(obj);
    img_test1(obj);
    img_test2(obj);
}
```

kisvegabor, 2021-07-08:
Thanks, I commented in the PR.

XuNeo, 2021-07-18:
Issue solved in #2284 .

kisvegabor, 2021-07-19:
Closing, thanks!

closed 2021-07-19 by kisvegabor