lvgl/lvgl #2288: lv_gauge_set_critical_value can't work correctly

xxxbbbooo, 2021-06-04:
### Perform all steps below and tick them with [x]
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

lv_gauge_set_critical_value can't work correctly

### To Reproduce

    m_rpmGauge = lv_gauge_create(lv_scr_act(), NULL);
    lv_obj_set_size(m_rpmGauge, GUAGE_SIZE, GUAGE_SIZE);
    lv_obj_align(m_rpmGauge, lv_scr_act(), LV_ALIGN_IN_RIGHT_MID, -40, 0);
    lv_gauge_set_scale(m_rpmGauge, 270, 25, 9);
    lv_gauge_set_range(m_rpmGauge, 0, 8);
    lv_gauge_set_critical_value(m_rpmGauge, 7);

### Expected behavior

color changes from correct place
[image]

### Screenshots or video

kisvegabor, 2021-06-07:
Thanks, it already caused issues several times.  We will take a look.

BTW, have you considered using v8? [lv_meter](https://docs.lvgl.io/8.0/widgets/extra/meter.html) is much more versatile then `lv_gauge`.

kisvegabor, 2021-06-17:
I've pushed a fix.

xxxbbbooo, 2021-06-21:
thanks for your fixing

closed 2021-06-21 by xxxbbbooo

fvanroie, 2021-07-06:
The fix seems to have an unintended consequence where the last major tick of the critical scale is always drawn.
[image]

Previous to the [last commit](https://github.com/lvgl/lvgl/commit/042e34cb880618ee8ba180a642697ce4653bd2af), you could use`lv_gauge_set_critical_value` with `value` >= `max` and the critical scale would not draw:
[image]

I've tracked it down to [`ext->lmeter.lvl_ofs = -1;`](https://github.com/lvgl/lvgl/blob/042e34cb880618ee8ba180a642697ce4653bd2af/src/lv_widgets/lv_gauge.c#L83) in `lv_gauge.c`. When I set this to `0` the last major tick of the critical scale is not drawn anymore and it works as before. I'm stuck trying to understand what is the purpose of `lvl_ofs` and how it could be fixed...

@kisvegabor is this expected behavior and could you give me any pointer about what may be the cause?

kisvegabor, 2021-07-08:
I've tried to fix it but the way gauge and critical value works in v7 are a little bit hacky and it seems difficult to find a good fix. 

The problem is that if you set a greater critical value than the max value LVGL will limit the critical value to max. So, for example, it will be 30 on your image. But critical value = 30 means to make the line at 30 colored.

I'm thinking about allowing to set larger critical values than the max value and limit it later during drawing.

But the line meter's "normal value" is used for the gauge's critical value. So finally the value limitation should be removed from the line meter. 

What do you think?

fvanroie, 2021-07-12:
> I'm thinking about allowing to set larger critical values than the max value and limit it later during drawing.

I fully agree with this statement. I trust your judgement as I'm not fully understanding what's going on under the hood.
From your description, I'm a bit wary about knock-on effect to the line meter...

To me it seems the gauge's *ext* might be better suited to hold the critical value *(or a flag to not draw the critical scale)*,
but that might be a breaking change in v7... again, not sure what's the best solution here.

kisvegabor, 2021-07-13:
Finally, I found a simple solution: 797212a3

Does it work well there as well?

fvanroie, 2021-07-14:
[image]

It seems to affect the starting major tick too...

kisvegabor, 2021-07-14:
Please send a code snippet to reproduce the issue.

fvanroie, 2021-07-14:
Sorry, my bad... I forgot to clear the project cache in PlatformIO. Now it pulls in the new commit fine.
I only needed to add a pointer cast to ext:
```cpp
    lv_linemeter_ext_t * ext = (lv_linemeter_ext_t *)lv_obj_get_ext_attr(gauge);
```

Major and minor tick draw OK now when the critical_value is lower, equal or higher then max!
I'll update my project with this fix, so our users can test it. Thanks!

kisvegabor, 2021-07-15:
Great! :) 

I fixed the cast as you suggested here 719fa3ce

commit 042e34cb (2021-06-17): fix(gauge)fix inaccuracy of gauge critical value

Fixes #2288