lvgl/lvgl #2296: I'm having problems with version 8.0

Miguel0101, 2021-06-07:
In version 7.11.0 the video driver works fine but when I migrated to v8 my screen renders but freezes.
Display configuration:
```
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint64_t w = (area->x2 - area->x1 + 1);
  uint64_t h = (area->y2 - area->y1 + 1);

  tft.startWrite();
  tft.setAddrWindow(area->x1, area->y1, w, h);
  tft.pushColors(&color_p->full, w * h, true);
  tft.endWrite();

  lv_disp_flush_ready(disp);
}

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf1[screenWidth * 50];
static lv_color_t buf2[screenWidth * 50];
lv_disp_draw_buf_init(&draw_buf, buf1, buf2, screenWidth * 50);

/*Initialize the display*/
static lv_disp_drv_t disp_drv;
lv_disp_drv_init(&disp_drv);
disp_drv.draw_buf = &draw_buf;
disp_drv.hor_res = screenWidth;
disp_drv.ver_res = screenHeight;
disp_drv.flush_cb = my_disp_flush;
disp_drv.full_refresh = 1;

lv_disp_t * disp;
disp = lv_disp_drv_register(&disp_drv);
```

kisvegabor, 2021-06-07:
Can you use a debugger to see where it freezes? Please also enable [logging in  LVGL](https://docs.lvgl.io/8.0/porting/log.html).

On a side note, if you don't use DMA to push the colors you don1t need to draw buffers.

kisvegabor, 2021-06-07:
Continue in https://forum.lvgl.io/t/screen-freezes-using-lvgl-v8-0/5723

closed 2021-06-07 by kisvegabor