lvgl/lvgl #2298: Arduino issues

dronecz, 2021-06-08:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Arduino 1.8.15
ESP32 support 1.0.6

I´m unable to compile Arduino example sketch. 

I download lv_demos repo from GH via ZIP file, import it to Arduino by functionality for ZIP libraries, but I´m unable to compile any demo. I can not compile sketch even when I comment out `#include "lv_demo.h" ` . I tried to change includes in some files in `lv_demos` and `lv_demos\src` folders but I can not make it work.   

If I try to uncomment `#include <lv_examples.h>` I get `lv_examples: Wrong lvgl version` 

I tried completely new Arduino setup on different machine (just with lvgl and TFT_eSP library) and that result in same compile error.   

### To Reproduce

I dd not try it in the PC simulator as this is problem specific to Arduno platform.

```c
#include <lvgl.h>
#include <TFT_eSPI.h>
/*If you want to use the LVGL examples,
  make sure to install the lv_examples Arduino library
  and uncomment the following line.
  #include <lv_examples.h>
*/

//#include <lv_demo.h>

TFT_eSPI tft = TFT_eSPI(); /* TFT instance */

/*Change to your screen resolution*/
static const uint32_t screenWidth  = 320;
static const uint32_t screenHeight = 240;

static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[ screenWidth * 10 ];

#if LV_USE_LOG != 0
/* Serial debugging */
void my_print( lv_log_level_t level, const char * file, uint32_t line, const char * fn_name, const char * dsc )
{
  Serial.printf( "%s(%s)@%d->%s\r\n", file, fn_name, line, dsc );
  Serial.flush();
}
#endif

/* Display flushing */
void my_disp_flush( lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p )
{
  uint32_t w = ( area->x2 - area->x1 + 1 );
  uint32_t h = ( area->y2 - area->y1 + 1 );

  tft.startWrite();
  tft.setAddrWindow( area->x1, area->y1, w, h );
  tft.pushColors( ( uint16_t * )&color_p->full, w * h, true );
  tft.endWrite();

  lv_disp_flush_ready( disp );
}

/*Read the touchpad*/
bool my_touchpad_read( lv_indev_drv_t * indev_driver, lv_indev_data_t * data )
{
  uint16_t touchX, touchY;

  bool touched = tft.getTouch( &touchX, &touchY, 600 );

  if ( !touched )
  {
    data->state = LV_INDEV_STATE_REL;
  }
  else
  {
    data->state = LV_INDEV_STATE_PR;

    /*Set the coordinates*/
    data->point.x = touchX;
    data->point.y = touchY;

    Serial.print( "Data x " );
    Serial.println( touchX );

    Serial.print( "Data y " );
    Serial.println( touchY );
  }

  return false; /*Return `false` because we are not buffering and no more data to read*/
}

void setup()
{
  Serial.begin( 115200 ); /* prepare for possible serial debug */
  Serial.println( "Hello Arduino! (V8.0.X)" );
  Serial.println( "I am LVGL_Arduino" );

  lv_init();

#if LV_USE_LOG != 0
  lv_log_register_print_cb( my_print ); /* register print function for debugging */
#endif

  tft.begin();          /* TFT init */
  tft.setRotation( 3 ); /* Landscape orientation, flipped */

  /*Set the touchscreen calibration data,
    the actual data for your display can be aquired using
    the Generic -> Touch_calibrate example from the TFT_eSPI library*/
  uint16_t calData[5] = { 275, 3620, 264, 3532, 1 };
  tft.setTouch( calData );

  lv_disp_draw_buf_init( &draw_buf, buf, NULL, screenWidth * 10 );

  /*Initialize the display*/
  static lv_disp_drv_t disp_drv;
  lv_disp_drv_init( &disp_drv );
  /*Change the following line to your display resolution*/
  disp_drv.hor_res = screenWidth;
  disp_drv.ver_res = screenHeight;
  disp_drv.flush_cb = my_disp_flush;
  disp_drv.draw_buf = &draw_buf;
  lv_disp_drv_register( &disp_drv );

  /*Initialize the (dummy) input device driver*/
  static lv_indev_drv_t indev_drv;
  lv_indev_drv_init( &indev_drv );
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read_cb = my_touchpad_read;
  lv_indev_drv_register( &indev_drv );

#if 0
  /* Create simple label */
  lv_obj_t *label = lv_label_create( lv_scr_act() );
  lv_label_set_text( label, "Hello Arduino! (V8.0.X)" );
  lv_obj_align( label, LV_ALIGN_CENTER, 0, 0 );
#else
  /* Try an example from the lv_examples Arduino library
     make sure to include it as written above.
    lv_example_btn_1();
  */

  // uncomment one of these demos
  //   lv_demo_widgets();            // OK
  //    lv_demo_benchmark();          // OK
  // lv_demo_keypad_encoder();     // works, but I haven't an encoder
  // lv_demo_music();              // NOK
  // lv_demo_printer();
  // lv_demo_stress();             // seems to be OK
#endif
  Serial.println( "Setup done" );
}

void loop()
{
  lv_timer_handler(); /* let the GUI do its work */
  delay( 5 );
}
```

Here is error: 

`LVGL_Arduino:109:21: error: invalid conversion from 'bool (*)(lv_indev_drv_t*, lv_indev_data_t*) {aka bool (*)(_lv_indev_drv_t*, lv_indev_data_t*)}' to 'void (*)(_lv_indev_drv_t*, lv_indev_data_t*)' [-fpermissive]
   indev_drv.read_cb = my_touchpad_read;`

### Expected behavior
It should compile without error.

tobozo, 2021-06-08:
just change `bool` to `void` in the function signature, and remove the `return false;` statement :

```C++
/*Read the touchpad*/
void my_touchpad_read( lv_indev_drv_t * indev_driver, lv_indev_data_t * data )
{
   uint16_t touchX, touchY;

   bool touched = tft.getTouch( &touchX, &touchY, 600 );

   if( !touched )
   {
      data->state = LV_INDEV_STATE_REL;
   }
   else
   {
      data->state = LV_INDEV_STATE_PR;

      /*Set the coordinates*/
      data->point.x = touchX;
      data->point.y = touchY;

      Serial.print( "Data x " );
      Serial.println( touchX );

      Serial.print( "Data y " );
      Serial.println( touchY );
   }
}
```

dronecz, 2021-06-09:
> just change bool to void in the function signature, and remove the return false; statement 

If I do that, function will not return if display was touched or not, is that right? I checked older sketch and there is also this function as `bool`.

tobozo, 2021-06-09:
maybe the signature was changed because this function doesn't need to return anything to have the touch working, and the maintainers didn't have time to update the example code?

embeddedt, 2021-06-09:
In v8 the `read_cb` function signature was changed. You can refer to the last entry in the [driver changes](https://github.com/lvgl/lvgl/blob/master/docs/CHANGELOG.md#driver-changes) section of the changelog. Essentially the return value is now stored in `data->continue_reading`.

dronecz, 2021-06-09:
Ok, thank you for clarification . That one is solved but what about problem with `#include lv_demo.h` ?

I get this error:

`\lv_demos\lv_demo.h:19:26: fatal error: ../lvgl/lvgl.h: No such file or directory` which refer to this [line](https://github.com/lvgl/lv_demos/blob/master/lv_demo.h#L19)

I tried  to find definition for  `LV_LVGL_H_INCLUDE_SIMPLE` but without luck..

kisvegabor, 2021-06-16:
`LV_LVGL_H_INCLUDE_SIMPLE` can be defined by you to tell the compiler hor to include `lv_conf.h`.

I found this about adding defined to the build:
https://forum.arduino.cc/t/passing-a-define-to-compiler-via-d-how-to-set-the-define-to-a-value/598826/2

kisvegabor, 2021-06-29:
@dronecz 
Could you try `LV_LVGL_H_INCLUDE_SIMPLE`?

dronecz, 2021-06-29:
I look at that link, but did not find where to put it (which line and where in the line)

I solved it by adding `|| ARDUINO >=100` behind `LV_LVGL_H_INCLUDE_SIMPLE`

I make some changes to the library (and made widget example working) but lost those as I needed to revert back to v7.

kisvegabor, 2021-06-29:
I believe there should be a generic way to do this instead of adding platform-specific defines. 

Anyway, it seems this topic can be closed. 

Feel free to comment here if you got back to this topic.

dronecz, 2021-06-29:
I looked at this again and here is what I find: 

- you have to add `-DLV_LVGL_H_INCLUDE_SIMPLE=.` flag to line `build.extra_flags` in `platform.txt`or even better make `platform.local.txt` and insert `build.extra_flags=-DLV_LVGL_H_INCLUDE_SIMPLE=.` in to it so this flag wont be overwritten when you update platform files
- you have to do this for every platform which are using in Arduino (AVR, SAMD, ESP8266, STM32, etc.) 

I understand your approach and solution, which I found, is quite easy to implement when you know what to do, but Arduino platform is (very often) used by people who do no know what tick under the hood and this is thing which can discourage some people from using lvgl in their project.  

Would there be such I problem add check if I´m using Arduino (`|| ARDUINO >=100`) in to demo files? I´m happy to test all variations in `LVGL_Arduino.ino` sketch and make PR for changes.  

Otherwise I would like to make PR for change in [docs ](https://github.com/lvgl/docs/blob/gh-pages/8.1/_sources/get-started/arduino.md.txt)so other people do not have to fight this issue.

kisvegabor, 2021-07-01:
I got an idea. We use this technique in `lv_conf_internal.h` to include `lv_conf.h` directly if possible:
```c
#if defined __has_include
#  if __has_include("lv_conf.h")
#   ifndef LV_CONF_INCLUDE_SIMPLE
#    define LV_CONF_INCLUDE_SIMPLE
#   endif
#  endif
#endif
```

I wonder if we could do something similar in `lv_demo.h`:
```
#if defined __has_include
#  if __has_include("lvgl.h")
#   ifndef LV_LVGL_H_INCLUDE_SIMPLE
#    define LV_LVGL_H_INCLUDE_SIMPLE
#   endif
#  endif
#endif
```

Could you try it out?

dronecz, 2021-07-01:
Hi, I tried that but it did not work. I had to change it also in files which are in `src\lv_demo_widgets\assets` folder (img_clothes.c, img_demo_widgets_avatar.c, img_lvgl_logo.c) but in the end there is lot more errors.. 🙁 

Here: 

`C:\Users\dronecz\Dropbox\@Arduino\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:29:28: error: unknown type name 'lv_obj_t'
 static void profile_create(lv_obj_t * parent);
                            ^
C:\Users\dronecz\Dropbox\@Arduino\Arduino\libraries\lv_examples\src\lv_demo_widgets\lv_demo_widgets.c:30:30: error: unknown type name 'lv_obj_t'
 static void analytics_create(lv_obj_t * parent);
                              ^
// I crop a lot of errors
`

kisvegabor, 2021-07-06:
The attached libs and sketch is working for me. Finally, the only this I changed is replacing the includes in the mentioned 3 assets with `#include "../../lv_demo.h"`.

[Arduino.zip](https://github.com/lvgl/lvgl/files/6769697/Arduino.zip)

Does it work for you too?

dronecz, 2021-07-06:
I had to make some changes to sketch and files, but yes, it is working for me as well. 

To be sure that everything works for everyone, I changed my folder (in` %appdata%/Local\Arduino15/preferences.txt`) for libraries and download again lvgl & lv_examples via Library manager. Copied `.conf` files for both libraries,  compared files in `/src/lv_demo_widgets/assets` (to be sure what changes you made), changed files in same way and compiled example sketch from lvgl library wihout issue. 

I checked all other examples in sketch and these can be compiled straight away: 

```
lv_demo_widgets();
lv_demo_keypad_encoder();
lv_demo_music(); 
lv_demo_stress();   
```

These have some issues: 

`lv_demo_printer();` is completaly missing example files in `/src` folder. I do not know if this is in purpose or not. 
`lv_demo_benchmark();` is missing `img_cogwheel_rgb`, `img_cogwheel_argb`, `img_cogwheel_chroma_keyed`, `img_cogwheel_indexed16`, `img_cogwheel_alpha16` which are for `7.11.0` lv_examples library in `/src/assets` folder which is missing for `8.0.1` example library

Thanks for help

kisvegabor, 2021-07-08:
Great, I pushed a fix to include `lv_demo.h`

> lv_demo_printer();

It's not ported to v8.

> lv_demo_benchmark();

Fixed.

kisvegabor, 2021-07-23:
It seems all issues are fixed so I close it now. 

Let us know if you find anything else.

closed 2021-07-23 by kisvegabor

dronecz, 2022-06-14:
@vanminh1310 It looks like your touch axis are inverted ;) Your are using TFT_eSPI? I had similar issue and I had to change last parameter in callData array and I´m also sending x to y variable and y data to x variable to fix my axis.

I did not had time to report that issue.

vanminh1310, 2022-06-14:
@dronecz Yes I use TFT_eSPI can you help me fix it

vanminh1310, 2022-06-14:
> @droneczCó, tôi sử dụng TFT_eSPI bạn có thể giúp tôi sửa nó được không

Problem solved. Thank you