lvgl/lvgl #2299: lv_style_set_size inconsistent with lv_obj_set_size

jupeos, 2021-06-08:
[V8] `lv_obj_set_size` supports 2 parameters corresponding to width and height. The corresponding style function `lv_style_set_size` takes a single argument resulting in a square object which is inconsistent with the behaviour of `lv_obj_set_size`.  All other `lv_obj_set_...` to `lv_style_set_...` associations are symmetric in both signature and function.

So a potential solution is
```
static inline void lv_style_set_size(lv_style_t * style, lv_coord_t value1, lv_coord_t value1) {
    lv_style_set_width(style, value1);
    lv_style_set_height(style, value2);
}
```

I appreciate this breaks the uniformity of a single value signature.
Alternatives are:
- Remove `lv_style_set_size`. Requires explicit use of `lv_style_set_width` & `lv_style_set_height`
- Rename `lv_style_set_size` something like `lv_style_set_size_equal`

kisvegabor, 2021-06-09:
You are right, it's really not so good as it is now. 
But unfortunately, we can't remove or change the `lv_style_set_size` without making an API breaking change.  :slightly_frowning_face:

kisvegabor, 2021-06-17:
I close this issue as I think we can't do anything now. :slightly_frowning_face: 

Let us know if you get a good idea about it.

closed 2021-06-17 by kisvegabor