lvgl/lvgl #2300: use image indic in meter maybe abnormal in lvgl8.0 release

XiaAndragon, 2021-06-09:
### Perform all steps below and tick them with [x]
- [X ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X ] Update lvgl to the latest version
- [X ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Use lvgl8.0 release ,I wan to create a clock by using image and meter,but it seems not work very well.I wan the image indic in meter reflash itself.
### To Reproduce

### Expected behavior

### Screenshots or video

[image]
[image]
[image]
[image]
[image]
[image]

kisvegabor, 2021-06-09:
Please send a code snippet to create the smaller meter. 
On my test's it was working fine.

XiaAndragon, 2021-06-09:
```c
  #pragma region 时钟图片
  lv_obj_t *clock_bg = lv_img_create(page1.panel1);
  lv_img_set_src(clock_bg,"./png/clock_bg.png");
  lv_obj_align_to(clock_bg,page1.panel1,LV_ALIGN_RIGHT_MID,-5,0);
  #pragma endregion
  #pragma region 时针
  hour_meter = lv_meter_create(page1.panel1);
  lv_obj_set_size(hour_meter,160,160);
  lv_obj_align_to(hour_meter,clock_bg,LV_ALIGN_CENTER,0,0);
  lv_obj_set_style_bg_opa(hour_meter,0,LV_PART_MAIN);
  lv_obj_set_style_bg_opa(hour_meter,0,LV_PART_INDICATOR);
  lv_obj_set_style_border_width(hour_meter,0,LV_PART_MAIN);

  lv_meter_scale_t * scale = lv_meter_add_scale(hour_meter);
  lv_meter_set_scale_range(hour_meter,scale,0,360,360,180);
  lv_meter_set_scale_ticks(hour_meter,scale, 0, 0, 0, lv_color_make(0,0,0));
  lv_meter_indicator_t * indic_hour = lv_meter_add_needle_img(hour_meter, scale, "./png/gau1bg.png", 7, 14);
  // lv_meter_indicator_t * indic_hour = lv_meter_add_needle_line(hour_meter, scale,3,lv_color_white(),0);
  // lv_meter_set_indicator_end_value(hour_meter, indic_hour, (tm.hour % 12) * 30+(tm.minute/2));
  // usr_hour.meter = hour_meter;
  // usr_hour.indic = indic_hour;
  // lv_timer_create(lv_update_t_hour,60000,NULL);

  #pragma endregion
  #pragma region 分针
  lv_meter_indicator_t * indic_min = lv_meter_add_needle_img(hour_meter, scale, "./png/gau2bg.png", 7, 14);
  // lv_meter_indicator_t * indic_min = lv_meter_add_needle_line(hour_meter, scale,3,lv_color_white(),1);
  // lv_meter_set_indicator_end_value(hour_meter, indic_min, tm.minute*6);
  // usr_min.meter = hour_meter;
  // usr_min.indic = indic_min;
  // lv_timer_create(lv_update_t_min,1000,NULL);

  /*Create an animation to set the value*/
  lv_anim_t a;
  lv_anim_init(&a);
  lv_anim_set_exec_cb(&a, set_value);
  lv_anim_set_values(&a, 0, 360);
  lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
  lv_anim_set_time(&a, 2000);     /*2 sec for 1 turn of the minute hand (1 hour)*/
  lv_anim_set_var(&a, indic_min);
  lv_anim_start(&a);

  lv_anim_set_var(&a, indic_hour);
  lv_anim_set_time(&a, 24000);    /*24 sec for 1 turn of the hour hand*/
  lv_anim_set_values(&a, 0, 360);
  lv_anim_start(&a);
  
  #pragma endregion

#pragma region 风速框
  lv_obj_t *speedbg = lv_img_create(page1.panel2);
  lv_img_set_src(speedbg,"./png/speedbg.png");
  lv_obj_center(speedbg);
  
  speed_meter = lv_meter_create(page1.panel2);
  lv_obj_set_size(speed_meter, 300, 300);
  lv_obj_set_style_bg_opa(speed_meter,0,LV_PART_MAIN);
  lv_obj_set_style_bg_opa(speed_meter,0,LV_PART_INDICATOR);
  lv_obj_set_style_border_width(speed_meter,0,LV_PART_MAIN);
  lv_obj_set_style_pad_all(speed_meter,0,LV_PART_MAIN);
  lv_obj_center(speed_meter);

  LV_IMG_DECLARE(speed_indic);
  lv_meter_scale_t * speed_scale = lv_meter_add_scale(speed_meter);
  lv_meter_set_scale_range(speed_meter,speed_scale,0,360,360,60);
  lv_meter_set_scale_ticks(speed_meter,speed_scale, 0, 0, 0, lv_color_make(0,0,0));
  lv_meter_indicator_t * indic_speed = lv_meter_add_needle_img(speed_meter, speed_scale, &speed_indic, 16, 16);
  
  lv_anim_init(&a);
  lv_anim_set_exec_cb(&a, speed_value);
  lv_anim_set_values(&a, 0, 240);
  lv_anim_set_repeat_count(&a, LV_ANIM_REPEAT_INFINITE);
  lv_anim_set_time(&a, 2000);     /*2 sec for 1 turn of the minute hand (1 hour)*/
  lv_anim_set_var(&a, indic_speed);
  lv_anim_start(&a);

#pragma endregion
```

XiaAndragon, 2021-06-09:
> Please send a code snippet to create the smaller meter.
> On my tests it was working fine.

Hi,the main test code snippet was uploaded.

kisvegabor, 2021-06-10:
Thanks, I've pushed a fix. 

Note that the needle images should point to the right. Like this:
[image]

closed 2021-06-10 by kisvegabor

XiaAndragon, 2021-06-10:
Thanks,I got it.

commit 54d8e817 (2021-06-10): fix(meter) fix needle image invalidation

Fixes #2300