lvgl/lvgl #2301: How to get all the name of buttons from a List ?

sfarwassim, 2021-06-09:
### Introduce the problem

How to get all the name of buttons from a List ? 

### Examples and cases

lv_obj_t *list_of_actions_btn;
I want to get all the buttons in   "list_of_actions_btn"

### Suggested solution
I dont have solutions

embeddedt, 2021-06-09:
You will need to loop over the buttons using [`lv_obj_get_child`](https://docs.lvgl.io/8.0/widgets/obj.html#parents-and-children). Then, you can use `lv_list_get_btn_text(list, btn)` to get the button text.

----------------------
We use GitHub issues for development related discussions.
Please use the [forum](https://forum.littlevgl.com/) to ask questions.

It's possible to sign in to the forum with your GitHub account... you don't need to make a separate account.

closed 2021-06-09 by embeddedt

sfarwassim, 2021-06-09:
@embeddedt  do u have some example please ?  because it  i trited that and it didnt work !
This is my code :
lv_obj_t *list_of_actions_btn;
 
uint32_t i;
for(i = 0; i < lv_obj_get_child_cnt(parent); i++) {
  lv_obj_t * child = lv_obj_get_child(list_of_actions_btn, i);
            printf("%s\n", lv_list_get_btn_text(child));

}

Tell me if i did somthing wrong ?

embeddedt, 2021-06-09:
* Which LVGL version are you using?
* You call `lv_obj_get_child_cnt` on `parent` but then `lv_obj_get_child` on `list_of_actions_btn`.
* Is `list_of_actions_btn` actually initialized? You seem to be defining it and then using it immediately without assigning a value. Or are you just providing it for context in your example?

sfarwassim, 2021-06-09:
- i am using LVGL 7.3
-   i fixed  , i am calling now lv_obj_get_child_cnt  on  list_of_actions_btn and still the same problem !
- Okay now i will show you how i created the list 👍 
lv_obj_t *list_of_actions_btn;
 list_of_actions = lv_list_create(screenAction1, NULL);
for (int i = 0; i < 3; i++)
                {
                    char *name = cJSON_GetObjectItem(arrayChannel, "name")->valuestring;
                    LV_IMG_DECLARE(img1620211926196_png);
                    list_of_actions_btn = lv_list_add_btn(list_of_actions, &img1620211926196_png, name);
              }
************ Now i need to get  all the button insdie the list to do some stuffs ! **************

Thank you sir