lvgl/lvgl #2303: C++ binding

kisvegabor, 2021-06-09:
### Introduce the problem

It'd be great to provide C++ API for LVGL. 

### Examples and cases
Almost every common and modern platform support C++ so the binding could be used on many places.

### Initial questions
- how should the API look like
- what C++ features to use
- automatic code generation
- manually added parts
- how to keep track of the new features of LVGL
- how and what kind of test shall we run  to test the binding

If you are interested in this topic, please comment here and share your ideas. Or even better contribute to the implementation. :slightly_smiling_face:

liebman, 2021-06-09:
I'm interested. For my own personal projects, v6 then v7 based, I created a [simple c++ wrapper for esp32 using esp-idf](https://github.com/liebman/lvgl_cpp).  I've not had time to look at v8 yet.  It would be better to generate the wrapper if possible, manual updates would be (and are) tedious and error prone.

fstuff-dev, 2021-06-09:
For the API what notation we should use ?

- CamelCase
- dromedaryCase

liebman, 2021-06-09:
`CamelCase` for classes / `dromedaryCase` for methods?

fstuff-dev, 2021-06-09:
For example :

class LvObject {
private:
...
protected:
...
public:
    LvObject();
    void setSize(unsigned int w,unsigned int h)
    ...
    virtual ~LvObject();

...
}

puzrin, 2021-06-10:
> how should the API look like

To reply this question, you need answer to question "what problem do you need to resolve". Not abstract "it would be nice to have cpp support"

For example, i see problem with boring declarations when i have to assign/define multiple classes/properties. This requires chained syntax, available in cpp.

IMO good cpp support for current c code will be endless pain. But "sugar via cpp" - possible with low cost. Some wrappers (like ETL for strings) to simplify local syntax.

kisvegabor, 2021-06-10:
> To reply this question, you need answer to question "what problem do you need to resolve". Not abstract "it would be nice to have cpp support"

True, let's collect them :slightly_smiling_face: 

@fstuff-dev 
> class LvObject {

Can/Should `Lv` be a namespace?

puzrin, 2021-06-10:
> True, let's collect them

## Generic things

I would propose to focus on low cost things. IMO full mappings of current C arch to CPP is NOT possible with ease. Drop all ideas about automated generators and so on. Just create manually (template?) wrapper for some base objects. And use CI to care about C api drifts.

## More low level details

1. LvStyle wrapper - most demanded, to assign properties (with `.chainable()` manner)
2. LvObject
    - assign styles (chained)
    - assign event listeners (chained, as lambdas)
3. Easy casts lv_obj <=> LvObject - for easy local equations
4. Method variants to simplify calls

IMO that's MVP, which will solve 90% of boring things. Good goal for first release, easy to reach & maintain. More advanced things can be discussed later.

kisvegabor, 2021-06-10:
> I would propose to focus on low cost things. IMO full mappings of current C arch to CPP is NOT possible with ease. Drop all ideas about automated generators and so on. Just create manually (template?) wrapper for some base objects. And use CI to care about C api drifts.

It'd mean constant maintenance to add new widgets/features. The MicroPython binding is fully autogenerated and very Pythonic (expect some special complex cases). 

So I think we should go for auto-generation with a few manual parts where required. 

> More low level details

I agree with all points. 

In addition ofr styles:
```
obj1.addStyle(&style); //default selector
obj1.addStyle(&style, LV_STATE_PRESSED); 
```

embeddedt, 2021-06-10:
> So I think we should go for auto-generation with a few manual parts where required.

I agree.

I had previously made a very basic prototype of an autogenerated C++ API with pycparser (the same library used for the MicroPython binding generator). I haven't had time to work on it recently, but if there's interest, I can put it on a GitHub repository. I made it for an earlier v8 snapshot but I should be able to update it to 8.0 easily.

Something that needs to be discussed is how to handle object creation/deletion, as objects created from C++ as well as objects created by LVGL in C should look identical.

kisvegabor, 2021-06-10:
> Something that needs to be discussed is how to handle object creation/deletion, as objects created from C++ as well as objects created by LVGL in C should look identical. That was a stumbling point for my proof-of-concept.

I don't use C++ that much so probably I'm not the best person to give suggestions but what was the problem? Using constructors similarly to MicroPython look at first. E.g. `LvBtn(parent)`.

fstuff-dev, 2021-06-10:
I agree with automatic code generator because take care of all C API it’s not easy at the moment …. 
i propose to implement also a special C++ object called “LvApplication” that cares about all initialization things and thread creation …

embeddedt, 2021-06-10:
Constructors work for creating new objects from C++, but how do we handle existing objects (like `lv_scr_act()`)?

kisvegabor, 2021-06-10:
> Constructors work for creating new objects from C++, but how do we handle existing objects (like lv_scr_act())?

`LvScrAct()` ? :slightly_smiling_face: 

Sorry, I don't see the problem, could you explain it?

ropg, 2021-06-10:
I once wrote something which, with the help of some preprocessor magic, allowed one to do:

`submit.on(E_TAPPED | E_PRESSED, doFunction {`

(Where `submit` was a button)

puzrin, 2021-06-10:
> So I think we should go for auto-generation with a few manual parts where required.

I suggested only things i can predict. If you know how to implement autogeneration fast - no problem.

> Something that needs to be discussed is how to handle object creation/deletion, as objects created from C++ as well as objects created by LVGL in C should look identical. That was a stumbling point for my proof-of-concept.

I'm not sure if this can be done well. But, from my point of view, this is not critical point (=> not mandatory). IMO simplified wrapper would be ok for 3-5 years, and later everything can be rewritten to CPP, as standalone project.

Under simple wrapper i mean ~ template for multiple chained calls. Not "true" cpp programming, just a lot of syntax sugar.

fstuff-dev, 2021-06-10:
Probably with simple wrapper you’ll have more control of what you’re doing !! 
it can be boring but you can add feature not implemented in the C library directly in C++ objects !!

kisvegabor, 2021-06-10:
> Probably with simple wrapper you’ll have more control of what you’re doing !!
it can be boring but you can add feature not implemented in the C library directly in C++ objects !!

It's ok, but we should ensure that to have at least a basic C++ version of every API function. That is all functions should be converted to class member functions and IMO it can be done by a script automatically.

fstuff-dev, 2021-06-10:
That’s right !

ValentiWorkLearning, 2021-06-11:
This API should be almost-without overhead. What do you think about providing just a simple set of primitives for dealing with the library?

ValentiWorkLearning, 2021-06-11:
The list of points which should be considered includes:

- API without exceptions
- huge usage of virtual functions sometimes can lead to the performance decreasing
- minimal supported standard for the wrapper
- more compile-time things

ValentiWorkLearning, 2021-06-11:
In my C++ project a few wrappers were used for LVGL-C++ integration instead of writing the whole wrapper for the library. 
For instance, for an object deletion when a view becomes invisible - the pointer wrapper was used in a such way:
```cpp
Meta::PointerWrapper<lv_obj_t, lv_obj_del> m_pStepsLabel;

m_pStepsLabel.reset(lv_label_create(_parentObject));

lv_obj_add_style(m_pStepsLabel.get(), &m_mainFontLight,
                 LV_PART_MAIN | LV_STATE_DEFAULT);
lv_label_set_text(m_pStepsLabel.get(), "3600");

```
The pointer wrapper is defined as:
```cpp
template <auto DeleterFunction>
using CustomDeleter =
    std::integral_constant<decltype(DeleterFunction), DeleterFunction>;

template <typename ManagedType, auto Functor>
using PointerWrapper = std::unique_ptr<ManagedType, CustomDeleter<Functor>>;
```
It can be considered as C++ style of wrapping C-objects, but without extra overhead.
Probably, instead of creating the full C++ API for the library it's better to provide a utility set for managing it, which can be lightweight.

fstuff-dev, 2021-06-11:
Sorry … what the advantage in using this technique instead using directly C library ?!

embeddedt, 2021-06-11:
> As an idea also- it's not necessary to keep all of C++ wrappers. Instead, they can be used as a proxy.

I like the proxy idea. It solves the issue I mentioned before.

@kisvegabor What I was referring to is that there needs to be some way of handling the fact that objects could either be created through a C++ declaration...

```cpp
lv::button btn;
// or
lv::button *btn = new lv::button;
```

...or through being returned from an LVGL API method:

```cpp
lv::widget screen = lv_scr_act(); /* how do you handle this? */
```

However, the consensus seems to be that a more lightweight wrapper would be simpler and easier to maintain, and I'm also in favor of that direction.

liebman, 2021-06-11:
> 
> ```c++
> lv::widget screen = lv_scr_act(); /* how do you handle this? */
> ```
> 
This would be the `operator=` right?

embeddedt, 2021-06-11:
That can work, but then (unless I'm missing something) there needs to be a flag to differentiate it from an object created through C++ (the first case I listed), as otherwise `screen` would get deleted when the function containing it returns, right?

liebman, 2021-06-11:
That is true :-(.

ValentiWorkLearning, 2021-06-12:
```cpp
lv::button btn;
// or
lv::button *btn = new lv::button;
```
@embeddedt seems to me, the best option should be lvgl-smartpointer like either std::unique_ptr or std::shared_ptr which depends on a user's intentions. At the first glance, the idea with `lv::button bt` seems cool, but this hides a lot of details about an LVGL object creation.
What about the manner with `make_` functions family or `create` functions family?

It can be like:
```cpp
namespace lvgl
{
template<typename TWidget>
using widget_pointer = std::unique_ptr<TWidget,lv_obj_del>;
using style_pointer = std:unique_ptr<lv_style_t,lv_style_reset>;
}
```

And this api can be simply used with the existing set of functions with widgets creation. It can be a quite challenge to wrap existing styles generation for concrete widgets, but it's still possible.

@fstuff-dev about your question for the necessity of a such wrappers set and their benefits. From my C++ experience and C libraries integration -  C++ api shouldn't be a simple copy of the existing API with a huge hierarcy of classes just for the keeping "Clean Object Oriented Design". It should be considered, that LVGL was originally designed for small MCUs and systems. Also, if you take a look at the C++ standard library design- you'll see that the library sometimes contains the "procedural-implemeneted " API.
The main benefit of C++ wrappers, imho, should be type-safety, reducing the error-prone usage of the LVGL library and simplifying the library memory management with providing the flexible set of smart-pointers.

As a conslusion,the main benefits are:
* smart memory management with explicitly shown a programmer intention( shared or unique owning of a widget)
* type safety with the C++ type system
* simple integration with the existing plain C-API even if some new functions were added - the simle raw handle to the widget can be obtained from the wrapper

fstuff-dev, 2021-06-12:
Clear like water …. I Thinq that this approch is better than class hiercachy …. 
i agree in using this method …
👍

puzrin, 2021-06-13:
@ValentiWorkLearning IMO i suggested the same, but via templates instead of `make_XXX`. The difference is in better compile-time optimizations.

That will be like "utilities for smart syntax" (every class is final), because cpp has different inheritance model.

But i'm not expert in CPP, probably i miss something.

scottandrew, 2021-06-14:
I did mine with smart ptrs.. And a weak ptr in user_data to handle events and current screen. Also supplied wrappers around functions to abstract away lvgl library.. 

`
#include "object.h"
#include <iostream>

using namespace std;

Object::Object(lv_obj_t *anObject) {
  object = anObject;

  auto callback = [](lv_obj_t *obj, lv_event_t event) -> void {
    try {
      auto data = lv_obj_get_event_dsc(obj, 0);
      Object *eventObject = (Object *)data->user_data;

      if (event == LV_EVENT_DELETE) {
        data->user_data = NULL;
        return;
      }
      eventObject->events.at(event)(eventObject);
    } catch (...) {
    }
  };

  lv_obj_add_event_cb(object, callback, this);
}

Object::~Object() { lv_obj_del(object); }

void Object::removeStyle(uint8_t part, lv_state_t state, lv_style_t* style) { 
  lv_obj_remove_style(object, part, state, style);
}

void Object::setSize(lv_coord_t width, lv_coord_t height) {
  lv_obj_set_size(object, width, height);
}

void Object::setWidth(lv_coord_t width) { lv_obj_set_width(object, width); }

void Object::align(lv_align_t alignment, lv_coord_t xOffset,
                   lv_coord_t yOffset) {
  lv_obj_align(object, alignment, xOffset, yOffset);
}

void Object::setBackgroundColor(uint8_t part, lv_state_t state, Color color) {
  lv_obj_set_style_bg_color(object, part, state, color.lvColor());
}

void Object::setLineColor(uint8_t part, lv_state_t state, Color color) {
  lv_obj_set_style_line_color(object, part, state, color.lvColor());
}

void Object::setBorderColor(uint8_t part, lv_state_t state, Color color) {
  lv_obj_set_style_border_color(object, part, state, color.lvColor());
}

void Object::setLineWidth(uint8_t part, lv_state_t state, lv_coord_t width) {
  lv_obj_set_style_line_width(object, part, state, width);
}

void Object::setPaddingAllSides(uint8_t part, lv_state_t state,
                                lv_coord_t padding) {
  lv_obj_set_style_pad_all(object, part, state, padding);
}

void Object::setPaddingLeft(uint8_t part, lv_state_t state,
                            lv_coord_t padding) {
  lv_obj_set_style_pad_left(object, part, state, padding);
}

void Object::setPaddingRight(uint8_t part, lv_state_t state,
                             lv_coord_t padding) {
  lv_obj_set_style_pad_right(object, part, state, padding);
}

void Object::setPaddingTop(uint8_t part, lv_state_t state, lv_coord_t padding) {
  lv_obj_set_style_pad_top(object, part, state, padding);
}

void Object::setPaddingBottom(uint8_t part, lv_state_t state,
                              lv_coord_t padding) {
  lv_obj_set_style_pad_bottom(object, part, state, padding);
}

void Object::setBorderWidth(uint8_t part, lv_state_t state, lv_coord_t width) {
  lv_obj_set_style_border_width(object, part, state, width);
}

void Object::setFont(uint8_t part, lv_state_t state, lv_font_t *font) {
  lv_obj_set_style_text_font(object, part, state, font);
}

void Object::setTextAlignment(uint8_t part, lv_state_t state, lv_text_align_t alignment) {
  lv_obj_set_style_text_align(object, part, state, alignment);
}
`

liebman, 2021-06-14:
Supporting lambdas for callbacks would be a huge plus.

scottandrew, 2021-06-14:
> > Constructors work for creating new objects from C++, but how do we handle existing objects (like lv_scr_act())?
> 
> `LvScrAct()` ? 🙂
> 
> Sorry, I don't see the problem, could you explain it?

I was writing abstracted wrappers.. So I had a Screen::current that decoded the user_data on the object since the Object stuffed self in there. 

```
shared_ptr<Screen> Screen::create() { return shared_ptr<Screen>(new Screen()); }

Screen::Screen() : Object(lv_obj_create(NULL)) {}

void Screen::load() { lv_scr_load(lvObjectPtr()); }

shared_ptr<Screen> Screen::current() {
  try {
    lv_obj_t *screen = lv_scr_act();
    auto data = lv_obj_get_event_dsc(screen, 0);
    // lets attempt to get the weak ptr.

    if (data->user_data != NULL) {
      shared_ptr<Object> object =
          ((Object *)data->user_data)->shared_from_this();

      return static_pointer_cast<Screen>(object);
    }
  } catch (...) {
  }

  return shared_ptr<Screen>();
}
```

ShenRen, 2021-06-15:
Glad that LVGL plans to launch its own C++API starting with version 8. In fact I have been working on a C++ wrapper for LVGL and hope it will be of some help. 

I started to explore C++ encapsulation from the 5th edition, and I am using the 6th edition for development. Now the 7th edition has reached the final perfection, and I have asked my colleagues to follow up the encapsulation work of the 7th edition.

[lvgl_cpp_v6x](https://github.com/Leadiffer/lvgl_cpp_v6x) 

 [LittlevGL_v5x_CPPPort](https://github.com/panda-3d-printer/LittlevGL_CPPPort)

Personally, I think the C++ API needs to accomplish these tasks：
1. encapsulation & inheritance & polymorphism,  
 ```c++
 class LvObject 
    : public lv_obj_t
 {
protected:
    LvObject (){ }
public:
    LvObject (LvObject * parent, LvObject * copy = nullptr){ lv_obj_create(parent,copy); }
    virturl ~LvObject (){ lv_obj_del(this); }

    void set_size(lvCoord w, lvCoord h)
    {
         lv_obj_set_size(this, w, h);  // Automatic conversion of Pointers
    }
 }

class LvLed 
    : public LvObject
    , public lv_led_t
{
public:
    LvLed (LvObject * parent, LvLed * copy = nullptr)
     : LvObject()
    {
         lv_led_create(parent, copy);
    }

    void on() { }
    void off() { }
}

```
2. C++ style object creation,  
```c++
// create
Lvobject * obj = new LvObject(parent); 
// delete
delete obj;
```
3. Object-oriented interfaces, 
```c++
obj->set_size(480,320);
```
4. Guarded pointers to LvObject 
```c++
// It can automatically sense if an object exists.
LvPointer<LvLed> led = new LvLed(parent);
if(led.isValid() ) // is true;
{
    // do some thing
    led->on();
}

// ......

delete led;  
if(led.isValid() ) // is false
{
    // do some thing
}
[... 243 lines cut ...]
    [LVCallBack.h](https://github.com/Leadiffer/lvgl_cpp_v6x/blob/lvgl_cpp_v6x/LVCore/LVCallBack.h)

+ Signal and Slot
    [LVSignalSlot.h](https://github.com/Leadiffer/lvgl_cpp_v6x/blob/lvgl_cpp_v6x/LVCore/LVSignalSlot.h)
```c++
 LVSignal signal;
 LVSlot slot;
slot.setSlotFunc([&](){
  // do some thing
});
 signal.connect(&slot) // Connect the signal to the slot
// ......
signal.emit(); // Sends a signal to trigger the slot function to work
```
+  Smart pointer 
    [LVScopedPointer.h](https://github.com/Leadiffer/lvgl_cpp_v6x/blob/lvgl_cpp_v6x/LVCore/LVScopedPointer.h)
    [LVSharedPointer.h](https://github.com/Leadiffer/lvgl_cpp_v6x/blob/lvgl_cpp_v6x/LVCore/LVSharedPointer.h)
+ Guarded pointers to LvObject 
    [LVPointer.h](https://github.com/Leadiffer/lvgl_cpp_v6x/blob/lvgl_cpp_v6x/LVCore/LVPointer.h)
+ Screen Management
    [LVScreen.h](https://github.com/Leadiffer/lvgl_cpp_v6x/blob/lvgl_cpp_v6x/LVApp/LVScreen.h)
+ Tasks attached to the screen
    [LVScreenTask.h](https://github.com/Leadiffer/lvgl_cpp_v6x/blob/lvgl_cpp_v6x/LVApp/LVScreenTask.h)

Hopefully these rough tips will help you. I recently took a closer look at the code for LVGL version 8 and saw that the framework has changed a lot, but I'm still going to try to wrap it in C++ for my development work

ShenRen, 2021-06-15:
how should the API look like.
Personally, I think it's best to keep the method names as similar as possible to the C language interface. This can greatly reduce the user's learning cost, read the C language documentation can also used the C++ interface immediately .
+  Class and type name
```c++
// Remove underline, use Camel-Case, Consider completing words
lv_obj_t  => LvObj  or LvObject
lv_led_t  => LvLed
lv_btn_t  => LvBtn  or LvButton

LvLed led;
led.set_size();
led.on();

// That's fine, just a little awkward.
lv_obj_t  => lv_obj
lv_led_t  => lv_led
lv_btn_t  => lv_btn

lv_led led;
led.set_size();
led.on();

//  even remove the LV prefix from the front of the class
//  Put everything in a namespace lv

namespace lv
{
    class obj
    {
    }

    class led
    {
    }
}

lv_obj_t  => lv::obj
lv_led_t  => lv::led
lv_btn_t  => lv::btn

````
+ Enumeration name
```c++
LV_ALIGN_CENTER  => Lv::Align::Center
// or
LV_ALIGN_CENTER  => LV::ALIGN::CENTER

// definition
namespace LV
{
    enum Align
    {
        Center = LV_ALIGN_CENTER , 
        // Other enumerated values
    }
}
```
+  Method name, Just remove the prefix from the method name
```c++
lv_obj_set_width() => LvObj::set_width()
lv_led_on()  => LvLed::on()

class LvObj
{
public:
    // lv_obj_set_width
    void set_width(){ }  // Just remove the prefix from the method name
    // Other interface methods
}

class LvLed
{
public:
    // lv_led_on
    void on(){ }  // Just remove the prefix from the method name
    // Other interface methods
}

```

scottandrew, 2021-06-15:
If we are looking at light wrappers I would say no name space.. I like LED, Object, Button, etc.. I think hierarchy is important in that there are a lot of things that take just an object and it mirrors the C hierarchy.  Storing self in user_data I think is important for memory and shared object usage. Things like current screen etc. If it can be figured out how to use a smart ptr that is reference counted it also means you don't need to always hold on to an instance it could be unwrapped. Also storing self allows lambadas and getting the current screen class possible without creating a new instance of Screen (or LVScreen) what ever its called.

I also think its important to have the classes look like C++ classes and take advantage of C++ not just the type safety. I think the overhead is minimal.. I always look at these things like we we should rarely have to dip in tot he C part of code.

ValentiWorkLearning, 2021-06-15:
@ShenRen I've explored your idea with API. Thanks.
Seems to me, a few proposals can be cool:
1) LvObj should be templated. With CRTP we can get rid of the extra-inherited parent class.
Before:
```cpp
class LvLed 
   : public LvObject
   , public lv_led_t
{
public:
   LvLed (LvObject * parent, LvLed * copy = nullptr)
    : LvObject()
   {
        lv_led_create(parent, copy);
   }

   void on() { }
   void off() { }
}
```
After:

```cpp
class LvLed 
   : public LvObject<lv_led_t>
{
public:
   LvLed (LvObject * parent, LvLed * copy = nullptr)
    : LvObject()
   {
        lv_led_create(parent, copy);
   }

   void on() { }
   void off() { }
}
```

Note, that `lv_led` contains the nested `lv_obj_t` inside. It is unnecessary to inherit `LvLed` from `lv_led_t`.

2. About self-written smart pointers. IMHO, It's better to keep on using C++ standard library pointers instead of self-written. std::shared_ptr with std::unique_ptr are better choise, also the custom deleter can be provided for them.
3. Prefer `enum class` over simple `enum`
4.  Don't try to shorting the names. It's pretty uncomfortable sometimes. Instead of `lv::btn` it's better to use `lv::button`.
5. `LVCallBack` can be reimplemented with the `FastPimpl` idiom. 

As a conclusion, your framework wrapper contains a lot of cool ideas. It'll be cool to agreed changes and start working on the lvgl cpp bindings.

fstuff-dev, 2021-06-15:
In my framework i’ve implemented lvcallback as a template class…. So callbacks can be callad with operator() and Call() method ! 
Take a look at https://github.com/fstuff-dev/lvglpp/blob/3b638777de6e7e9da5132688c010b4c6b7932e5f/lv_cpp/lvCore/lvCallback.h

zapta, 2021-06-20:
I am joining late, so can't say I read every post here but here are my thoughts, just in case.

Problem to solve:  Currently LVGL tries to model an object oriented system but tracking the relationships between objects hierarchy (e.g. label is a special class of obj) , parts (what enum parts apply to what object, and methods (what function prefixes apply to what object) is left to the programmer and the IDEs provide little help since they are not aware of those relationships. This results in a hard to explore and work with API, loss of productivity, useful features not being found, and potentially some users giving up. 

Binding:  the use of the word 'binding' suggests some afterthought where a legacy code is dressed with a facade of a C++ API. Hopefully this will end up with an intuitive and 'native' C++ API that doesn't leak internal C limitations.

Coding style: C++ I suggest to pick one fo the common coding styles and stick with them. This way you don't need to waste time on coming with a coding style of your own (with endless debates) and use a coding style that is supported out of the box by common IDEs.

Success criteria (in an arbitrary order):
1. Native C++ API. Doesn't have oddities that are not natural or intuitive in C++, due to LVGL's C origins. Since C++ will be the main LVGL language, it should feel as if LVGL was designed from the ground up in C++.
2. Discoverability. LVGL features can easily be explored and used with a common IDE and code completion. E.g. One can finds his way with code completion to construct a statement such as  my_chart.series(0).add_point(adc_value).
3. Minimize things the user must remember from the documentation. For example, currently one need to remember what 'parts' apply to what 'object' and what properties of each part means for the for the actual widget (e.g. series bg part control chart's grid). This should all be represented in the API structure, e.g. my_chart.background().line_style().set_width(2).
4. Reasonable resource utilization and performance. Not much worse than currently.
5. Extensible. One can extend existing widgets and create new ones with no much effort.

Achieving this will make LVGL a pleasure to work with and is inline with the current trend of smarter IDEs and more capable and less expensive MCUs.

kisvegabor, 2021-06-21:
> Currently LVGL tries to model an object oriented system but tracking the relationships between objects hierarchy (e.g. label is a special class of obj) , parts (what enum parts apply to what object, [...]

Thanks for the feedback. In v7 there were pats like `LV_CHART_PART_MAIN` but it was redundant in a lot of points (e.g. every widget has MAIN part) and it wasn't obvious that `LV_CHART_PART_MAIN` and `LV_TEXTAREA_PART_MAIN` are basically the same thing. Anyway, I'll think about adding helper parts such as `LV_PART_INDICATOR -> LV_CHART_PART_LINE_BULLET`.
It's a different topic and I'd love to talk about the issues of the C API too. So if you (or anybody) has the interest to discuss it please open an issue.  :slightly_smiling_face: 

> Success criteria (in an arbitrary order):

I agree with all the points. :+1:

kisvegabor, 2021-06-29:
Hi guys, 
how to continue on this? We got a lot of interesting ideas and approaches and I think we can pick one and start to work on it. 

I'm not good enough in C++ to make the right decisions so I wonder if there is someone who is willing to lead the C++ binding project.

ValentiWorkLearning, 2021-06-29:
@kisvegabor I can try to add a small utility header with C++ smartpointers wrapper:)

fstuff-dev, 2021-06-29:
Sooo…. Someone ha e to create the repository and start the project !

> @kisvegabor I can try to add a small utility header with C++ smartpointers wrapper:)

Let’s start with that !!

kisvegabor, 2021-06-29:
@fstuff-dev 
Good idea :slightly_smiling_face: 
The repo is created here: https://github.com/kisvegabor/lv_binding_cpp

@ValentiWorkLearning 
Let's continue there. I've just opened the first issue.

closed 2021-06-29 by kisvegabor

kisvegabor, 2021-07-05:
I've just noticed that I opened the repo on my personal account.
Transferred here: https://github.com/lvgl/lv_binding_cpp