lvgl/lvgl #2305: Image button (imgbtn) default width is incorrect

jupeos, 2021-06-09:
[V8] image width is slightly larger than actual image, by default, causing repeats. Run in the sim. Same code worked for V7.

When attempting to create a single image (not repeating) button, the button is automatically sized to be slightly larger than the image, which results in a repetition.

To reproduce
```
lv_obj_t *imgbtn1 = lv_imgbtn_create(lv_scr_act());
lv_obj_remove_style_all(imgbtn1);
lv_imgbtn_set_src(imgbtn1, LV_IMGBTN_STATE_RELEASED, NULL, &common_button_next_up, NULL);
lv_obj_update_layout(imgbtn1);
lv_coord_t w = lv_obj_get_width(imgbtn1);
lv_coord_t h = lv_obj_get_height(imgbtn1);
```
Here w=**130** & h=88 for the given **(124, 88)** image and this is the visual result
[image]
Additionally, when attempting to use `LV_SIZE_CONTENT` with an `imgbtn` the resulting size is 0 for both w & h, i.e. `lv_obj_set_size( &common_button_next_up, LV_SIZE_CONTENT, LV_SIZE_CONTENT )` does not work as expected.

The image used was recently run through the online Image Converter and shows the correct size in the generated C file. I have attached the image for reference but believe this will occur with any image.
[image]

kisvegabor, 2021-06-10:
Hi, 

it happens because the image button doesn't set its width automatically.  130 (LV_DPI_DEF) is the default width which is close to 124 and therefore it looks buggy.

Normally, if you set left, center, and right parts a fixed-width works well because the center part will be repeated and you will get a "healthy" button. Actually, I think setting the width automatically, in this case, would be rather unintuitive because the point of the widget is to have an image which size can be adjusted. 
 
However, I agree if only the center part is specified and the object's width is `LV_SIZE_CONTENT` the width could be set to the image source automatically.

jupeos, 2021-06-25:
> However, I agree if only the center part is specified and the object's width is `LV_SIZE_CONTENT` the width could be set to the image source automatically.

This is definitely a bug that needs fixing.

kisvegabor, 2021-06-28:
I've just fixed it. You still need to set `lv_obj_set_width(imgbtn, LV_SIZE_CONTENT)` manually.

commit 7e49f488 (2021-06-28): fix(imgbtn) consider width==LV_SIZE_CONTENT if only mid. img is set

fixes #2305
commit 8080b5f7 (2021-07-16): fix(imgbtn) consider width==LV_SIZE_CONTENT if only mid. img is set

fixes #2305
closed 2021-06-28 by kisvegabor