lvgl/lvgl #2311: LV_COLOR_MAKE16 problem

ali-rostami, 2021-06-13:
### Describe the bug
After I upgraded to v8 I got to this compile error:
"expected expression before {  token"
from GCC compiler.
after checking the v7 and changed this line:
```
# define LV_COLOR_MAKE16(r8, g8, b8)  {{(uint8_t)((b8 >> 3) & 0x1FU), (uint8_t)((g8 >> 2) & 0x3FU), (uint8_t)((r8 >> 3) & 0x1FU)}}
```

to this:
```
# define LV_COLOR_MAKE16(r8, g8, b8) (_LV_COLOR_MAKE_TYPE_HELPER{{(uint8_t)((b8 >> 3) & 0x1FU), (uint8_t)((g8 >> 2) & 0x3FU), (uint8_t)((r8 >> 3) & 0x1FU)}})
```
It got solved. any idea?

embeddedt, 2021-06-13:
I think that's the correct fix. It's what gets used for `LV_COLOR_MAKE8` (but strangely, not `LV_COLOR_MAKE32`).

kisvegabor, 2021-06-14:
@ali-rostami Could you send a PR with the fix?

ali-rostami, 2021-06-15:
> @ali-rostami Could you send a PR with the fix?

yes, so to make sure I'm asking, this is O.K:
```
#if LV_COLOR_16_SWAP == 0
# define _LV_COLOR_ZERO_INITIALIZER16 {{0x00, 0x00, 0x00}}
# define LV_COLOR_MAKE16(r8, g8, b8) (_LV_COLOR_MAKE_TYPE_HELPER{{(uint8_t)((b8 >> 3) & 0x1FU), (uint8_t)((g8 >> 2) & 0x3FU), (uint8_t)((r8 >> 3) & 0x1FU)}})
#else
# define _LV_COLOR_ZERO_INITIALIZER16 {{0x00, 0x00, 0x00, 0x00}}
# define LV_COLOR_MAKE16(r8, g8, b8) (_LV_COLOR_MAKE_TYPE_HELPER{{(uint8_t)((g8 >> 5) & 0x7U), (uint8_t)((r8 >> 3) & 0x1FU), (uint8_t)((b8 >> 3) & 0x1FU), (uint8_t)((g8 >> 2) & 0x7U)}})
#endif
```
right?

kisvegabor, 2021-06-15:
I think so, but I don't see warnings on my side, so I'm not totally sure if it's correct. :slightly_smiling_face:

kisvegabor, 2021-06-29:
As the issue was [somewhere](https://github.com/lvgl/lvgl/pull/2325#issuecomment-864773873) else I close this issue.

closed 2021-06-29 by kisvegabor