lvgl/lvgl #2315: Version confusion with PlatformIO library registry

maxgerhardt, 2021-06-15:
### Perform all steps below and tick them with [x]
(none of these are relevant)
- [X] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [X] Update lvgl to the latest version
- [X] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

The library versions as uploaded and described on https://platformio.org/lib/show/6598/lvgl do not match, by content and date, the versions uploaded in this repository.

Take e.g. the 8.0.0 version from the PlatformIO library registry

[image]

It says that version was uploaded on the 16th March, so pretty much 3 months ago. Looking at your branch https://github.com/lvgl/lvgl/tree/v8.0.0, the latest commit says "Release v8.0.0", from 14 days ago (20th May). How does that work out? A pre-release version was uploaded there, tagged as 8.0.0? Does the 8.0.0 releaseversion keep changing over time?
 
When people use PlatformIO with the library version expression

```ini
lib_deps =
     lvgl/lvgl @ 8.0.0
```

they are getting some 3-month old "8.0.0" version and ask us on what in the world that library version is and why it doesn't match the repo (see: ["Strange version being installed by library manager?"](https://community.platformio.org/t/strange-version-being-installed-by-library-manager/21992)).

### To Reproduce

Create a PlatformIO project and add the above shown [`lib_deps`](https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps) expression to the `platformio.ini`, build the project, then compare the contents of `.pio\libdeps\<env>\lvgl` with what is in the repo for version 8.0.0

### Expected behavior

The versions uploaded to the PlatformIO library manager should be the same as in this repo. Tags should be frozen.

### Screenshots or video
<none applicable>

embeddedt, 2021-06-15:
https://forum.lvgl.io/t/v8-0-is-released/5668/9?u=embeddedt

8.0.0 was mistakenly released to PlatformIO before the actual release. 8.0.1 is just a patch release, so the simplest solution is probably to just remove 8.0.0 from PlatformIO.

@kisvegabor Do you agree?

kisvegabor, 2021-06-15:
Yes, I agree, but for some reason v8.0.1 is not listed:
[image]

BTW, how to remove a release from PIO?

embeddedt, 2021-06-15:
If I had to guess, I would say that it's because PIO is only checking the `master` branch, not the release branches. I think it would actually prevent this issue in the future if it did the opposite, but I'm not sure that's possible.

kisvegabor, 2021-06-15:
I've tried manually `pio package publish` from `release/v8.0` Let's see what happens. (Needs 24 hours)

On a sidenote: [Arduino](https://www.arduinolibraries.info/libraries/lvgl) picked it up correctly.
[image]

kisvegabor, 2021-06-15:
Oh, it was updated immediately. But it was only v8.0.1.

I've published master too and now it looks like this:
[image]

maxgerhardt, 2021-06-15:
>I've tried manually pio package publish from release/v8.0 Let's see what happens. (Needs 24 hours)
>Oh, it was updated immediately. But it was only v8.0.1.

Yes I would expect that too because the `version` field of the `library.json` in the `release/v8.0` branch says `8.0.1`. 

https://github.com/lvgl/lvgl/blob/release/v8.0/library.json#L3

I think re-publishing the repo exactly as seen in the `v8.0.0` tag would work out though, since its version field is 

https://github.com/lvgl/lvgl/blob/c597d257984e2cd3a1c883dc97a26d4512b5e60a/library.json#L3

(on a sidenote though I wasn't actually aware of that you could re-publish the library under the same version multiple times).

Again, if you need a specific version deleted from the PIO registry, I can tag ivankravets here (PlatformIO CEO).

kisvegabor, 2021-06-16:
I'm still confused. PIO picked up releases automatically so far. Or someone published them manually?

embeddedt, 2021-06-16:
I suspect PIO is scanning `master` on a regular basis and using whatever release data is there.

maxgerhardt, 2021-06-16:
>I suspect PIO is scanning master on a regular basis and using whatever release data is there.

They way I have read it in the community forums is that a repository can either use the 'legacy' autocrawler, which scans for new releases or the default branch, or is setup for manual publishing. 

Maybe @ivankravets can assist here with a better library version cleanup.

kisvegabor, 2021-06-22:
Thank you, Ivan!

I've published the master branch (`v8.1.0-dev`) and the `v8.0.1` tag of `lvgl` and `lv_drivers`.
The `library.json` file was incorrect for `lv_driver` on `v8.0.0` but was correct for `lvgl`, so `lvgl` `v8.0.0` is also published.

The only issue is the release dates are incorrect. Is it possible to set them manually?

I'm not sure what policy fits best to PIO. Shall/Can we publish the `master` branches periodically where we have the latest unreleased features and fixes or publish on the releases?

ivankravets, 2021-06-23:
> I've published the master branch (v8.1.0-dev)

I removed again all DEV versions. We don't recommend publishing unstable packages into the registry. If someone needs the latest upstream version, he can use GIT re[psitory directly.

> The only issue is the release dates are incorrect. Is it possible to set them manually?

Thanks, I fixed them manually.

>  Shall/Can we publish the master branches periodically 

Please publish new package versions when you make final release.

kisvegabor, 2021-06-25:
Thank you @ivankravets. We will do it as you suggested.

As the versions are cleaned up I close this issue.

closed 2021-06-25 by kisvegabor