lvgl/lvgl #2318: undefined reference to `lv_example_tabview_2'

Johnxjj, 2021-06-16:
When I try to use the lv_example_tabview_2() example, a compilation error is reported. Is it because the file lv_example_tabview_2.py is missing.

Johnxjj, 2021-06-16:
[image]

kisvegabor, 2021-06-16:
`lv_example_tabview_2` was added a few days ago.

Be sure to update lvgl to the latest version and Refresh the Eclipse project.

kisvegabor, 2021-06-29:
As the [lv_example_tabview_2](https://github.com/lvgl/lvgl/blob/master/examples/widgets/tabview/lv_example_tabview_2.c) is really there I close this issue.

closed 2021-06-29 by kisvegabor

Johnxjj, 2021-06-30:
@kisvegabor I am not referring to lv_example_tabview_2.c, but lv_example_tabview_2.py. Whether this file is useful for emulation on PC.

kisvegabor, 2021-07-01:
Oh, thank.
@uraich Could you take a look at this example?

uraich, 2021-07-02:
@kisvegabor @Johnxjj :
For the moment, I uploaded it to https://github.com/uraich/lv_mpy_examples_v8/tree/main/widgets/tabview
If you are happy with it, I can create a PR to integrate it into lvgl. To me, it looks like it behaves the same way as the C original.

kisvegabor, 2021-07-06:
Thank you @uraich.

Yes, please send a PR :slightly_smiling_face:

uraich, 2021-07-06:
It is already merged into master.

kisvegabor, 2021-07-06:
I'm catching up with my email, it seems I didn't get there yet :slightly_smiling_face: