lvgl/lvgl #2320: set_px_cb or disp_flush is not invoked

epikao, 2021-06-16:
### Describe the bug

### To Reproduce

embeddedt, 2021-06-17:
Can you enable logging and see if anything gets printed?

epikao, 2021-06-17:
> Can you enable logging and see if anything gets printed?
If activate LOG, I get following warning when compiling:
[image]

with LV_LOG_LEVEL_WARN:
```
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_mem.c@203->lv_mem_alloc
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_mem.c@307->lv_mem_realloc
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_mem.c@516->_lv_mem_buf_get
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_debug.c@127->lv_debug_log_error
```

with LV_LOG_LEVEL_TRACE:
```
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_refr.c@309->_lv_disp_refr_task 
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_task.c@192->lv_task_handler    
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_widgets\lv_label.c@79->lv_label_create 
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@237->lv_obj_create       
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@392->lv_obj_create       
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_widgets\lv_label.c@167->lv_label_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_task.c@74->lv_task_handler     
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_refr.c@177->_lv_disp_refr_task 
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_refr.c@309->_lv_disp_refr_task
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_task.c@192->lv_task_handler
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_widgets\lv_label.c@79->lv_label_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@237->lv_obj_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@392->lv_obj_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_widgets\lv_label.c@167->lv_label_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_task.c@74->lv_task_handler
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_refr.c@177->_lv_disp_refr_task
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_refr.c@309->_lv_disp_refr_task
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_task.c@192->lv_task_handler
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_widgets\lv_label.c@79->lv_label_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@237->lv_obj_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@392->lv_obj_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_widgets\lv_label.c@167->lv_label_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_task.c@74->lv_task_handler
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_refr.c@177->_lv_disp_refr_task
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_refr.c@309->_lv_disp_refr_task
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_task.c@192->lv_task_handler
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_widgets\lv_label.c@79->lv_label_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@237->lv_obj_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@392->lv_obj_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_widgets\lv_label.c@167->lv_label_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_task.c@74->lv_task_handler
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_refr.c@177->_lv_disp_refr_task
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_refr.c@309->_lv_disp_refr_task
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_misc\lv_task.c@192->lv_task_handler
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_widgets\lv_label.c@79->lv_label_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@237->lv_obj_create
my_print DEBUG
.pio\libdeps\teensy31\lv_arduino\src\src\lv_core\lv_obj.c@392->lv_obj_create
```

kisvegabor, 2021-06-17:
Where/How do you call [lv_tivk_inc()](https://docs.lvgl.io/master/porting/tick.html)?

epikao, 2021-06-17:
instead of "Serial.println..." I put a counter++ into disp_flush callback, and Serial.println(counter) into set_px_cb callback.
Now it looks both callbacks are invoked, the counter is incremented.

So I think the problem was that my Serial.println in the set_px_cb blocked(or something else) the Serial.println of disp_flush callback. 

Sorry for the circumstances

kisvegabor, 2021-06-18:
Glad to hear it works. I close this issue then.

closed 2021-06-18 by kisvegabor