lvgl/lvgl #2321: lv_btnmatrix_get_selected_btn not working within an event callback

amartini-hopper, 2021-06-17:
### Perform all steps below and tick them with [x]
- [ x ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)
^ I tried for some time to get the GDB debugging working with Big Sur to run teh simulator on OSX. Unfortunately, I failed but do have a simple [sketch](https://github.com/acumartini/lvgl-playground) working on my teensy device.

### Describe the bug

Within a btnmatrix event callback `lv_btnmatrix_get_selected_btn` returns `65535` for both `event->current-target` and  `event->target` regardless of having explicitly set a selected btn with `lv_btnmatrix_set_selected_btn` outside of the event callback.

### Expected behavior

- `lv_btnmatrix_get_selected_btn` should return the id of the selected btn or preferably -1 when nothing is selected, although I see the return type is a uint16_t.
- `lv_btnmatrix_get_selected_btn` should work both inside and outside of an event callback context. When within an event callback context I would expect `event->current_target` to point to the btnmatrix object.

### Screenshots or video

amartini-hopper, 2021-06-17:
Sry I accidentally opened this issue with my work account. `acumartini` is my personal and I have contributed in the past to the lv_arc widget. Thanks!

amartini-hopper, 2021-06-17:
Note: this behavior worked as expected in lvgl v7

amartini-hopper, 2021-06-17:
OK to be more specific. The issue is that `lv_btnmatrix_get_selected_btn` on `event->code == LV_EVENT_VALUE_CHANGED`, but is set to `65535` when the events is `event->code == LV_EVENT_RELEASED || event->code == LV_EVENT_PRESS_LOST`. If this is expected behavior I can keep track of the `LV_EVENT_VALUE_CHANGED` id state for use with subsequent events.

embeddedt, 2021-06-17:
I haven't had a chance to try the code sample yet, but if you are unchecking the button, that sounds like the expected behavior, as there won't be a selected button anymore when `LV_EVENT_RELEASED` gets fired.

amartini-hopper, 2021-06-17:
Correct that is what I am doing. The btn state was never unchecked after release in v7 so I had to do it manually, and it appears that behavior remains the same in v8. It looks like what has changed is that the btmn state is registering that there is no selected btn after release, but the state of the actual btn remains `CHECKED`.

amartini-hopper, 2021-06-17:
The real expected behavior here is that pressing a `CHECKED` btn on the btnm should uncheck its state, but because it did not work that way in v7 I was thinking that was not the intended behavior and wanted to focus on what changed here in v8.

embeddedt, 2021-06-17:
> the btmn state is registering that there is no selected btn after release

The behavior change might have been caused by the move to using events for internal widget logic as well. It appears that `btn_id_sel` is [explicitly cleared](https://github.com/lvgl/lvgl/blob/master/src/widgets/lv_btnmatrix.c#L491) in the button matrix's RELEASED event handler, which I am pretty sure runs before yours. The simplest solution does appear to be to use VALUE_CHANGED as you suggested, as that gets fired before the selected button is cleared.

In the future it would be good for us to have unit tests for this kind of thing, which is being worked on slowly in another branch. That way the expected high-level behavior is known and shouldn't change even if the API changes.

> The real expected behavior here is that pressing a CHECKED btn on the btnm should uncheck its state, but because it did not work that way in v7 I was thinking that was not the intended behavior and wanted to focus on what changed here in v8.

The conditions for a button being unchecked automatically are listed [here](https://github.com/lvgl/lvgl/blob/7066c8fbbb082934bc1da8251571d23da7b4d979/src/widgets/lv_btnmatrix.c#L465-L467). Do you have the one_check mode enabled on your button matrix? If so, the button will never be unchecked, as the expectation there is that you always want exactly one button checked at any given time (not zero or two).

amartini-hopper, 2021-06-17:
> Do you have the one_check mode enabled on your button matrix? If so, the button will never be unchecked, as the expectation there is that you always want exactly one button checked at any given time (not zero or two).

Yes that is exactly what is going on. I forgot that was the condition that changed the unchecking behavior. By manually keeping track of the last_id and deselecting the selected btn on release I am essentially created a third btnm mode, which would be something like `ONE_OR_NONE_CHECKED`.

But I do think that have the btnm selected state change when releasing a selected btn in `ONE_CHECK` modes seems wrong.

embeddedt, 2021-06-18:
(cc @kisvegabor)

I think the name "selected" is misleading. In v7 there were [three separate states](https://github.com/lvgl/lvgl/blob/042e34cb880618ee8ba180a642697ce4653bd2af/src/lv_widgets/lv_btnmatrix.h#L57-L59). In v8 these were flattened into one "selected" state, but "selected" is not the best term to use here, because it really means whichever button is currently focused or pressed, not whichever button was last touched and is now selected.

kisvegabor, 2021-06-18:
I've flattened the three states into one because they were almost the same and I thought it's difficult for the users the select what kind of ID should they get. Besides focused, pressed and act could be different which is also misleading.

With "selected" my intention was to show which was lastly changed/affected button by focusing or clicking. What term would be better? "active"? 

@amartini-hopper 
Just be sure I understood the issue correctly. The problem is that you want one or zero buttons checked so trying to manually uncheck the previously checked button if it's different from the button just clicked. 

In v7 it worked because `ONE_CHECK` mode didn't force exactly one button to be checked, but one or zero. But v8 forces one button to be checked.

Is this summary correct?

amartini-hopper, 2021-06-18:
I think the flattened state makes sense, but the behavior for the selected state seems to be a bit off during a `LV_EVENT_RELEASED` or `LV_EVENT_PRESS_LOST` event.

@kisvegabor 
No not quite the right summary. Here are the order of operations to recreate the problem:
- set btnm to `ONE_CHECK` mode
- select one of the buttons
- pressing the selected button triggers a `LV_EVENT_VALUE_CHANGED` at which point `lv_btnmatrix_get_selected_btn` works as expected
- releasing the button triggers a `LV_EVENT_RELEASED` event  at which point `lv_btnmatrix_get_selected_btn` returns `65535`

I am not exactly sure why the behavior was different in v7. The behavior for `ONE_CHECK` was the same as it is in v8 where it "forced exactly one button to be checked", so I had to uncheck the button manually similar to the sketch above. What has changed in v8 is the behavior of `lv_btnmatrix_get_selected_btn` in the context of `LV_EVENT_VALUE_CHANGED` and `LV_EVENT_RELEASED` event callbacks, which appears to be connected to the flattening of the btn state. Although, I don't think we have pinpointed the issue yet.

kisvegabor, 2021-06-22:
Thanks, got it now. :slightly_smiling_face: 

This event handler should do the job:
```c
static void event_handler(lv_event_t * e)
{
    static uint16_t last_id = LV_BTNMATRIX_BTN_NONE;
    static uint16_t act_id = LV_BTNMATRIX_BTN_NONE;
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    if(code == LV_EVENT_VALUE_CHANGED) {
        act_id = lv_btnmatrix_get_selected_btn(obj);
        const char * txt = lv_btnmatrix_get_btn_text(obj, act_id);
        LV_LOG_USER("%s was pressed", txt);

    }
    else if(code == LV_EVENT_RELEASED || code == LV_EVENT_PRESS_LOST) {

        if(last_id == act_id) {
            lv_btnmatrix_clear_btn_ctrl(obj, act_id, LV_BTNMATRIX_CTRL_CHECKED);
        }
        last_id = act_id;
    }
}

```

amartini-hopper, 2021-06-23:
Thanks @kisvegabor! This method is working perfectly for now. Not sure if you want to leave this open or close it. I am satisfied with the behavior, but thought it might be a bug worth further investigation.

kisvegabor, 2021-06-28:
I don't think it's buggy just different from v7.

It might be more intuitive to keep the selected button in the RELEASED event too but it's not so simple in v8.

Anyway, as the code above works, I close this issue for now.

closed 2021-06-28 by kisvegabor