lvgl/lvgl #2326: Drawing polygon interface, how to make it support concave polygon

ihuoxin, 2021-06-19:
### Perform all steps below and tick them with [x]

- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Draw a polygon. Only convex polygons are supported

Drawing polygon interface, how to make it support concave polygon

### To Reproduce

### Expected behavior

Drawing polygon interface,support concave polygon.

### Screenshots or video

Must present

ihuoxin, 2021-06-21:
Ask for expert help

kisvegabor, 2021-06-21:
It's planned to implement a more versatile polygon drawer but it's rather difficult considering:
- anti-aliasing support
- no float type
- small flash usage
- small RAM usage for intermediate buffers
- fast rendering

If you have less strict requirements you can use a custom polygon drawer and use [lv_canvas](https://docs.lvgl.io/master/widgets/core/canvas.html) to display the resulting image.

ihuoxin, 2021-06-22:
Thank you for your help. I have implemented a simple polygon interface.

ihuoxin, 2021-06-22:
Thank you for your help. I have implemented a simple polygon interface.

> It's planned to implement a more versatile polygon drawer but it's rather difficult considering:
>
> * anti-aliasing support
> * no float type
> * small flash usage
> * small RAM usage for intermediate [...]

Thank you for your help. I have implemented a simple polygon interface.

kisvegabor, 2021-06-22:
Great! If you are interested in contributing to LVGL feel free to open a PR.
New widgets like this should go to the [extra](https://github.com/lvgl/lvgl/tree/master/src/extra) folder.

closed 2021-06-22 by kisvegabor