lvgl/lvgl #2328: Support use of ps_malloc

pscs, 2021-06-21:
On an ESP32-WROVER there is lots of PSRAM available, but it appears that it can't be used with LVGL's built-in memory pool functions, so you would have to use LV_MEM_CUSTOM_ALLOC set to ps_malloc, etc.

lv_mem_init can use #define LV_MEM_ADR to use a memory pool at a fixed address but that won't work with an address determined at runtime by ps_malloc. Ideally you'd be able to use 'ps_malloc()'to allocate the pool, and then pass that address/size to lv_init at runtime.

kisvegabor, 2021-06-21:
> On an ESP32-WROVER there is lots of PSRAM available, but it appears that it can't be used with LVGL's built-in memory pool functions, so you would have to use LV_MEM_CUSTOM_ALLOC set to ps_malloc, etc.

What is the problem with setting LV_MEM_CUSTOM_ALLOC to ps_malloc?

pscs, 2021-06-21:
> What is the problem with setting LV_MEM_CUSTOM_ALLOC to ps_malloc?

Memory fragmentation and no limit on size of memory usage. The lv_mem_init function already supports creating the tlsf pool in a specific part of memory, it's just that the address cannot be specified at runtime.

I could easily fork lvgl and add the couple of lines necessary to allow the address to be passed as a parameter, but it seems that it would be something that may be of general interest (possibly replacing the LV_MEM_ADR compile-time setting)

embeddedt, 2021-06-21:
So, if I'm understanding correctly, you're saying that ESP32 provides no way to statically allocate something in PSRAM, and their dynamic allocator is not suitable for use with LVGL?

If so, the simplest and least invasive option here is probably to allow users to specify a custom function in `LV_MEM_ADR` which returns the address of the heap. We would guarantee that the function will only be invoked when `lv_init` runs. Then, you can call `ps_malloc(LV_MEM_SIZE)` inside that function and return a chunk of PSRAM which TLSF would use.

pscs, 2021-06-21:
> So, if I'm understanding correctly, you're saying that ESP32 provides no way to statically allocate something in PSRAM

Not that I've been able to find anywhere (I'm willing to be proven wrong here). The ESP32 heap functions don't seem to have any way to exclude an address range from the heap (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/mem_alloc.html)

> and their dynamic allocator is not suitable for use with LVGL?

Well, it's as suitable as 'malloc' is - but given that lvgl (by default) doesn't use malloc for allocating its own blocks, and instead uses tlsf, then I presume that there's a reason for that...

> If so, the simplest and least invasive option here is probably to allow users to specify a custom function in `LV_MEM_ADR` which returns the address of the heap. We would guarantee that the function [...]

That would also work.

kisvegabor, 2021-06-21:
> If so, the simplest and least invasive option here is probably to allow users to specify a custom function in LV_MEM_ADR which returns the address of the heap. We would guarantee that the function [...]

I like this idea. I've tested it with `#  define LV_MEM_ADR          malloc(LV_MEM_SIZE)`. It works but the `#if LV_MEM_ADR == 0` in [lv_mem_init()](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_mem.c#L85) gives error:
```
Missing binary operator before token "("
```
 
Do you have any idea how the replace `#if LV_MEM_ADR == 0`?

pscs, 2021-06-21:
I don't think you could just do that because I think () in a #if conditional statement is invalid

You could add another #define

eg

#if LV_MEM_ADR == 0
#ifdef LV_MEM_ADR_FN
    tlsf = lv_tlsf_create_with_pool((void *)LV_MEM_ADR_FN(LV_MEM_SIZE), LV_MEM_SIZE);
#else
    /*Allocate a large array to store the dynamically allocated data*/
    static LV_ATTRIBUTE_LARGE_RAM_ARRAY MEM_UNIT work_mem_int[LV_MEM_SIZE / sizeof(MEM_UNIT)];
    tlsf = lv_tlsf_create_with_pool((void *)work_mem_int, LV_MEM_SIZE);
#endif
#else LV_MEM_ADR
    tlsf = lv_tlsf_create_with_pool((void *)LV_MEM_ADR, LV_MEM_SIZE);
#endif

kisvegabor, 2021-06-22:
> I don't think you could just do that because I think () in a #if conditional statement is invalid. The expression has to be an integer expression.

I've asked it on StackOverflow and got some support tricky solutions. I've tested this and it worked well
https://stackoverflow.com/a/68074912/1999969

embeddedt, 2021-06-22:
That's some very complicated preprocessor logic just to handle one backwards-compatibility case (very creative though). In my opinion, @pscs' solution makes more sense to the code reader and the extra macro doesn't clutter `lv_conf.h` that much. However, totally up to you which direction you wish to proceed. :slightly_smiling_face:

kisvegabor, 2021-06-22:
I'm also worried about the complexity of the macros and  not sure every compiler can digest it :smile: 

So probably a new define would be better. What about `LV_MEM_POOL_ALLOC` instead of `LV_MEM_ADR_FN`?

embeddedt, 2021-07-07:
Not stale; it would be ideal from an overall portability perspective to allow allocating a custom pool at runtime without fully committing to using the platform's provided heap manager.

This is a simple change to tackle if someone would like to contribute. :slightly_smiling_face:

kisvegabor, 2021-07-07:
I agree to leave it open for contributors!

kisvegabor, 2021-08-14:
> Even editing LV_MEM_CUSTOM_ALLOC to ps_malloc.

Why wasn't it working? What happened?

Miguel0101, 2021-08-14:
> > Even editing LV_MEM_CUSTOM_ALLOC to ps_malloc.
> 
> Why wasn't it working? What happened?

Because my ESP32 keeps restarting.

kisvegabor, 2021-08-17:
What id the log on restart?

Miguel0101, 2021-08-17:
> What is the log on restart?

Memory fragmentation and no limit on size of memory

kisvegabor, 2021-08-17:
Okay, we have left this open because it can be easily implemented by contributors. 

If you are interested in sending a PR with this feature please check this issue [from this comment](https://github.com/lvgl/lvgl/issues/2328#issuecomment-865332591).

commit 36ddbd07 (2021-08-17): feat(mem) LV_MEM_POOL_ALLOC (#2458)

fixes  #2328

* Add LV_MEM_POOL_ALLOC #2328

* Add LV_MEM_POOL_ALLOC #2328

* Update lv_conf_template.h
closed 2021-08-17 by kisvegabor