lvgl/lvgl #2334: lv_obj_align_to fails when using flexlayout and LV_SIZE_CONTENT

grarup, 2021-06-27:
### Perform all steps below and tick them with [x]
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
lv_obj_align_to givs a wrong result when used in a flex layout with a and an other object uses size for content
using lv_obj_align seems to work.
### To Reproduce
Run this code.
```c
    lv_obj_t * root = lv_obj_create(NULL);

    lv_obj_set_layout(root, LV_LAYOUT_FLEX);
    lv_obj_set_flex_flow(root, LV_FLEX_FLOW_ROW);
    lv_obj_set_flex_align(root, LV_ALIGN_CENTER, LV_ALIGN_CENTER, LV_ALIGN_CENTER);

    lv_obj_t * left = lv_obj_create(root);
    lv_obj_set_width(left, LV_SIZE_CONTENT);
    lv_obj_set_height(left, LV_PCT(100));

    lv_obj_t * center = lv_obj_create(root);
    lv_obj_set_style_bg_color(center, lv_color_hex(0xFF0000), LV_STATE_DEFAULT | LV_PART_MAIN);
    lv_obj_set_style_bg_opa(center, 255, LV_STATE_DEFAULT | LV_PART_MAIN);
    lv_obj_set_style_pad_all(center, 50, LV_STATE_DEFAULT | LV_PART_MAIN);
    lv_obj_set_height(center, LV_PCT(100));

    lv_obj_t * right2 = lv_obj_create(root);
    lv_obj_set_width(right2, 100);
    lv_obj_set_height(right2, LV_PCT(100));

    lv_obj_set_flex_grow(center, 1);

    lv_obj_t * label = lv_label_create(center);
    lv_label_set_text(label, "Hello world");
    lv_obj_align_to(label, center, LV_ALIGN_CENTER, 0, 0);

    lv_obj_t * label2 = lv_label_create(left);
    lv_label_set_text(label2, "long text here why?");
    lv_obj_set_align(label2, LV_ALIGN_CENTER);

    lv_scr_load(root);

```
### Expected behavior
Object is aligned correctly

### Screenshots or video
[image]

kisvegabor, 2021-06-28:
Hi,

It happens because `lv_obj_align_to` aligns the object once, and doesn't realign it when the parent's size changes. And the parent's size changes when you create `label2`.

On the contrary, `lv_obj_align` realigns the object when needed. 

So the solution is either:
1. Use `lv_obj_align`
2. Use `lv_obj_align_to` when all sizes are set (all the objects are created in this case) 
3. Use a layout instead of `lv_obj_align`.

grarup, 2021-06-28:
Shouldn't `lv_obj_align_to` subscribe to an event of what it aligns to and then change when it fires?

kisvegabor, 2021-07-01:
It'd be great but the problem is that the "object to align" needs to keep a reference to an unrelated object ("base") . Therefore any object can be a potential "base" of an object. So if an object's size or parent changes or it's deleted all the other objects need to be checked if any of the other objects have this object as "base".

So it's too complicated, error-prone and performance-hungry.. :slightly_frowning_face:

grarup, 2021-07-16:
Couldn't this also be solved with subscribing to LV_EVENT_DELETE, LV_EVENT_SIZE_CHANGED on the base object, and then updating when this is fired. Then only the objects with base needs to update.

kisvegabor, 2021-07-16:
In this case, the "base" should store the related "objects to align" in an array. And "object to align" should also store a reference to "base". This way the objects can see all the connections regardless to they are "bases" of "objects to align".

Using `LV_EVENT_DELETE` and `LV_EVENT_SIZE_CHANGED` can work but if the base is scrolled or moved all the objects needs to be realigned. 

So it seems possible but IMO too complicated compared to the gain. In the projects I've already used v8 I found that when `lv_obj_align_to` seems useful a grid layout is more powerful and a better choice.

closed 2021-08-08 by lvgl-bot